tock/tock #2564: Tracking: Update `static_init!()` to `singleton!()` semantics

bradjc, 2021-04-27:
This is the tracking issue for the discussion on #1447. At a high level, if static_init is called twice it can lead to mutable memory aliasing.

- [ ] Update `static_init!()` to have `singleton!()` semantics.
  - Singleton: https://docs.rs/cortex-m/0.7.2/cortex_m/macro.singleton.html
  - Will not be able to disable interrupts.

Optional, or more discussion needed:
- [ ] Provide a safe interface for `static_init!()` if there is some way to guarantee it's safe whenever used.
  - https://github.com/tock/tock/issues/1447#issuecomment-827336421
  - Intent is it is only used during initialization before the kernel loop starts.

kupiakos, 2022-08-05:
Note: if a type is [`FromBytes`](https://docs.rs/zerocopy/latest/zerocopy/trait.FromBytes.html), meaning any bit pattern is valid, it could be safely mutated on a single-core system even during an interrupt, as the most any race could do is create an arbitrary byte pattern in the value, which is safe to do according to `FromBytes`'s safety guarantees. This doesn't solve the multiple `&mut` issue.

hudson-ayers, 2022-10-17:
https://github.com/tock/tock/pull/3239 took us almost all the way there, all that is left is making the macro safe to call from single-core systems

bradjc, 2023-05-26:
I propose we close this.

hudson-ayers, 2023-06-25:
I still think a safe `static_init!()` would be a very significant upgrade to Tock, so I would prefer we keep this open, even if no one is working on it now.

kupiakos, 2023-07-07:
An idea I'd brought up in a previous call: rather than protecting each singleton with a flag, we could protect a group of singletons with a single flag, or even better, have a primitive for ensuring a block of code can only run once.