tock/tock #2568: RFC: Allocate a "KernelGrant" type when allocating `Grant<T>`, use this for syscall correctness tracking

bradjc, 2021-05-03:
This approach, would, I think, allow us to enforce all allow and subscribe syscall restrictions at the cost of memory space in a process's grant region. The high-level idea is every time a grant is allocated for a process, the kernel would allocate its own structure as well. That structure would store any metadata required to properly verify that processes and capsules are correctly following all of our syscall requirements.

## Design

[image]

The new `KernelGrant` type would be dynamically allocated with the grant's `T` type in process memory when the grant is allocated for a process. The type might look something like:

```rust
struct KernelGrant {
  appslices: [(*const u8, usize); NUM_APPSLICES],
  upcalls: [Option<DriverNum>; NUM_UPCALLS],
}
```

- Appslices: when a new appslice is created, the kernel iterates every allocated grant region for that process, and verifies no aliasing.
- Upcalls: upcalls are tagged with `Option<DriverNum>` as described here: https://github.com/tock/tock/pull/2462#issuecomment-796843381. But, since the `KernelGrant` stores a copy of the last `Option<DriverNum>` this resolves the issue where a capsule could return `None` every time since the kernel now knows the upcall had been subscribed before.

## Drawbacks

This uses more process memory for grants. Hopefully not too much (how many upcalls and appslices do capsules use? one or two?), and I think that redudant information between `T` and `KernelGrant` can be removed. For example, appslice ptr and length only needs to be stored in `KernelGrant`, and the `AppSlice` type can lookup that information in `KernelGrant` (I think). And the memory requirement could be statically calculated just by looking at the code, so processes could theoretically calculate how much grant memory they need based on which capsules they use.

I can imagine how to building this in C very easily (as in the code structure, not necessarily getting the code correct), but I'm not sure what Rust roadblocks could exist.

## Advantages

We would now have a type and mechanism for allowing the kernel to actually manage grant allocations, rather than the mostly ad-hoc mechanism we have now (where the core kernel doesn't even know which capsule is responsible for which grant). 

There would be no one-grant-per-capsule restriction for upcalls.

The grant allocation code already exists, so it wouldn't require building an entirely new mechanism for tracking appslices.

lschuermann, 2021-05-03:
I like this approach in general, it's also very close to what @hudson-ayers and I had in mind when considering the implications of different approaches for preventing appslice buffer aliasing.

As discussed on Slack, it should be rather easy to determine the number of `AppSlice`s and `Upcall`s by using a `GrantDefault` trait similar to #2462, and constructing the `AppSlice`s and `Upcall`s using a factory pattern. This factory can also produce structs which just index in the `KernelGrant` region, as you described.

##### Upcalls

However, since I've given this some thought before, I'm not entirely sure whether we'd see any benefits over the mechanisms to be introduced in #2462, for `Upcall`s at least. The proposed approach there introduces very little cost at runtime, both in terms of memory and computational overhead, since it makes use of the Rust type system to the farthest we could reasonably make it work (i.e. using the `GrantDefault` to create process-bound `Upcall`s which hold the `ProcessId`, `driver_num` and `subdriver_num` anyways). It also means that accessing the `Upcall` data for scheduling doesn't go through an additional deference step. This might of course allow multiple Grant regions per capsule.

##### AppSlices

Further, it's still going to be somewhat complex to efficiently decide whether a to-be allowed `AppSlice` overlaps with one already allowed for a process, as we'd have to iterate through each `AppSlice` of each `Grant` to be sure it doesn't overlap with another one already allowed. Compared to some arbitrary data structure not bound to the different `Grant` regions, here we'd be limited to the way `AppSlice`s are distributed across `Grant`s and can't sort them to be able to perform a potentially more efficient binary search.

I imagine a combination of the mechanisms introduced for upcalls in #2462, together with either this, or a volatile / `UnsafeCell`-based accessing method of `AppSlices` might be the most efficient solution and a good balance between required memory and computational overhead. I'm currently writing up a document to outline whether & why `UnsafeCell` might work for us here. I suppose benchmarking this and the other volatile/`UnsafeCell` approach will tell us the route to take.

bradjc, 2021-07-02:
This has been implemented for Upcalls in #2639.

closed 2021-07-07 by bors[bot]