tock/tock #2579: Tracking: Release new version of tock-register-interface

andre-richter, 2021-05-08:
Now that https://github.com/tock/tock/pull/2517 has landed, please consider releasing a new version on crates.io.

This will allow me to deprecate https://github.com/rust-embedded/register-rs and refer people to `tock-register-interface` for future use.

Thanks in advance,
Andre

@ppannuto @lschuermann

lschuermann, 2021-05-09:
Thanks for this tracking issue! Releasing a new version is definitely warranted due to this change, so it has been on my list for some time (rather, poking someone to do it :smile:).

That being said, I still have a few short term todos which I think should be done prior to a release:
- [ ] splitting up the internal structure, such that the interfaces, register types, and helpers are cleanly and semantically separated in different modules (that is already somewhat reflected in Tock's `kernel`-crate re-exports of this library, but should really be done in the library itself)
- [ ] adding a feature-flag to exclude the "default" register types (`ReadOnly`, `ReadWrite`, etc.), allowing this to be used as an interface-only library (following the previous change, essentially just disabling importing the module containing these types). A major advantage is that enabling this feature should then remove all unsafe code from the library. 
- [ ] doing a final pass over the doc-comments, and adding some documentation tests and more usage examples

I have some changes ready locally, and I expect to have all the boxes ticked in 1-2 weeks.

Thanks for your input on this, @andre-richter!

ppannuto, 2021-05-10:
Acking this thread, and eager for @lschuermann's updates! 

1 and 3 here are definite wins. I felt like there was possibly some unresolved discussion around 2 in the call last Friday, but we can re-examine those questions where there is a concrete proposal to discuss.

andre-richter, 2021-06-14:
@lschuermann please see https://github.com/tock/tock/pull/1489

hnj2, 2021-06-21:
Hello, I was wondering what the status of the release is?
According to #2618 it is nearly ready, but if that isn't the case a small version bump of `tock-registers` to `0.6.1` to make #2589 available might be nice for nightly users?

lschuermann, 2021-06-23:
@hnj2 Sorry for the wait! It seems that the only blocking PR for a release is #2551 and writing a Changelog, and I would've hoped that this is a rather uncontroversial set of changes. However, there seems to be some (entirely valid) discussion around this.

I'm not sure. Depending on what @ppannuto and @bradjc think a point release may be justified as the nightly breakage seems to be a rather pressing issue for the wider community.

v0.6.0 corresponds to 72a1e1ea759359, I'll backport #2589 to that on another branch such that we can create a v0.6.1 point release if @ppannuto agrees.

lschuermann, 2021-06-23:
Mhm, backporting this puts us in a weird position, as `tock-register-interface` is maintained in the Tock repository: by updating the Rust version, we break a lot of existing code in Tock, so we'd create a point release on something with a very red CI :smile:. It's not really feasible to reconstruct a version of Tock compiling on a newer nightly (2021-05-27) on top of 72a1e1ea759359. Therefore, I've created a commit a892097f57f9c5 (branch [tock-register-interface-v0.6.1](https://github.com/tock/tock/tree/tock-register-interface-v0.6.1)) which is based on 72a1e1ea759359 and only bumps the Rust version + backports the essential change of #2589. `tock-register-interface` seems to work fine, `cargo {check, build, test, doc}` work, but the rest of tock does not. Also, we're then having a point release of `tock-registers` not corresponding to any commit on master and any tag. Let's see what others think.

commit 0e1b1489 (2021-07-01): tock-registers: release v0.7

**This is expected to be the final release before 1.0**

v0.7 CHANGELOG

 - #2642: Rename `IntLike` to `UIntLike` to match semantics
 - #2618: Reorganize, document, and feature-gate modules and exports
 - #2589: Upgrade nightly for `const_fn` -> `const_fn_trait_bound`
 - #2517: Use traits for accessing / manipulating registers
 - #2512: Fix `Copy` and `Clone` implementation on `Field`
 - #2300: Add support for `usize`
 - #2220: Remove duplicate code, make local register copy read-write
 - #2210: Add `u128` to `IntLike`
 - #2197: Accept trailing comma in bitfields  [... cut]
closed 2021-07-02 by bradjc