tock/tock #2593: tock-registers: Make Register Access Unsafe

Cryptjar, 2021-05-22:
_**TL;DR** IMO the current `get` and `set` functions are potentially unsound, thus I propose to declare them all as `unsafe` or at least allowing to do so optionally._

First of all, let me say, that I'm a user of the `tock-registers` crate as it is on crates.io, I do not have any insights into Tock itself, and I'm arguing here only in regard to `tock-registers` as a library.

I'm aware of the recent changes due to #2517 and the current release tracking issue #2579, actually, that is partially why I come up with this now. In my personal opinion, this (#2517) is going in a very good direction. But I believe, we can do even better. And since, there is already an upcoming breaking release planed, why not include this too, so only a single breaking release has to be done. I.e., I propose this to be included in #2579.

Motivation
----------

One of the most magnificent features of Rust (IMO) is the distinction between `unsafe` Rust and Safe Rust. This allows Safe Rust to invoke any safe function in any way (that the type system permits), and it will always be _sound_, i.e., it will never invoke _Undefined Behavior_.

Going directly to MMIO, there are at least two things that must be upheld when writing or reading to/from an MMIO memory address:
	1) that an address (i.e. a pointer) must be valid as required by Rust and points to the intended device
	2) the respective device and system must be in a valid state, MMIO is _no memory_ and each access individually must be valid

In my opinion, the first point has been already dealt with (i.e., requiring the constructor to ensure this contract is upheld, so the methods can rely upon that 1) is upheld). However, to the best of my knowledge, the second one is yet not handled satisfactorily (IMO), i.e. practically the constructor also has to ensure that all subsequent read & writes (in any order and with any value) will be valid, which practically is borderline impossible (depending on the device of course).

To illustrated let me give you a practical example that I recently encountered:

I'm currently working with a Raspberry Pi 4 board, and it has (like most others) an UART, in particular a so call 'Mini UART', which, for reference, is specified in Chapter 2 of the [manual](https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf#chap-Auxiliaries). In particular, it has the `AUX_ENABLES` register besides the other UART access registers. One paragraph in its specification caught my attention:

> If the enable bits are clear\[ed\] you will have **no access** to a peripheral.
> You can not even read or write the registers!

This clearly states that whether it is sound (i.e. not UB) to write or even read any of the other UART registers depends on whether this `AUX_ENABLES` register is set (i.e. contains a `1`). This is a clear contract that the hardware defines. And to the best of my knowledge, there is no way to express this satisfyingly with `tock-registers`. This is basically due to the fact that all read & write functions (i.e. `set` & `get`) are defined as _safe_ functions that must be sound to be called in any situation (this is essentially what Rust means with _safe_).

In other words, if I, as a library user, would simply put both the `AUX_ENABLES` register and the others into a single register struct, one could first write a `0` into `AUX_ENABLES` and then access one of the other registers, both without any `unsafe` usage, which makes this entire struct fundamentally **unsound**.

The Solution
------------

Well, there are actually a lot of options here, some don't even involve `tock-registers` itself, e.g. the current way to fix this unsoundness, is, as it is teased above, to **not** put such registers into a single `tock-registers::register_structs!` struct. But, in my opinion, this is neither an intuitive nor practical solution.

Instead, consider what I actually did (shame on me, but I assume that other would do that too), I put it all together and just say: 'it might be (internally) unsound, but it's stuffed inside a private module, so that unsoundness is at least not exposed' sort of things. Which is actually a very bad thing to do, because _implicitly_ that entire private module would be _implicitly unsafe_, that is a concept that technically does not exist in Rust and is rather a C-like mindset.

The better way, obviously, would be to make said private module _explicitly_ unsafe that is by using the `unsafe` keyword where appropriate. However, as a user of `tock_registers::register_structs` macro, I can't do this. And this is where the `tock_registers` crate comes into the picture.

So, the lazy solution, in my opinion, would be to simply declare all relevant read and write methods on MMIO registers as `unsafe` (considering #2517 that would mean the trait functions). I think, tho, that this is perfectly fine, because, I (at least) use this library only to get symbolic names for my memory addresses and getting the nasty bit manipulations done right. This is, in my opinion, what this library is essentially about. Then downstream, I use this symbolic interface to create a device-specific safe-wrapper anyway, which since I have to comply with the hardware requirements has an _unsafe tone_ to it anyway. So, I wouldn't care whether these registers, which are from my perspective an internal API, are `unsafe` or not, in fact making them `unsafe` would make said unsafe tones explicit via the `unsafe` keyword.

Open Questions
----------------------

I think there is quite some wiggle room on how to solve this (thus I called it above 'the lazy solution'). To mention a more flexible approach I like to point to the [`VolAddress` crate](https://docs.rs/voladdress/0.4.1/voladdress/struct.VolAddress.html) (specifically the `R` & `W` generics). It also intends to wrap accesses to MMIO memory locations, and it actually encodes in the type system, whether it is henceforth safe to read and/or write the respective MMIO location, i.e. whether the `unsafe` keyword is required, indicating some underlying contract. So in my example, for instance, I would declare reading the `AUX_ENABLES` register save, but writing it or accessing any other register `unsafe`.

lschuermann, 2021-05-22:
Thanks for this issue. This is only a preliminary response, I have some things to catch up on apparently.

As far as I understand this, you are in particular concerned about memory-mapped registers which have conditional access restrictions at runtime (such as the example you mentioned, where the state of one register determines whether another one can be accessed). For registers which are _always_ either readable, writable, or both, the _types themselves_ as currently present in `tock-registers` seem sound to me. Citing the [`voladdress` doc](https://docs.rs/voladdress/0.4.1/voladdress/struct.VolAddress.html#safety): "This type’s safety follows the <<unsafe creation, then safe use>> strategy." By creating a `ReadOnly`, `WriteOnly` or `ReadWrite` over some memory (MMIO) location (which will involve `unsafe` at some point, e.g. using `core::ptr::transmute`), you as a user are guaranteeing that this memory will allow accesses as specified by the type. Hence I conclude that when using these types exclusively over registers which don't have conditional access restrictions, the library should be usable without invoking unsound (undefined) behavior.

Now, saying "we don't support this use case" would of course be an unsatisfactory response. These seem to be valid concerns, and I can imagine we might have issues like these present in some chip supported by Tock today. However, I do fear that simply making `get` and `set` `unsafe` unconditionally (which really would mean we'd have to mark every trait method accessing the register as `unsafe`) will lead to much more opportunities to introduce actual unsound behavior, without much of the usual respect of the presence of `unsafe`. Furthermore, it provides many more opportunities for unsafe code to be invoked accidentally / "hidden" (think `unsafe { self.regs.bdrate.set(some_very_dangerous_fn()) }`). Last, but not least, for some parts of the codebase Tock uses restrictions on the use of `unsafe` as an isolation mechanism (see [threat model](https://github.com/tock/tock/blob/master/doc/threat_model/Capsule_Isolation.md#isolation-mechanism)). Making the methods on the access traits (`Readable`, `Writeable`, ...) unconditionally `unsafe` would prevent the use of entirely safe constructs such as `InMemoryRegister`s.

I'll read up on what exactly Rust considers unsound / UB when in comes to volatile accesses to MMIO. In the past I found that the definitions are a bit fuzzy here, as representing registers as memory means that ultimately, the register's semantics must be representable as a memory-bus access. Part of what makes this issue difficult is that presumably this is all hardware-dependent (hence _undefined_ behavior), so in the end we might not get around an `unsafe` interface. I'd probably vouch for a second, unsafe set of traits (e.g. `UnsafeReadable`) then. Do you have any further sources as to what Rust's / LLVM's understanding of this is?

Cryptjar, 2021-05-22:
> As far as I understand this, you are in particular concerned about memory-mapped registers which have conditional access restrictions at runtime

Yes, that is correct.

> Hence I conclude that when using these types exclusively over registers which don't have conditional access restrictions, the library should be usable without invoking unsound (undefined) behavior.

I would agree with that.

> Making the methods on the access traits (Readable, Writeable, ...) unconditionally unsafe would prevent the use of entirely safe constructs such as `InMemoryRegister`s.

Yes, this is precisely why I mentioned that there are various option to address this issue, and why I explicitly called my proposed solution the 'the lazy solution', i.e. lazily to be implemented within `tock-registers`, but with probably the most intrusive effect on the users. In fact, I personally would prefer a solution that allows to specify which registers are (unconditionally) safe to read and/or write, only requiring `unsafe` where actually required. This why I specifically brought up the example of the `voladdress` crate.

To that extent, I don't want to imply that I would know (as of now) which solution has the best design. I do consider this an open question, and I am practically asking for suggestions.

The idea of having two types of traits sounds interesting, but I don't grasp the full consequences that would imply. So, I wonder whether for instance a single trait that is generic similar to the `voladdress` approach wouldn't be easier to maintain.

Cryptjar, 2021-05-22:
@lschuermann 
> Do you have any further sources as to what Rust's / LLVM's understanding of this is?

Well, no, except maybe generally point to the [Rustonomicon](https://doc.rust-lang.org/nomicon/). I mean, in this context, I did not consider the validity of e.g. `std::ptr::read_volatile` upon MMIO addresses. To the best of my knowledge, this is perfectly fine.

I'm more concerned with the more general concept that Safe Rust must be always sound. Or formally called the ['soundness property'](https://doc.rust-lang.org/nomicon/safe-unsafe-meaning.html) which states:

> No matter what, Safe Rust can't cause Undefined Behavior.

And it goes without saying, `unsafe` Rust must make sure that this holds, otherwise it is unsound itself. Following this, I presented in my example how a struct that contains all these registers in combination with all the `set` & `get` functions being _safe_ then allows Safe Rust to violate the above rule, implying: something is wrong.

In fact, given the philosophy of "unsafe creation, then safe use" implies clearly, in my example, the unsound part was the creation of such a combined struct, as I implied before. So, I'm **not** trying to say that `tock-registers` were fundamentally unsound. Instead, I rather believe that given the existence of such complicated conditional hardware, that the current design of `tock-registers`, how should I say, might mislead users into writing unsound code. I'm trying to state that as something that is not serious, but something that should be improved.

lschuermann, 2021-05-22:
Right, I understand where you're going towards. I was not trying to imply that you were saying tock-registers was fundamentally unsound, rather I wanted to make sure we are on the same page about this.

Fundamentally, I agree that this seems bad; in particular that we're likely at least lacking explicit documentation that for these kinds of hardware, the creation of the registers types (unsafe) can lead to unsoundness. I'll add that to the documentation I'm writing as part of #2579.

I'd also like to see us have a solution for this. How exactly that will look like I can't say - it's not only up to me, and in any case I'd like to survey Rust documentation & the ecosystem first. I have raised my concerns about making `get`/`set` unsafe because I'm confident that would have a range of issues, not because I have a better solution in mind. I appreciate your input and the proposed ways to solve this, hope this clears up any misunderstandings :smile:.

Cryptjar, 2021-05-22:
However, @lschuermann having brought up the point about the [Capsule Isolation](https://github.com/tock/tock/blob/master/doc/threat_model/Capsule_Isolation.md#isolation-mechanism) of Tock, (and still I don't know anything beyond this file), starts to worry me. I mean, using the unsafe/safe split to create an isolation mechanism sounds very exciting to me, but then clearly you want IMO to give the benefit of the doubt, and rather declare every `get` & `set` as `unsafe` with only a few cautious safe-wappers to be forwarded to such capsules, than having just a single function that is accidentally missing its `unsafe`. Because, it seems to me, that such a single function would break the isolation, wouldn't?

When I think about it, there are also some register such as an MMU translation table address register (and I don't know if those exist in the context of Tock) seem to me inherently unsafe to write to, i.e. I can't think of any constructor that would ensure that writing arbitrary addresses to such a register would be always sound. This sort of things makes me wonder whether this problem is actually more serious that I thought. Or, maybe I'm overthinking this.

jrvanwhy, 2021-05-22:
> If the enable bits are clear[ed] you will have _**no access**_ to a peripheral. You can not even read or write the registers!

This states very little about what happens if software tries to read or write a register when the peripheral is disabled. I looked through the document, and I don't see anything that explains what happens. I would expect it to fire some sort of exception interrupt (i.e. a hard fault). If that is the case, then there is no UB, and using `tock_registers` is still sound.

If it's truly undocumented, then you're right that using `tock_registers` for that use case is unsound.

> Do you have any further sources as to what Rust's / LLVM's understanding of this is?

Because Rust lacks a specification (the Rust Reference is a work in progress and is not currently correct), there is no single source. The Rust Reference's [list of undefined behavior](https://doc.rust-lang.org/reference/behavior-considered-undefined.html) is a decent high-level overview, but very insufficient to cover details. I recommend poking around the [Unsafe Code Guidelines WG](https://github.com/rust-lang/unsafe-code-guidelines)'s repository, including their issues.

For example, nothing in the Rust Reference or the standard library docs say that having a `&UnsafeCell<u32>` that points to a MMIO'd register causes UB, but it currently does: https://github.com/rust-lang/unsafe-code-guidelines/issues/33.

Cryptjar, 2021-05-22:
@jrvanwhy: well can agree on that the Rust documentation does not, by any mean, fully cover embedded unsafe programming. In fact, most of the documentation IMO is rather application centric than system (i.e. OS) centric. So, I admit that I resort to some sort of creative thinking. Tho, I want to point-out, that the [list of undefined behavior](https://doc.rust-lang.org/reference/behavior-considered-undefined.html) dose state: "The following list is **not** exhaustive". Therefore, I'd rather taken a conservative approach, by which I mean, I rather try to prove why I think something is _sound_ instead of trying to show that it is _unsound_. So, for the benefit of the doubt, if I can't confidentially prove it is _sound_, I just **assume** that it is _unsound_.

Having said this, when some documentation literally tells me **not** to do something, I _assume_ it must be unsound. But for the sake of argument, take the division-by-zero, when I'm speaking as a system development with C-mindset, would argue, that (on most platform) this is a well-defined operation, i.e. it definitely (as it is well documented) traps to an exception handler. However, when arguing from the point of view of a Rust application developer, trapping into the OS, is sort of unintended (we can call it a programming error), I mean semantically speaking the result of the mathematical expression `1÷0` is something that [arguable is _undefined_](https://en.wikipedia.org/wiki/Division_by_zero), so I would argue it _is_ UB in the context of Rust. And in fact, the `unsafe` [`unchecked_div` intrinsic](https://doc.rust-lang.org/nightly/std/intrinsics/fn.unchecked_div.html) dose clearly define this as UB: "[…] resulting in **undefined behavior** where y == 0 […]" (where `y` refers to the divisor)

jrvanwhy, 2021-05-22:
> But for the sake of argument, take the division-by-zero, when I'm speaking as a system developer with C-mindset, I would argue, that (on most platform) this is a well-defined operation, i.e. it definitely (as it is well documented) traps to an exception handler.

Division by zero in C causes undefined behavior. From [N1570](http://chimera.roma1.infn.it/SP/COMMON/iso-iec-9899-1990.pdf) (roughly C11):

> The result of the / operator is the quotient from the division of the first operand by the second; the result of the % operator is the remainder. In both operations, if the value of the second operand is zero, the behavior is undefined.

Trapping to an exception handler is a hardware behavior, and as such it does not exist in the C abstract machine.

I spent some more time looking through the BCM2711 peripherals datasheet you linked. The "Peripheral access precautions for correct memory ordering" section says the following:

> Normally a processor assumes that if it executes two read operations the data will arrive in order. So a read from location X followed by a read from location Y should return the data of location X first, followed by the data of location Y. Data arriving out of order can have disastrous consequences. For example:
> ```
> a_status = *pointer_to_peripheral_a;
> b_status = *pointer_to_peripheral_b;
> ```
> Without precautions the values ending up in the variables `a_status` and `b_status` can be swapped around.

In my opinion, this is far more problematic for `tock_registers` than concerns about accessing disabled peripherals. Maybe `tock_registers` just isn't suited to the BCM2711?

Cryptjar, 2021-05-23:
@jrvanwhy you are of course right, I appreciate your effort to contradict me. It seems, my little example with division-by-zero with hindsight was badly state, I mean, I don't fully get what I was trying to say. Of course, it is undefined whenever one is interested in the quotient where the divisor happens to be zero, this extends to every language including C. I guess, I was rather tying to say, that if one tries to write an OS (a kernel for that matter), one has considered the real machine which includes exception handling (especially if coming from a user application, which does have to be written in C), and thus as a kernel developer you can't really stay confined to the C abstract machine, can you? (tho I have to admit I never read any formal C standard, so I'm practically guessing here).

I hope this clears a bit up what I was trying to say, I didn't want to start confusion.

---

Go forward:

> The "Peripheral access precautions for correct memory ordering" section says the following:
>
>
> In my opinion, this is far more problematic for `tock_registers` than concerns about accessing disabled [...]

While this is right, it is not as bad as it might sound. I do not expect you to read that entire manual, so just let me add two more citations from that section:

> Accesses  to  the  same  peripheral  will  always  arrive  and  return  in-order.

and

> It  is  not  required  to  put  a  memory  barrier  instruction  after  each  read  or  write  access.  Only  at  those  places  in  the  codewhere  it  is  possible  that  a  peripheral  read  or  write  may  be  followed  by  a  read  or  write  of  a  different  peripheral.  This  isnormally at the entry and exit points of the peripheral service code

I think that makes it much easier to grasp, tho I have to admit that I did not find a precise definition of what constitutes as a 'peripheral', so I'm practically _assuming_ that whatever register list Broadcom puts into a single table counts as a single peripheral, e.g. I actually assume that the 'three Auxiliary peripherals' are, with regard to memory ordering, considered a single peripheral i.e. that all accesses to the addresses that the document defines as `0x7e215000` through `0x7e215100` (section 2.1.) are in this respect a single peripheral.

So, given that assumption, I did not feel it is important when only considering, say, a single UART as in OP.

> Maybe `tock_registers` just isn't suited to the BCM2711?

So, having things made (hopefully) clearer, I think it is perfectly suitable. For instance, in the current version (that is the one on crates.io), I would simply say, the constructor of the register struct has the additional (platform specific) contract, that it must ensure that no references to any register struct of any other peripheral coexists with the one to be created, and of course it has to issue the appropriate memory barriers (including one in the destructor). That given, Safe Rust, could ever only interact (arbitrarily) with a single peripheral, where this memory ordering is **not** an issue.

Having said that, I seriously believe that, this platform in particular, would benefit from the addition of `unsafe` `get` & `set` methods, because then the constructor would not necessarily have to prevent the coexistence of such peripheral register structs, instead it would be the contract of the individual `get` & `set` methods that appropriate memory barriers have been/will be issued. Which in my opinion makes things more practical. If one is actually writing some sort of safe-wrapper then each method would simply start & end with the appropriate memory barriers and within that function, of course, only a single peripheral is accessed.

jrvanwhy, 2021-05-23:
> So, having made things (hopefully) clearer, I think `tock_registers` is perfectly suitable here. For instance, in the current version (that is the one on crates.io), I would simply say, the constructor of the register struct has the additional (platform specific) contract, that it must ensure that no references to any register struct of any other peripheral coexists with the one to be created, and of course it has to issue the appropriate memory barriers (including one in the destructor). That given, Safe Rust, could ever only interact (arbitrarily) with a single peripheral, where this memory ordering is **not** an issue.

Actually, I think there's a workable solution without making the `get` and `set` methods `unsafe`. Don't include `AUX_ENABLES` in your `tock_registers`-based register declarations. Then, in the function that constructs the register struct reference, you need to do two things:

1. Have a memory barrier (as you suggested)
2. Ensure the peripheral that is being switched to is enabled.

Then use of the returned struct can be used safely.

Am I missing something?

Cryptjar, 2021-05-23:
> Don't include `AUX_ENABLES` in your `tock_registers`-based register declarations.
> […]
> Then use of the returned struct can be used safely.

I agree, in fact, that is pretty much what I wrote as, so to speak, _first solution_ in OP (first paragraph). I think, that this is a perfectly valid (i.e., _sound_) approach to do it. However, as I also already pointed out, I think this kind of library design is not very convenient for the user in this particular use-case. Furthermore, I imagine, that this sort of design might even tease someone in accidentally writing something that is unsound, because they happen to put them together anyway (similar to how it is 'placed' in the documentation).

And this is my reason, why I propose to change `tock_registers` to, in my opinion, better accommodate this particular use-case.

> Am I missing something?

I just want to add, for clarification, that still, the constructs has to make sure that no other peripheral (if they are also somehow wrapped via `tock_registers`) dose coexist at run time with that returned struct (i.e. limiting life-times).

lschuermann, 2021-05-23:
> why I propose to change `tock_registers` to, in my opinion, better accommodate this particular use-case.

Having thought about this for some time, I'm not sure whether making `get`/`set` `unsafe` is beneficial here. Assuming the proposition is valid and accessing a disabled register is in fact undefined behavior (which it is not for every platform!), simply marking register accesses as `unsafe` doesn't really tell us anything about when it is safe to access some register (i.e. the contract with that register). I agree with @jrvanwhy that it might be a better solution to just limit the lifetimes of the respective register wrappers, such that they are only in scope (with their respective access levels) when it is safe to access the register's contents. This would, conceptually, extend the "unsafe creation, safe use" paradigm to encompass these restrictions.

To further illustrate: since #2517 `tock-registers` encompasses a traits-based interface for accessing registers (_Readable_, _Writable_, etc.). Part of the reason this change was introduced was to be able to be agnostic over the precise register type (such as `ReadWrite`, `ReadOnly`, _`YourCustomType`_), and still use it through a standard interface. We are able to use generics / trait objects for accepting registers of unknown types. If methods on these traits were marked unsafe, the interface would convey that sometimes it is possible to access these registers, whereas sometimes it is not. A helper / struct taking these registers would not know any of the specifics of your hardware, as to when exactly one can access these. It appears to make more sense to match the lifetimes of the respective registers to the scope in which these registers are safe to access, using - for example - a syntax as in the following contrived example:

```rust
self.access_uart(|uart_regs| uart_regs.baudrate.set(9600))?;
```

(Which would, prior to calling the closure, ensure that the register struct reference passed in is in fact safe to access.)

lschuermann, 2021-05-23:
Nonetheless, as stated above, I am in favor of making the precise assumptions of the existing register structs (`ReadWrite`, `ReadOnly`, `WriteOnly`) be explicit in the documentation, specifically that the memory wrapped through these structs must be valid to access as restricted by the types for the entire lifetime of said structs.

Cryptjar, 2021-05-24:
I'm getting the impression that we're having a discussion about when exactly something is sound here. So, to illustrate things, I'd like to give a further example based on the [Nordic nRF52840 chip](https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.2.pdf) that I happen to know superficially and that is supported by Tock, so you might know it too. In its section 2.3.2 "Permissions", it defines a read-only register as "[a] Field [that] can only be read. A write will be ignored."

Following this definition, I make two observations:

1) It does not semantically make sense to write to a read-only register, because it does not have any intentional effect (i.e. non at all)
2) The hardware is perfectly fine with a write to a read-only register and there is nothing _bad_ happening (i.e. nothing at all).

Based on semantics (1), I would personally conclude that writing to a read-only register is _unsound_.

I do wonder, whether you (@lschuermann and @jrvanwhy) would agree with me on that example.

bradjc, 2021-05-24:
> However, I do fear that simply making `get` and `set` `unsafe` unconditionally (which really would mean we'd have to mark every trait method accessing the register as `unsafe`) will lead to much [...]

I agree with this point. Register accesses are used so frequently that using `unsafe` would become a somewhat meaningless flag, and the incentive to mark entire functions `unsafe` would be high. And since virtually all uses would meet the safety requirements, there would be very little incentive to document why each use of unsafe for registers is actually safe.

jrvanwhy, 2021-05-24:
> I do wonder whether you (@lschuermann and @jrvanwhy) would agree with me on that example?

In Rust, an API is _unsound_ if it allows safe Rust code (i.e. code that does not use `unsafe`) to invoke UB, and sound otherwise.

If you use `core::ptr::write_volatile` to write to a nRF52840 read-only register, and never create references (`&` or `&mut`) to the register, then that write will not result in UB. Therefore, you can build a sound API that sends a write to a nRF52840 read-only register.

Cryptjar, 2021-05-27:
Sorry, for the late answer, but I see that we have a certain disagreement here. I don't feel I have anything to add. So, I actually moved on and consider writing my own MMIO register crate. I still consider my points given here to be valid, but if you shall come to the conclusion that it wouldn't be worth it, I don't mind if you close this issue. After all, it's your decision to make.

lschuermann, 2021-05-27:
I do want to emphasize that we very much appreciate these inputs and even if it seems like there is some hesitance with respect to your proposal, it's important to have discussions about these kinds of issues. Especially when talking about an interface which must work across vastly different platforms and be used in different software, there for sure are different goals, each having their own set of compromises. I think it's important we've established neither of these approaches is wrong -- which one `tock-registers` should go with is a different question.

Cryptjar, 2021-05-28:
BTW: My idea goes along the lines of something like this:

```rust
/// Safety: A value of `Condition` must only exist when it is sound to access
/// this register, i.e. it reperesents an access token.
pub unsafe trait ReadableWithCondition<Condition> {
    type T: IntLike;
    /// Read this register
    fn get(&self, _cond: &mut Condition) -> Self::T {
        // Example content: assuming this is a MMIO reg at `self.ptr()`:
        unsafe {
            // This is sound, because this trait must only be implemented,
            // if the existence of a `Condition` value guarantees that it is sound
            // to access it, and we have here an exclusive `_cond`.
            core::ptr::read_volatile(self.ptr())
        }
    }
    // ... other items
}
```

A type which implements `ReadableWithCondition<()>` than represents something that `tock-registers` currently represents with `Readable` (i.e. always sound to access). Important tho, is that this `Condition` must be specifiable differently for each register, whereas, I still like the concept to declare a register-block as a whole.

lschuermann, 2021-05-28:
I've put together a small example of a possible API [here](https://gist.github.com/lschuermann/ae78cf23684b86f2c5b9b2e7eb8f55eb#file-conditional_tock_regs-rs-L136). Please have a look, maybe it satisfies your use cases.

Most importantly, this does not change the `tock-registers` APIs (`Readable`, `Writable`, `ReadWriteable`) in any way. It does so by introducing three helper structs, which are bound to your capability's lifetime, and implement the respective `tock-registers` traits. Even better, you can implement these types outside of this crate and still be compatible with the macros and other interfaces! Unfortunately this isn't on crates.io just yet.

I know, the code looks a bit confusing (and isn't particularly well documented), but conceptually it's pretty simple: you have a `CondReg` and call either `.readable`, `.writeable`, or `.read_writeable`, pass in your capabilities and get back a struct which supports the regular `tock-registers` API (e.g. `get`, `modify`). It takes care that your capabilities must be mutably borrowed for as long as you hold your intermediate type.

Cryptjar, 2021-05-28:
> That example would've been great from the start of this discussion

I very much agree, but unfortunately, for a matter of fact, this idea had cross my mind just recently.

> Whether we'd want to have a `tock-registers` API that uses capabilities is a different question than making the interface methods `unsafe`.

Well, I don't want to roll up this entire discussion, but in my personal opinion, if there is some contract that must be upheld (for whatever reason), one can either express it with the type system or with `unsafe` and come-out and say it in the docs. And, since I had **not** any good idea when I started this thread, I went with the "lazy" idea of going with `unsafe`. I never said it would be particularly good, it was just one solved of many (as I said).

So, in my personal opinion, it is rather whether `tock-registers` wants to support that particular use-case, where either option is helpful. Obviously, some of these options comes with drawbacks for other use-cases.

And, it seems to me that you kinda seem to agree (at least up to some point) with that, because you made in your example the "base functions" (i.e. `get` & `set`) of `CondReg` indeed `unsafe`. And around them then is the fancy type system to provide for a safer alternative.

> I've put together a small example of a possible API here. Please have a look, maybe it satisfies your use cases.

It does look interesting, I think it goes definitively into the right direction and I appreciate this.

> Even better, you can implement these types outside of this crate and still be compatible with the macros and other interfaces!

Is it? But, even if it were, doesn't that somehow defeat the purpose of a library? I mean, it sounds a bit like, what I'm doing currently anyway: write my one types and stuff to suit my use-case.

And doing so, I feel my use-case might be even better suited by my more customary solution. For instance, I'm also experimenting with const-generics for the register address instead of using `UnsafeCell`. Another thing is, I'd like that fields that are declared as represented by an enum to be _only_ accessible via that enum and not an integer. So, some more minor things that most boil-down to personal preference, and I didn't feel as if I could make a point for either of them here, anyway.

lschuermann, 2021-05-28:
> Is it? But, even if it were, doesn't that somehow defeat the purpose of a library? I mean, it sounds a bit like, what I'm doing currently anyway: write my one types and stuff to suit my use-case.

Actually, the purpose of #2517 was (at least in part) to allow exactly this: implement custom register types. By making the APIs traits, and adding some default implementations and clever impl-logic for `ReadWriteable`, as long as your type implements `Readable::get` and/or `Writeable::set`, you can use it with the `tock-registers` API and the `register_bitfields!` macro etc. We even use that internally, for RISC-V CSR registers:

https://github.com/tock/tock/blob/60a9eb15baeb40e521e9d0fda8e89b0ce71bfffb/libraries/riscv-csr/src/csr.rs#L269-L284

If types are sufficiently generic and useful we can incorporate them into `tock-registers`, but even for registers in your own codebase, you can opt for `tock-registers` to do the heavy lifting with the API & macros on your very own types (as long as they're roughly compatible). By having a standard traits API, you can also be agnostic over the actual types and instead take just a `Readable<T=u32>`, for instance.

The fact that I was able to implement this proof of concept using public interfaces seems to confirm that this is valuable.

> And doing so, I feel my use-case might be even better suited by my more customary solution. For instance, I'm also experimenting with const-generics for the register address instead of using [...]

Sure thing. All those ideas sound interesting, but questionable as to whether they could make it into `tock-registers` in the near future. If you do build your own crate, be sure to give us a pointer.

lschuermann, 2021-05-29:
I think based on your previous comments we can close this issue. Feel free to reopen or comment. Again, thanks for bringing this up.

closed 2021-05-29 by lschuermann