tock/tock #2594: `Process` trait implementation is not optimised out

alistair23, 2021-05-24:
Although I can't find documentation stating this, it seems that any implementation of `pub` traits will not be optimsied out in the final binary.

A good example of this is the `print_full_process()` function. It's an implementation of the `Process` trait. I have made the following changes to a RISC-V and an ARM board:

```diff
diff --git a/boards/redboard_artemis_nano/src/io.rs b/boards/redboard_artemis_nano/src/io.rs
index a31982232..821d2992a 100644
--- a/boards/redboard_artemis_nano/src/io.rs
+++ b/boards/redboard_artemis_nano/src/io.rs
@@ -42,12 +42,5 @@ pub unsafe extern "C" fn panic_fmt(info: &PanicInfo) -> ! {
     let led = &mut led::LedLow::new(led_pin);
     let writer = &mut WRITER;
 
-    debug::panic(
-        &mut [led],
-        writer,
-        info,
-        &cortexm4::support::nop,
-        &PROCESSES,
-        &CHIP,
-    )
+    loop {}
 }
diff --git a/boards/swervolf/src/io.rs b/boards/swervolf/src/io.rs
index 0ccf74c16..d01fdaa03 100644
--- a/boards/swervolf/src/io.rs
+++ b/boards/swervolf/src/io.rs
@@ -37,8 +37,6 @@ impl IoWrite for Writer {
 pub unsafe extern "C" fn panic_fmt(pi: &PanicInfo) -> ! {
     let writer = &mut WRITER;
 
-    debug::panic_print(writer, pi, &rv32i::support::nop, &PROCESSES, &CHIP);
-
     // By writing to address 0x80001009 we can exit the simulation.
     // So instead of blinking in a loop let's exit the simulation.
     write_volatile(0x8000_1009 as *mut u8, 20);
```

This ensures that the `print_full_process()` is never called. This can be verified by inspecting the objdump and confirming that `print_full_process()` is never called, yet the entire function is included.

Depending on the board `print_full_process()` is somewhere between the 3rd and 15th largest function. There are other `Process` debug functions like `print_full_process()` that also have the same problem and are also very large.

A quick look at the SweRV binary shows that just these two unused print functions take up about 7% of the binary size. The string formatting is also a large part.

This has come up in the past, see https://github.com/tock/tock/pull/1830 and https://github.com/tock/tock/issues/1995

I had a quick look at using a separate debug trait, but that quickly becomes messy. Would a process configuration option be ok to use here?

hudson-ayers, 2021-05-24:
> Although I can't find documentation stating this, it seems that any implementation of pub traits will not be optimsied out in the final binary.

I believe this is only the case if an instance of an implementation of that trait is constructed as a trait object.

This is really just another demonstration of how high the cost of using trait objects throughout the Tock kernel is. This particular overhead could be resolved if Rust supported LLVM whole program devirtualization in the way that Clang does. There is a tracking issue for this support (https://github.com/rust-lang/rust/issues/68262) but seems to be little progress as of now. In fact, I think that LLVM whole program devirtualization could make a huge difference in the Tock kernel given how many traits are only implemented once, but used as trait objects anyway.

The specific feature that would fix this is "dead virtual function elimination", for which support was added to LLVM in 2019 (https://reviews.llvm.org/D63932).

bradjc, 2021-05-25:
Just for context, are any boards running against code size limits right now? Is this something we need to address in the short term?

hudson-ayers, 2021-05-25:
We have had to remove userspace capsules from Imix once already, though I think that was probably a good decision regardless of the fact that code size limits inspired it.

bradjc, 2021-10-15:
As of #2759 there is now a kernel config option to remove the debugging functions which are not optimized out.

closed 2021-10-15 by bradjc

alistair23, 2022-06-15:
`rustc` should also be better at this now: https://github.com/rust-lang/rust/pull/96285