tock/tock #2603: Tock syscall fuzzing - application and kernel side crash

Menzo96, 2021-05-29:
I am working on a fuzzing project on Tock OS and I have encountered some crashes and I couldn't find the cause yet.

I have created a simple fuzzer with libtock-c that tries a few Tock OS system calls. 
The issue encountered manifests in different ways for different Tock OS versions:

- For v 2.0 as a crash on the application side on a cmp instruction.
- For v 1.6 as a crash on the application side, however upon further investigation it seemed to crash in kernel space on a bx lr instruction.

I have detailed the issue on a libtock-c fork here: [https://github.com/Menzo96/libtock-c/tree/dumb_fuzzer/examples/dumb_fuzzer](url)

Do you have any idea why this happens? Do you have any advice on how to investigate this further? Is there something that I'm doing wrong (syscall interface, using tock, libtock-c)?

Thank you

alistair23, 2021-05-30:
A good place to start is to turn on the debug logging, you can find it here: https://github.com/tock/tock/blob/master/kernel/src/config.rs#L47

I don't know enough about ARM to get a good sense of the type of fault (if you can reproduce this on RISC-V I could help) but based on your `Faulting Memory Address` there is some NULL pointer access.

It seems like you have figured out the faulting line. Can you figure out where the faulting NULL pointer comes from? Is it being popped off the stack or does it come from somewhere else?

ppannuto, 2021-05-31:
[image]

If I'm reading this correctly, it looks like the `bx lr` at `0x000083e6` is the last instruction that runs before the panic? To me that points to some corruption in the application stack. Right before that, we loaded up the link register with the magic value `0xffff_fff9`. When that "address" is loaded into the PC, it triggers the hardware to do a context switch. That causes several things to happen, but most importantly, a bunch of registers, including the new PC for the application, is read from the application stack pointer. This suggests that -- somehow -- the application stack is getting corrupted before the return from the kernel.

hudson-ayers, 2021-05-31:
Thanks for the detailed writeup! I think one thing that is missing is a minimal reproducible example. The apps you shared are helpful, but they issue a bunch of system calls, allocate a bunch of memory, etc. It would be useful to narrow down the source of the failure rather than just relying on the panic dump (which can become unrealiable in the case of memory getting overwritten with incorrect values in advance of the failure). It's possible that a single system call alone leads to this failure -- that would be really helpful to know. Have you tried cutting out most of the code in your example app to find the minimal set of system calls that have to be issued to trigger this failure? 

It would be nice to have an app that just sequentially calls a small number of system calls with deterministic values to produce this bug. Even better if that app does not allocate memory or call rand() -- though of course it is possible those things are necessary to reproduce the failure. Even that information would be useful.

Menzo96, 2021-06-04:
**Thank you all for the responses.**

> A good place to start is to turn on the debug logging, you can find it here: https://github.com/tock/tock/blob/master/kernel/src/config.rs#L47
>
> I don't know enough about ARM to get a good sense of [...]

**I will enable debug logging come back with an update.**

> [image]
>
> If I'm reading this correctly, it looks like the `bx lr` at `0x000083e6` is the last instruction that runs before the panic? To me that points to some corruption in the application stack. [...]

Yes, the `bx lr` at `0x000083e6` is the last instruction. I could step through the mov instructions (with `si`) and when stepping to the `bx lr` instruction the panic starts. I will try and inspect the application stack.

> Thanks for the detailed writeup! I think one thing that is missing is a minimal reproducible example. The apps you shared are helpful, but they issue a bunch of system calls, allocate a bunch of [...]

**My assumption is that the crash si due to the excessive use of printf. I will test and provide a minimal application code that replicates this issue.**

Menzo96, 2021-07-31:
Hello,

I have enabled the debug logging in the kernel and cleared the code to the minimal reproducible examples. I have tested it on a few scenarios. The bug still appears during the svc call and it seems to be caused by printfs. The findings together with the debug console logs, disassembly files and C programs are in my fork: https://github.com/Menzo96/libtock-c/tree/dumb_fuzzer/examples/dumb_fuzzer

Do you have any idea on how to continue debugging? Or what causes it?

Thank you and sorry for the delay.

phil-levis, 2021-07-31:
Something doesn't add up here.

`cmp` doesn't access memory. So how can it cause an access fault? The faulting address is 0x0, but 0x0 is only in R6 and R11. The HFSR and CFSR look right, and since it's a memory access error, the fault address should point at the problem. This doesn't look like an unstacking error (caused, for example, when restoring the saved context on the stack has a memory access fault because the pointer to the stack is off), because that is a separate error case.

So for me, the key thing is that the processor state output doesn't make sense with the fault. Perhaps the disassembly is off and it's actually a different instruction causing the fault?

Could you please put your outputs and errors in this discussion? I worry that keeping this discussion synced with a changing branch README will be confusing.

Menzo96, 2021-08-01:
Thank you.
I have moved here the files.

From my previous analysis, it seems that the crash happens during the **svc call**, on the kernel side in the **svc_handler()** when executing the last **bx lr** instruction to switch back to the application side.

phil-levis, 2021-08-01:
> 
> 
> Thank you.
> I have moved here the files.
> 
> From my previous analysis, it seems that the crash happens during the **svc call**, on the kernel side in the **svc_handler()** when executing the last **bx lr** instruction to switch back to the application side.

Just trying to write down my thinking... hope this is helpful.

How does this happen? One thing the `bx lr` in a context switch will do is unstack the registers/restore the context. But if there is an MPU access error doing so you'd expect this to result in a context unstacking error, not a general data access violation error. See page 4 of:

https://www.keil.com/appnotes/files/apnt209.pdf

I'd also expect the error to be triggered in supervisor mode, not user mode. 

But these expectations are from my experiences on imix: it could be that the microbit is triggering a different exception?

Why is the last syscall entry in the process dump Memop instead of Allow? The register values in the userspace process look like a return type of FailureU32, with an error of FAIL and a value of `0x20005140`. But Memop #1 (SBRK) only returns an error of ENOMEM and Allow doesn't return a FailureU32.

The one time I've seen an error like this (on unstacking), it was due to the kernel portion of the grant region being corrupted. It was caused by my miscalculating offsets into the region, such that the ring buffer of Upcalls was overlapping with the process struct. The upcalls were therefore corrupting the process struct.

phil-levis, 2021-08-02:
@bradjc and I looked at this in a call today.

The key thing we saw in your output https://github.com/Menzo96/libtock-c/blob/dumb_fuzzer/examples/dumb_fuzzer/Tock_v2.0/output3/debug_console.txt is this:
```
[0] subscribe(0x0, 0, @0x0, 0x0) = SubscribeSuccess(0x40249, 0)
[0] memop(1, 0x6c) = Failure(NOMEM)
```

So it looks like sbrk() is failing and the program ignores this. This is the last system call made. The registers line up with an allow being made, but *not* with a return value being in there (see my comments above). 

The thing which just seems completely strange is there's an error on allow, but it isn't the last system call made. How is that possible? 

Our best guess is that the fact that sbrk()/malloc() has returned a failure isn't being checked, and that is corrupting memory somehow. This *looks* like a C bug, not a kernel bug. Brad will dig some more, hopefully.

bradjc, 2021-08-02:
The root issue is here: https://github.com/tock/libtock-c/blob/master/libtock/sys.c#L82

In the 2.0 changes the return value of `memop()` (the syscall) changed. sbrk expects a status code, and memop returns null. This means that a malloc fails but nobody is the wiser.

It isn't checked, but eventually one of these mallocs fail: https://github.com/Menzo96/libtock-c/blob/dumb_fuzzer/examples/dumb_fuzzer/main.c#L29. I'm guessing because only str is freed (not aux). https://github.com/Menzo96/libtock-c/blob/dumb_fuzzer/examples/dumb_fuzzer/main.c#L97

Menzo96, 2021-08-03:
Thank you both very much for your help.
I missed that aux, my bad. I'm sorry for this.
I will do some testing, but most probably that's the problem.

phil-levis, 2021-08-04:
Thanks for finding this bug in `malloc`!

Menzo96, 2021-08-09:
Glad I could help!

hudson-ayers, 2021-08-09:
Feel free to close this once you have confirmed the fix works!

closed 2021-09-21 by bradjc