tock/tock #2630: [question] Subscribe function pointer validation?

alexandruradovici, 2021-06-23:
While I was taking a closer look to the inner functioning of the system calls I noticed that there is no check for the function pointers provided to the subscribe system. 

My assumption is that of an application sends an invalid pointer (not in its memory or not in memory at all), the application will crash.

I was wondering if this is the desired behavior or should subscribe perform some kind of pointer validation?

jrvanwhy, 2021-06-23:
The kernel can check that a function pointer is in bounds for the process, but it cannot guarantee that pointer is *correct*. Such a check might be a useful diagnostic, but would fall far short of guaranteeing the callback pointer is incorrect.

The only case where I've heard of a subscribe pointer being incorrect is if the process binary is statically-linked at the incorrect address. However, the userspace runtime can check for an incorrect location and report that error. The Tock 2.0 `libtock-rs` does this, I don't know if `libtock-c` does.

Overall, I don't think the benefits are worth the cost, except possibly as part of some sort of "debug build" kernel (which Tock does not currently have).

alexandruradovici, 2021-06-23:
> The kernel can check that a function pointer is in bounds for the process, but it cannot guarantee that pointer is _correct_. Such a check might be a useful diagnostic, but would fall far short of guaranteeing the callback pointer is incorrect.
> 
That is why I assumed that the lack of any check is intentional, but I wanted to be sure.

> The only case where I've heard of a subscribe pointer being incorrect is if the process binary is statically-linked at the incorrect address. However, the userspace runtime can check for an [...]

Agree

lschuermann, 2021-06-23:
@alexandruradovici This has been extensively discussed here: https://github.com/tock/tock/issues/2270

alexandruradovici, 2021-06-23:
Thank you, I missed that.

lschuermann, 2021-06-23:
Re @jrvanwhy: 

> except possibly as part of some sort of "debug build" kernel (which Tock does not currently have).

I agree that this could be a debug check, but before someone goes to great lengths and adds a kernel config for that, it should be carefully considered whether we'd actually want this. My two ct on the issue are:
- having different checks in different builds of the kernel is always a bit tricky. In this case it might be fine because subscribe would just return an additional error in debug, but not in release builds. However, this requires more code paths and creates a significant difference between in-development and production behavior, which one can't even exploit: passing an invalid pointer in production can't be handled gracefully and will _not_ produce an error, and thus UB or a fault once the callback is invoked.
- most often, passing an invalid pointer will immediately fault the app as it is being scheduled. As far as I'm concerned with diagnostics, I think that gives a ton of useful information and let's developers identify the error quickly (the PC should be insightful).