tock/tock #2637: Violation of Rust's aliasing rules with DMA operations

lschuermann, 2021-06-29:
It's a common paradigm for chips in Tock to take some buffer which is written to using a DMA operation. Because this buffer must be handed back to the client when the operation completes, we need to hold on to a reference to it. This seems to be done almost exclusively through `TakeCell`s. For example, for the nRF52 family of chips:

https://github.com/tock/tock/blob/94172ecf293e6df598192111b87db9dbbcaad178/chips/nrf52/src/uart.rs#L162-L169

https://github.com/tock/tock/blob/94172ecf293e6df598192111b87db9dbbcaad178/chips/nrf52/src/uart.rs#L496-L510

However, by putting these buffers into a `TakeCell`, we are still holding on to a mutable Rust reference to the respective slice:

https://github.com/tock/tock/blob/94172ecf293e6df598192111b87db9dbbcaad178/libraries/tock-cells/src/take_cell.rs#L14-L16

If the DMA operation now mutates these buffers, we've violated Rust's assumption of there only being a single mutable reference to some object, and that object not being mutated otherwise. This, as far as I understand it, makes the kernel unsound.

---

The solution appears to be quite simple: we need to drop the slice reference prior to starting a DMA operation which might mutate the memory in question, and only store the pointer (as a `*mut u8`) and length of the respective buffer. Rust does not impose the strict aliasing requirements on pointers. Once the DMA operation is finished, it should be safe to reconstruct a proper Rust slice.

Because a `core::slice::from_raw_parts_mut` looks really scary, and furthermore is easy to mess up, I propose introducing a `DMABuffer` (naming WIP) type, which does exactly this: take a static mutable reference to a slice of bytes in the constructor, decompose it into its pointer and length, and allow to reconstruct a slice again consuming the type. This would of course need to be applied across the entire code base.

---

What do others think? Have I missed something or is this really a soundness issue?

jrvanwhy, 2021-06-29:
I agree this is a soundness issue, and that the fix is to store the slice as a raw pointer and length.

ppannuto, 2021-06-30:
+1, I agree this is an issue. I might bikeshed to `DMACell`, as the cell metaphor is well understood, and I think we can explain it as a cell whose contents is (1) empty, (2) owned by underlying hardware, (3) present and owned by software. Plus Tock has a history of custom `Cell` types.

lschuermann, 2021-06-30:
Thanks for the feedback! Will create a PR to propose a new type to handle this and subsequently apply it to all chip crates.

Agreed that `DMACell` seems more reasonable. Although I think we can really start the name nitpicking once the type's semantics and API are figured out 😬.

kupiakos, 2021-10-29:
FYI, `*const [u8]` is a very useful type to not have to store the pointer/length separately

lschuermann, 2021-11-01:
> FYI, `*const [u8]` is a very useful type to not have to store the pointer/length separately

Indeed it is. I have already started working on this locally some time ago and based my solution on that (I believe). I should have fixes for at least some platform up in a week or so.