tock/tock #2641: Quick RFC: Tock Registers: `IntLike` to `UIntLike`

ppannuto, 2021-06-30:
#2623 highlighted that really our register interface lets us turn register operations into operations on well-defined unsigned numbers. We called this trait magic `IntLike`, presumably as we were saying that the underlying data could be manipulated like numbers, but I think the implementation is more accurately `UIntLike` (indeed, all of the example types are `u8`, `u16`, ...).

Since we're going to have some churn with this upcoming release, and since it will hopefully be a 1.0 interface, I'd propose we rename `IntLike` to `UIntLike`, both to more accurately reflect what's happening, and leave the door open to doing something intelligent in a non-breaking manner with registers that hold signed integer -like values in the future.

This should be a quick `s/IntLike/UIntLike` if we opt to make the change.

Thoughts? @lschuermann @andre-richter @bradjc @brghena

lschuermann, 2021-06-30:
I think this is a good change. It just clarifies some things, and the other changes in the upcoming release are breaking existing code already, so I'm in favor of introducing this change now as well. Feel free to add a commit on to my PR, alternatively I can do that in ~4-5h when I'm back at my PC.

I'd be much more hesitant to introduce registers over signed integers or arbitrary types, but leaving this up for discussion and enabling to introduce this in a non-breaking manner is definitely a win.

hudson-ayers, 2021-06-30:
+1 to this change

commit 79b7ebe0 (2021-07-01): tock-registers: Rename `IntLike` to `UIntLike`

This better captures the semantics of this interfaces.

Closes #2641.

Depends on #2618.
commit 116b8be6 (2021-07-01): Merge #2642

2642: Tock reg UIntLike r=lschuermann a=ppannuto

### Pull Request Overview

tock-registers: Rename `IntLike` to `UIntLike`

This better captures the semantics of this interfaces.

Closes #2641.

Depends on #2618.

### Testing Strategy

This pull request was tested by compiling.

### TODO or Help Wanted

N/A.

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
PR #2642 (2021-07-01): Tock reg UIntLike
### Pull Request Overview

tock-registers: Rename `IntLike` to `UIntLike`

This better captures the semantics of this interfaces.

Closes #2641.

Depends on #2618.

### Testing Strategy

This pull request was tested by compiling.

### TODO or Help Wanted

N/A.

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2021-07-01 by bors[bot]