tock/tock #2649: Add "overview" to tock-registers doc

bradjc, 2021-07-02:
Since we distribute tock-registers on crates.io: https://crates.io/crates/tock-registers, it would be nice to have a gentler overview in the readme on what the crate is before digging into how to use the API.

lschuermann, 2021-07-02:
Agreed. I think that this, as well as a better introduction in the Rustdoc root module documentation is justified.

potto216, 2025-09-27:
What is left to do on this issue? Since it was created it looks like https://github.com/tock/tock/blob/master/libraries/tock-register-interface/README.md has improved its overview.

@lschuermann is the Rustdoc root module https://github.com/tock/tock/blob/master/doc/README.md or https://github.com/tock/tock/blob/master/libraries/README.md?