tock/tock #2676: virtual_alarm will never fire client if dt is max ticks

jettr, 2021-07-20:
Looking at the [alarm callback within virtual_alarm][0], you can see that if `dt` is `max_ticks`, then `now` will always be between `reference` and `reference + dt`. This means we will never fire the client callback for a `max_tick` `dt`. This problem also extends to ticks near `max_ticks` if the bottom half of virtual_alarm handler doesn't run immediately and gets a `now` time that is already past the original reference.I see a couple of potential solutions:

* When someone tries to schedule a new alarm for ~`max_ticks`, we reduce that by some factor to ensure that the bottom half of virtual alarm handler has time to fire and loop through clients before the `now` goes back into `[reference, reference + dt)`
  *   This is not very desirable since the amount of ticks for each system and timer should be tuned to their parameters
* Add logic to detect if we are setting a `dt` with the highest bit available set, then toggle that bit and set the alarm for half of the `max_tick` duration. When we fire the alarm, look for all extended `dt`s that need to be set again, and the wait the remaining time left from the `dt`.
  *  This is more robust and allows for waiting max_tick without having a race window to worry about
  *  This takes more bookeeping memory to know if the future `dt` we are waiting for is an extended `dt` that need to have its top bit flipped again

We are especially vulnerability to `max_tick` being used for `set_alarm` functions since the helper `ticks_from_<time>` function will saturate at `max_ticks` if you try to convert too large of a time.

[0]:https://github.com/tock/tock/blob/0c8c7127a0e820966a5336c79d961846765bd926/capsules/src/virtual_alarm.rs#L199

jettr, 2021-07-23:
I wrote a unit test example that shows the issue at https://github.com/jettr/tock/tree/virtual-alarm

The bug happens when there is any time delay from when the hardware timer fires and the bottom half of the virtual timer handler runs (which is expected on real hardware with 1MHz timers).

phil-levis, 2021-07-24:
> Looking at the [alarm callback within virtual_alarm](https://github.com/tock/tock/blob/0c8c7127a0e820966a5336c79d961846765bd926/capsules/src/virtual_alarm.rs#L199), you can see that if `dt` is `max_ticks`, then `now` will always be between `reference` and `reference + dt`. 

I don't think this is true. E.g. suppose I have `now = 100` and a `dt = max_ticks`. Then `start=100` and `end=99`. The inequalities on `within_range` are defined such:

> Returns whether the value is in the range of [`start, `end`)

So if the current time is 99, the alarm should fire. E.g.:

```
    fn within_range(self, start: Self, end: Self) -> bool {
        self.wrapping_sub(start).0 < end.wrapping_sub(start).0
    }
```

will return `false` if `self=99`, `start=100` and `end=99`.  `99.wrapping_sub(100) == 99.wrapping_sub(100)` so the method will return false.

That being said, **for any other value of self** it will not fire. This is definitely be a problem if alarms are bumped a little into the future to make sure they fire. It means that the alarm will fire if and only if the current counter is equal to end, which is too narrow a window.

> I see a couple of potential solutions:
>
>     * When someone tries to schedule a new alarm for ~`max_ticks`, we reduce that by some factor to ensure that the bottom half of virtual alarm handler has [...]

The most common approach I've seen to solve this problem is to simply not allow interval/dt values equal to the maximum number of ticks.

Since we're in the process of updating 101 anyways, another solution is to have the `Alarm` and `Timer` traits have `max_dt` and `max_interval` methods, which tell you the maximum value you can pass for these arguments. You can make them substantively shorter enough than `max_ticks` that you leave a sufficient window for alarms/timers to fire. Or we just say that these traits have a maximum interval/dt value of half of the max ticks and check it. This pushes your second solution to the caller.

jettr, 2021-07-26:
I heavily lean towards fixing this issue in the virtual alarm implementation versus restricting the max dt to half of the available resolution.

Reasons for my preference
  * The implementation of `Alarm` on hardware driver has no issue with dealing with max ticks dt
  * The fix within virtual alarm will likely cost only one more word per virtual alarm in flash
  * Limiting max dt would just be one more thing for users to have to keep track of and potential do incorrectly
    * The more we can make the core libraries just work the way users would expect, the better

lschuermann, 2021-07-26:
I strongly agree that we should avoid making this API more complex if that's a reasonable option. All complexity which is not covered by the type system will eventually lead to users making mistakes.

jettr, 2021-07-27:
> I strongly agree that we should avoid making this API more complex if that's a reasonable option. All complexity which is not covered by the type system will eventually lead to users making mistakes.

Agreed. See PR #2695 for an implementation that doesn't change the API

closed 2021-07-30 by jettr

lschuermann, 2021-07-30:
Thanks @jettr!