tock/tock #2684: TicKV: reading unknown value size

wright-joe, 2021-07-22:
Hi @alistair23, It appears to me that TicKV's get_key() method requires the caller to know the size of the value being retrieved. If the length of the value is returned, and a slight change to the CRC check, the buffer supplied could be larger than the value.

Was this the intention or am I missing something?

Thank you for the TicKV library, it has been quite handy!

alistair23, 2021-07-23:
Hey!

First I'm glad you are using it! It would be awesome if you could share what you are doing with it.

Second TicKV is still in a kind of beta, so changes/fixes/improvements are more then welcome. I plan on finalising it once https://github.com/tock/tock/pull/2177 is merged. If you have improvements please feel free to send a PR or raise an issue.

As for the size, we should currently handle a larger buffer (although you are right in that we don't return the size). Can you provide more details on why you think the CRC needs changes?

wright-joe, 2021-09-24:
Hey @alistair23, 

I have a pull request prepared, but I don't have access to create a new branch.

Basically, the code changes allow the use of a read buffer uncoupled to the size of the value being retrieved.

On line 592 (tickv.rs), the entire buf that was passed is added to the CRC, so this change also patches that functionality.

ppannuto, 2021-09-24:
> I have a pull request prepared, but I don't have access to create a new branch.

You don't need to make a branch on the main Tock repo to create a pull request. If you make your own fork, you can create a pull request from a branch you own. Here's some more info in the GitHub documentation: https://docs.github.com/en/github/collaborating-with-pull-requests/proposing-changes-to-your-work-with-pull-requests/creating-a-pull-request#changing-the-branch-range-and-destination-repository

closed 2023-05-30 by bradjc