tock/tock #2727: Potential issue with `unused_kernel_region_number` in rv32i/pmp.rs

jettr, 2021-08-04:
Looking at the [unused_kernel_region_number][0] function in rv32i/pmp.rs, it appears that we are trying to iterate through the available PMP regions backwards since we want the app regions to be before the kernel regions, but we end up mismatching the `region` from its actual array index/`number`.

It seems like the below snippet would be correct instead. If that isn't the case, what am I missing?

```
    fn unused_kernel_region_number(&self, locked_region_mask: u64) -> Option<usize> {
        for (number, region) in self.regions.iter().rev().enumerate() {
            if self.app_memory_region.contains(&number) {
                continue;
            }
            // This region exists, but is locked
            if locked_region_mask & (1 << number) > 0 {
                continue;
            }
            if region.is_none() {
                return Some(number);
            }
        }
        None
    }
```

Basically, remove the 
`let number = MAX_AVAILABLE_REGIONS_OVER_TWO - num - 1;` line and rename `num` to `number`

@alistair23 , can you show me what I am missing here?

[0]:https://github.com/tock/tock/blob/master/arch/rv32i/src/pmp.rs#L277

alistair23, 2021-08-04:
Yeah, I think you are right. Are you seeing a failure?

jettr, 2021-08-04:
No, I am not seeing an issue, but I was just reading through the code as I was changing things. It couldn't understand what is was trying to do.

jettr, 2021-08-31:
Should be move forward with a fix for this?

alistair23, 2021-09-02:
Yeah, I think so. It looks incorrect

jettr, 2021-09-02:
I figured out what is going on here. We should be calling `enumerate` before `rev` and the confusing line is correcting for calling `rev` before `enumerate`. I upload a "fix" that doesn't change behavior, but just documents what is happening more explicitly in #2806.

See also this [rust playground example](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=1b66e17de3c22e49cb855dde854cce09)

commit 4cc35cab (2021-09-03): Merge #2806

2806: rv32i/pmp: add clarifying comment to enumeration r=hudson-ayers a=jettr

### Pull Request Overview

It is subtle that calling enumerate() on a rev() iterator does not
produce values in a backwards order. However, we can call enumerate,
then rev to get index values that match the array indexes.

This also closes #2727 

See the [rust playground](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=1b66e17de3c22e49cb855dde854cce09) for a demonstration.

### Documentation Updated

- None

### Formatting

- [x] Ran `make prepush`.
PR #2806 (2021-09-03): rv32i/pmp: add clarifying comment to enumeration
### Pull Request Overview

It is subtle that calling enumerate() on a rev() iterator does not
produce values in a backwards order. However, we can call enumerate,
then rev to get index values that match the array indexes.

This also closes #2727 

See the [rust playground](https://play.rust-lang.org/?version=stable&mode=debug&edition=2018&gist=1b66e17de3c22e49cb855dde854cce09) for a demonstration.

### Documentation Updated

- None

### Formatting

- [x] Ran `make prepush`.
closed 2021-09-03 by bors[bot]