tock/tock #2734: Problem with running example program on Nexys Video board.

student180529, 2021-08-06:
I` m working on Open Titan on Nexys Video board. I have correctly built and uploaded the software for earlgrey-nexysvideo.

I00161 boot_rom.c:66] Boot ROM initialisation has completed, jump into flash!
OpenTitan initialisation complete. Entering main loop

I built examples from libtock-c with a command: make RISCV=1. The problem arises when I try to run a example program  e.g. c_hello. I link the Tock's elf file with the rv32imc.0x20030080.0x10005000.tbf file like this:
riscv32-unknown-elf-objcopy --update-section .apps=/home/ptumiali/projects/test/rv32imc.0x20030080.0x10005000.tbf /home/ptumiali/projects/test/earlgrey-nexysvideo.elf /home/ptumiali/projects/test/earlgrey-nexysvideo_c_hello.elf

riscv32-unknown-elf-objcopy -O binary earlgrey-nexysvideo_c_hello.elf ready_c_hello.bin

Then I send .bin file in Open Titan directory. 
build-bin/sw/host/spiflash/spiflash --input=/home/ptumiali/projects/test/ready_c_hello.bin

The feedback from the board is too fast, so I recorded it with my phone and posted in my Google Drive. 
Link [https://drive.google.com/file/d/15RLk0wM5PMFzJBOkMuYnY8DPcVpijCzI/view?usp=sharing](url)
I added some !debug in main.rs and kernel.rs to check where appears the problem. 
Where the problem occurs is line 565 in main.rs, 
let (board_kernel, earlgrey_nexysvideo, chip, _peripherals) = setup();
The setup function exits and returns the variables and I think this makes a problem.
How can I fix it?

Thank you in advance for your help.

alistair23, 2021-08-10:
Can you attach a log of what the problem is?

student180529, 2021-08-10:
This is a .lst file from app`s folder. 
[rv32imc.0x20030080.0x10005000.txt](https://github.com/tock/tock/files/6960138/rv32imc.0x20030080.0x10005000.txt)

bradjc, 2021-08-10:
Is there any chance you are using an old version of libtock-c? Are you on the latest libtock-c git master?

student180529, 2021-08-10:
I `m using Tock and Libtock-c release-1.6. I also tried with tock 2.0 and libtock-c 1.6 but then I didn`t get any feedback messege from a board.

bradjc, 2021-08-10:
We're in the middle of preparing a new release, so different versions of the kernel and libtock-c are incompatible. Your best bet is to `git pull` on both the tock and libtock-c repos to the latest commit, and try again.

bradjc, 2021-08-11:
> I would also like to show you a small error.
> In the library libtock-c in Configuration file (line 110) toolchain for rv32imc and rv32imac is riscv64-unknown-elf. You only need to replace the 64 bit compiler to 32 bit to make everything work properly.

Thanks for the note. We've been using the toolchain with `--multilib` enabled, meaning they support both 32 and 64 bit architectures. But if there are platforms where it is convenient to get the riscv32 toolchain we will need to find a way to support that.