tock/tock #2754: [arty] sensors fault

alexandruradovici, 2021-08-10:
When running E21 on Arty A7-100 The sensors app faults with the following message:

```
Initialization complete. Entering main loop.
Loading processes from flash=0x40430000-0x4052FFFF into sram=0x80004000-0x8000FFFF
Loaded process[0] from flash=0x40430000-0x4044FFFF into sram=0x80004000-0x80006367 = "sensors"

panicked at 'Process sensors had a fault', kernel/src/process_standard.rs:323:17
	Kernel version release-2.0-rc2

---| No debug queue found. You can set it with the DebugQueue component.

---| RISC-V Machine State |---
Last cause (mcause): Store/AMO access fault (interrupt=0, exception code=0x08000007)
Last value (mtval):  0x0000000C

System register dump:
 mepc:    0x404053BE    mstatus:     0x00000088
 mcycle:  0x119F3F78    minstret:    0x000A665E
 mtvec:   0x40400102
 mstatus: 0x00000088
  uie:    false  upie:   false
  sie:    false  spie:   false
  mie:    true   mpie:   true
  spp:    false
 mie:   0x00000000   mip:   0x00000000
  usoft:  false               false 
  ssoft:  false               false 
  msoft:  false               false 
  utimer: false               false 
  stimer: false               false 
  mtimer: false               false 
  uext:   false               false 
  sext:   false               false 
  mext:   false               false 

---| App Status |---
𝐀𝐩𝐩: sensors   -   [Faulted]
 Events Queued: 0   Syscall Count: 3   Dropped Upcall Count: 0
 Restart Count: 0
 Last Syscall: Some(Memop { operand: 11, arg0: 2147505076 })

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x80006368═╪══════════════════════════════════════════╝
             │ ▼ Grant         948 |    948          
  0x80005FB4 ┼───────────────────────────────────────────
             │ Unused
  0x800053B4 ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   3072               S
  0x800053B4 ┼─────────────────────────────────────────── R
             │ Data           2996 |   2996               A
  0x80004800 ┼─────────────────────────────────────────── M
             │ ▼ Stack        2048 |   2048          
  0x80004000 ┼───────────────────────────────────────────
             │ Unused
  0x80004000 ┴───────────────────────────────────────────
             .....
  0x40450000 ┬─────────────────────────────────────────── F
             │ App Flash    130976                        L
  0x40430060 ┼─────────────────────────────────────────── A
             │ Protected        96                        S
  0x40430000 ┴─────────────────────────────────────────── H

 R0 : 0x00000000    R16: 0x00000000
 R1 : 0x40433C76    R17: 0x800051B4
 R2 : 0x80004750    R18: 0x80004800
 R3 : 0x00000000    R19: 0x40438588
 R4 : 0x00000000    R20: 0x40438542
 R5 : 0x4043143E    R21: 0x404384E6
 R6 : 0x0000000F    R22: 0x404384A6
 R7 : 0x00000200    R23: 0x00000000
 R8 : 0x80004BBC    R24: 0x00000000
 R9 : 0x00000000    R25: 0x00000000
 R10: 0x80004C18    R26: 0x00000000
 R11: 0x00000000    R27: 0x00000000
 R12: 0x00000008    R28: 0x00020000
 R13: 0x4043145A    R29: 0xFFFFFFFF
 R14: 0x80004C18    R30: 0x80004800
 R15: 0x00010009    R31: 0x8000518C
 PC : 0x40433CB0

 mcause: 0x08000007 (Store/AMO access fault)
 mtval:  0x0000000C

 Total number of grant regions defined: 4
  Grant  0 : --          Grant  2 : --        
  Grant  1 : --          Grant  3 : --        
 PMP regions:
  [0]: addr=0x40430000, size=0x00020000, cfg=0xD (r-x)
  [1]: addr=0x80004000, size=0x000013B4, cfg=0xB (rw-)

To debug, run `make lst` in the app's folder
and open the arch.0x40430060.0x80004000.lst file.

```

bradjc, 2021-08-11:
I'm thinking this app might just be too big for the arty-e21. It might be trying to run `_c_start_nopic()` with addresses that are too large. I'm thinking this because the app is failing very early (inside crt0.c), and the size in flash is 130976 (huge!). I tried on my board, and the size is 70640 (still huge!). (I can't explain why these numbers are different, however.)

I'm inclined to just ignore this error as 1) there are no sensors on the arty board anyway and 2) we want to delete the platform as soon as we have any other good riscv hardware.

alexandruradovici, 2021-08-11:
Most of the apps have this behavior, they start, run some memops (usually 0, 10, 11), wait for a second and than fault (https://github.com/tock/tock/issues/2429#issuecomment-894633056)

bradjc, 2021-08-11:
Using a better newlib (https://github.com/tock/libtock-c/pull/237) fixes this issue.

bradjc, 2021-08-11:
I actually have a new hypothesis on why the apps fault. I now think they are not being written to the arty correctly. Tockloader+openocd seem to only work when writing 0x10000 bytes (which must be why I rounded all apps up to 0x10000 in size). Apps that are larger than 2^16 do not get rounded to 0x20000 by tockloader, and I think are not written to the board correctly.

bradjc, 2021-08-12:
I'm pretty sure at this point that this issue is a tockloader bug where tockloader is not able to successfully write everything to flash. With this issue, and limitations around writing really small apps, I've decided to just workaround limitations in openocd and jlinkexe by having tockloader only write complete pages on each board. That fixes this problem, and removes the need for other hacks like minimum app sizes.

New tockloader release is needed, but the changes are currently in git. Release will probably coincide with 2.0.

closed 2021-08-12 by bradjc

bradjc, 2021-08-12:
Both.

alexandruradovici, 2021-08-12:
I'll test again.

alexandruradovici, 2021-08-14:
I tested again. apps still fault in the same way./

bradjc, 2021-08-16:
I just tried with:

- tock at release-2.0-rc2
- libtock-c at newlib-riscv branch, commit 231534ad2c720ad89cf9de3cdbb6cfe34945348e
- tockloader master, commit 347c473eedce69fbd21be351d3b9aa955ffb9700

```
cd tock/boards/arty_e21
make install

cd libtock-c/examples/sensors
make RISCV=1
tockloader install --erase
```

and I get what is expected (as there are no sensors):

```
[Sensors] Starting Sensors App.
[Sensors] All available sensors on the platform will be sampled.

```

Maybe the output of `tockloader list --verbose` would provide some insight? I'm fairly certain the issue is with tockloader not flashing the app binary correctly.

alexandruradovici, 2021-08-16:
I get the same output, but afyer a few seconds the application faults.

bradjc, 2021-08-16:
Can you send the updated panic output, and maybe the .tab file? Mine just keeps printing new lines and doesn't crash.

alexandruradovici, 2021-08-17:
Install
```
$ tockloader install
[INFO   ] No TABs passed to tockloader.
[STATUS ] Searching for TABs in subdirectories.
[INFO   ] Using: ['./build/sensors.tab']
[INFO   ] Using openocd channel to communicate with the board.
[INFO   ] Using settings from KNOWN_BOARDS["arty"]
[STATUS ] Installing app on the board...
[INFO   ] Found sort order:
[INFO   ]   App "sensors" at address 0x40430060
[INFO   ] Finished in 3.602 seconds
```

Kernel: [arty_e21.zip](https://github.com/tock/tock/files/6996508/arty_e21.zip)
.tab: [sensors.zip](https://github.com/tock/tock/files/6996506/sensors.zip)

Fault:
```
Initialization complete. Entering main loop.

panicked at 'Process sensors had a fault', kernel/src/process_standard.rs:323:17
	Kernel version release-2.0-rc2-64-g3c0d323b8

---| No debug queue found. You can set it with the DebugQueue component.

---| RISC-V Machine State |---
Last cause (mcause): Store/AMO access fault (interrupt=0, exception code=0x08000007)
Last value (mtval):  0x0000000C

System register dump:
 mepc:    0x40405454    mstatus:     0x00000088
 mcycle:  0x1141DCC6    minstret:    0x000A1397
 mtvec:   0x40400102
 mstatus: 0x00000088
  uie:    false  upie:   false
  sie:    false  spie:   false
  mie:    true   mpie:   true
  spp:    false
 mie:   0x00000000   mip:   0x00000000
  usoft:  false               false 
  ssoft:  false               false 
  msoft:  false               false 
  utimer: false               false 
  stimer: false               false 
  mtimer: false               false 
  uext:   false               false 
  sext:   false               false 
  mext:   false               false 

---| App Status |---
𝐀𝐩𝐩: sensors   -   [Faulted]
 Events Queued: 0   Syscall Count: 3   Dropped Upcall Count: 0
 Restart Count: 0
 Last Syscall: Some(Memop { operand: 11, arg0: 2147505076 })

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x80006368═╪══════════════════════════════════════════╝
             │ ▼ Grant         948 |    948          
  0x80005FB4 ┼───────────────────────────────────────────
             │ Unused
  0x800053B4 ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   3072               S
  0x800053B4 ┼─────────────────────────────────────────── R
             │ Data           2996 |   2996               A
  0x80004800 ┼─────────────────────────────────────────── M
             │ ▼ Stack        2048 |   2048          
  0x80004000 ┼───────────────────────────────────────────
             │ Unused
[... 111 lines cut ...]
 R9 : 0x00000000    R25: 0x00000000
 R10: 0x80004C18    R26: 0x00000000
 R11: 0x00000000    R27: 0x00000000
 R12: 0x00000008    R28: 0x00020000
 R13: 0x4043145A    R29: 0xFFFFFFFF
 R14: 0x80004C18    R30: 0x80004800
 R15: 0x00010009    R31: 0x8000518C
 PC : 0x40433CB0

 mcause: 0x08000007 (Store/AMO access fault)
 mtval:  0x0000000C

 Total number of grant regions defined: 4
  Grant  0 : --          Grant  2 : --        
  Grant  1 : --          Grant  3 : --        
 PMP regions:
  [0]: addr=0x40430000, size=0x00020000, cfg=0xD (r-x)
  [1]: addr=0x80004000, size=0x000013B4, cfg=0xB (rw-)

To debug, run `make lst` in the app's folder
and open the arch.0x40430060.0x80004000.lst file.

```

gcc version: riscv64-unknown-elf-gcc (SiFive GCC-Metal 10.2.0-2020.12.8) 10.2.0

bradjc, 2021-08-17:
I don't see the `[Sensors] Starting Sensors App.`... output?

But it seems like you compiling the app without the new riscv newlib optimized for size. If you try building the sensors app with the latest libtock-c (the relevant PR was merged today) it should hopefully work.

alexandruradovici, 2021-08-17:
> I don't see the `[Sensors] Starting Sensors App.`... output?
> 
If you look at the system call trace, it seems that it faults before main.

> But it seems like you compiling the app without the new riscv newlib optimized for size. If you try building the sensors app with the latest libtock-c (the relevant PR was merged today) it should hopefully work.
Thanks, I'll try.

alexandruradovici, 2021-08-18:
It works :smile: