tock/tock #2815: Does tock have device driver for spi in lowrisc/earlgrey?

nicolast0604, 2021-09-10:
Hi 
I am a beginner to tock and trying to use opentitan nexysvideo as an experiment board.
After checking the directory, I am curious if tock have device driver for SPI as I found it in RP2040 but not found in lowrisc or earlgrey.

Thanks a lot.

alistair23, 2021-09-10:
Hello,

Unfortunately Tock does not have support for OpenTitan SPI. I have written a driver in the past (https://github.com/alistair23/tock/commits/alistair/spi) but at the time SPI wasn't exposed on the board so I was unable to test it.

Adding SPI support would be a good learning experience to Tock though if you are interested in doing it

closed 2021-09-10 by alistair23

nicolast0604, 2021-09-17:
Hi Alistair
Great thanks for your quick response.  I am very interested in adding SPI support to gain experience on tock.

As I am a beginner for this field, I am confused why SPI wasn't exposed on the nexys video baord as I found pinout/pinmux 
https://docs.opentitan.org/hw/top_earlgrey/ip/pinmux/doc/autogen/pinout_nexysvideo/index.html

Do I miss anything or misunderstood?

I am wondering if I can expose the SPI of nexys video to the PMOD and test the SPI device driver by adding two nexys video board or adding an external board by PMOD. 
Do you have any suggestion?

Nicolas

alistair23, 2021-09-19:
From reading the link you provided I don't think you can change the pins for the SPI. It seems like they are directly connected and not muxed

nicolast0604, 2021-09-23:
Hi Alistair
Thanks for your feedback but I am confused as I am beginner and know little. 
If the SPI are directed connected, can I change the pin out when I synthesis in vivado and get it connect externally through PMOD?
The other thing I am confused is that I do see SPI host1 (not host0) which is muxed.

spi_host1_sd[0] | muxed | - | kTopEarlgreyPinmuxOutselSpiHost1Sd0 / kTopEarlgreyPinmuxPeripheralInSpiHost1Sd0 |  
-- | -- | -- | -- | --
spi_host1_sd[1] | muxed | - | kTopEarlgreyPinmuxOutselSpiHost1Sd1 / kTopEarlgreyPinmuxPeripheralInSpiHost1Sd1 |  
spi_host1_sd[2] | muxed | - | kTopEarlgreyPinmuxOutselSpiHost1Sd2 / kTopEarlgreyPinmuxPeripheralInSpiHost1Sd2 |  
spi_host1_sd[3] | muxed | - | kTopEarlgreyPinmuxOutselSpiHost1Sd3 / kTopEarlgreyPinmuxPeripheralInSpiHost1Sd3

Thanks again for your support and please feel me to me know your feedback and advice.

alistair23, 2021-09-23:
Ah, I didn't see `spi_host1_sd`. That does appear to be muxed and might work. You could also change the design before you generate it, if you are comfortable doing that.

Sorry, I can't be of more help. I'm not a hardware designer and I haven't spent any time on the pinmux/SPI signals