tock/tock #2833: console: cannot distinguish between failure and success in upcall return

jettr, 2021-09-21:
In the console capsule, it is impossible to distinguish from a successful write of 9 bytes and an error of `NOMEM` since the first parameter of the upcall return is the bytes written in the [success case][0] or the positive failure code in the [failure case][1].

I recommend moving the bytes written to the second parameter, and reserve the first parameter for status code. If the status code is `0`, then the second parameter will contain the number of bytes written.

We would also need to update the [documentation][2].

If people agree that this is an issue and the proposed solution fixes it, then I will upload a PR with the fix.

[0]:https://github.com/tock/tock/blob/master/capsules/src/console.rs#L307
[1]:https://github.com/tock/tock/blob/master/capsules/src/console.rs#L318
[2]:https://github.com/tock/tock/blob/master/doc/syscalls/00001_console.md#subscribe-number-2

bradjc, 2021-09-21:
Are these the `// XXX This shouldn't ever happen?` cases?

In general I think you are exactly right and the change you propose is what we want. However, we decided to guarantee stability for the console syscall interface (https://github.com/tock/tock/pull/2769). So we can't move the return value without either 1) going to 3.0 or 2) violating our guarantees about 2.0 app stability.

What we can do is keep what we have, and add a new subscribe call and upcall with the desired argument ordering.

Since it doesn't look like the error case will ever happen in the current implementation, I'm opposed to breaking our compatibility guarantee until 3.0.

jettr, 2021-09-21:
Can we at least make the error number negative so it doesn't alias with success before 3.0? That is probably how it existed in 1.x before we switch from ReturnCode to ErrorCode/StatusCode, right?

bradjc, 2021-09-21:
> Can we at least make the error number negative so it doesn't alias with success before 3.0? That is probably how it existed in 1.x before we switch from ReturnCode to ErrorCode/StatusCode, right?

We explicitly moved away from this (hence the specificity around ErrorCode and StatusCode) because it leads to exactly this problem, and I don't think we want to re-introduce it in our most used capsule.

jettr, 2021-09-21:
> We explicitly moved away from this (hence the specificity around ErrorCode and StatusCode) because it leads to exactly this problem, and I don't think we want to re-introduce it in our most used capsule.

Agreed that it isn't my first choice, but I do prefer that to having errors and success be indistinguishable.

Since console is already not following its API contract, why not fix the code and the API documentation contract right now since 2.0 is still very early, and at least the libtock-rs standard implementation for console doesn't care either way.

bradjc, 2021-09-21:
I'm only one person, but I at least am very opposed to changing existing console syscalls, or anything we said was stable in that PR. Any bug of these sorts will seem small, or be early, or a minor change, but the result is that we don't actually have any stability guarantees in tock. I think as an OS kernel we have to stick to our promises.

jettr, 2021-09-21:
> I'm only one person, but I at least am very opposed to changing existing console syscalls, or anything we said was stable in that PR. Any bug of these sorts will seem small, or be early, or a minor [...]

I can get behind that if we are being strict about it. That means we should correct the upcall `2` code for console since it is not following the [promise][0], then right?

[0]:https://github.com/tock/tock/blob/master/doc/syscalls/00001_console.md#subscribe-number-2

bradjc, 2021-09-21:
> That means we should correct the upcall `2` code for console since it is not following the [promise](https://github.com/tock/tock/blob/master/doc/syscalls/00001_console.md#subscribe-number-2), then right?

I think the code is the better source of ground truth. We do our best with documentation, but, we're only so many people.

ppannuto, 2021-09-21:
I think I feel like the best solution to this is mark subscribe `2` as deprecated (but leave it otherwise unchanged) and add a subscribe `3` with the proposed new semantics. This won't break any existing code (which will be unable to distinguish ENOMEM and 9-byte write). The `libtock-c/rs` implementations can just quit using `2`, and whenever we go to Tock 3.0 we can drop the old code path.

phil-levis, 2021-09-21:
Another approach is to just use a new device number for a new console device, which has a more strongly specified API.

bradjc, 2021-09-21:
> Introducing a bug becomes ground truth?

If it's part of the userspace API, userspace applications rely on it, and we said it would be stable, then yes, I would say a bug becomes ground truth.

bradjc, 2021-09-21:
Wait but I still don't see what the issue is in this case. The documentation does not say that the first argument in upcall 1 can return an error. Only that it will return the number of bytes.

Unless I'm missing something, the console code only calls `schedule_upcall(1, ...` in a match statement after calling `self.send_continue()`. If we look at the implementation of `send_continue()`:

```rust
    fn send_continue(
        &self,
        app_id: ProcessId,
        app: &mut App,
    ) -> Result<bool, Result<(), ErrorCode>> {
        if app.write_remaining > 0 {
            self.send(app_id, app);
            Ok(true)
        } else {
            Ok(false)
        }
    }
```

It can only ever return `Ok()`. If we then look at the usage of `send_continue()` (there are a couple blocks like this):

```rust
                match self.send_continue(appid, app) {
                    Ok(more_to_send) => {
                        if !more_to_send {
                            // Go ahead and signal the application
                            let written = app.write_len;
                            app.write_len = 0;
                            upcalls.schedule_upcall(1, (written, 0, 0)).ok();
                        }
                    }
                    Err(return_code) => {
                        // XXX This shouldn't ever happen?
                        app.write_len = 0;
                        app.write_remaining = 0;
                        app.pending_write = false;
                        upcalls
                            .schedule_upcall(
                                1,
                                (kernel::errorcode::into_statuscode(return_code), 0, 0),
                            )
                            .ok();
                    }
                }
```

The `schedule_upcall(1, ...` call only happens in the error case...which can never happen.

So while we do seem to be guilty of having some misleading dead code in the console capsule, I don't think we are either aliasing the first argument or out of sync with our documentation.

lschuermann, 2021-09-21:
> I think I feel like the best solution to this is mark subscribe `2` as deprecated (but leave it otherwise unchanged) and add a subscribe `3` with the proposed new semantics. This won't break any existing code

FWIW, we've done that with the alarm capsule command syscall number 4 ("set absolute expiration") and that worked pretty well: https://github.com/tock/tock/pull/2781

hudson-ayers, 2021-09-21:
For subscribe it is a little messier because we have to pay the cost of storing that upcall in the grant region indefinitely, since subscribe numbers have to be contiguous and start at 0. Though we could shift all subcribe numbers down by 1 in a future major release.

bradjc, 2021-09-21:
As part of the 2.0 work send was updated to not fail, and instead queue to send later: https://github.com/tock/tock/commit/acc67d556b77c429adc6de44b7497969d4018caf That change just didn't propagate past the `send()` function. We can update that now and resolve this issue that way.

phil-levis, 2021-09-21:
> > Introducing a bug becomes ground truth?
> 
> If it's part of the userspace API, userspace applications rely on it, and we said it would be stable, then yes, I would say a bug becomes ground truth.

Does that mean by definition a stable API can't have a bug in it? Or a bug only exists if there is a mismatch between the implicit expectations of userspace based on their logic and the implicit behavior of the capsule based on its implementation? This is kind of saying that the API is not actually what's stable, it's instead the coupling of userspace code and the capsule.

I'm strongly against this framing: this means there is no such thing as a specification, just an artifact. It means there is no such thing as a distinction between design and implementation: the design is defined by the implementation.

hudson-ayers, 2021-09-27:
> Introducing a bug becomes ground truth?

I think reasoning about what is ground truth is not the right framing for this discussion. I think it is clearly the case that for APIs for which we have finalized documentation, the documented API represents "ground truth" when there is a disagreement between the documentation and the code. That does not, however, mean that you can change the API presented to userspace without a major version bump. If the bug was present in the original major version release, fixing that bug requires bumping the major version. If an API breaking bug is introduced as part of a minor version, it can be fixed in a minor version, but not if the breakage was part of a major release (https://semver.org/).

phil-levis, 2021-09-29:
> If the bug was present in the original major version release, fixing that bug requires bumping the major version.

It seems very strange to me that bug fixes require a major version bump. Would we say the same about the kernel? Suppose there's a bug in how allow is handled in a particular edge case: would fixing this bug require bumping to Tock 3.0?

ppannuto, 2021-09-29:
I believe so, yes. The semver FAQ language is a little dismissive on this point, but does discuss it: https://semver.org/#if-even-the-tiniest-backwards-incompatible-changes-to-the-public-api-require-a-major-version-bump-wont-i-end-up-at-version-4200-very-rapidly.

As I understand it, the philosophy / dogma is such that code that worked at a given major version MUST continue to work at subsequent releases of that major version (even if it only worked because it relied on a bug).

phil-levis, 2021-09-29:
I think the semver point is about the API, not its implementation. I am defining a bug as "the implementation does not do what the API specifies," not "there is a mistake in the API." I think the next point in the FAQ is particularly relevant here and perhaps the real root of our problem. :)

hudson-ayers, 2021-09-29:
>> I think the semver point is about the API, not its implementation. I am defining a bug as "the implementation does not do what the API specifies," not "there is a mistake in the API." I think the next point in the FAQ is particularly relevant here and perhaps the real root of our problem. :)

To steal a quote from elsewhere

> If you develop an app and it works, then you update a library you use from 2.0 to 2.1 and it breaks, and you complain, and the library developer says "well you see, i know our api changed, but only to match the documentation that we promised, so im sure you can see we were right to break your application", most developers would say "well of course!! i should have programmed my app to the documentation, not the actual implementation, that way it would have been broken from the start"

I think this scenario gets to the root of the problem here. Obviously the actually desirable thing would be for users to submit a bug report in cases where they find these discrepancies, but that will not always happen. Users should in general be able to make minor version updates without having to extensively retest their code -- this is a desirable thing to maintain because it encourages users to update regularly to pull in bug fixes without requiring them to extensively retest. If a given library starts breaking working code in minor releases by changing the API, it can discourage those users from updating to new minor versions in the future.

phil-levis, 2021-10-02:
> Users should in general be able to make minor version updates without having to extensively retest their code -- this is a desirable thing to maintain because it encourages users to update regularly to pull in bug fixes without requiring them to extensively retest.

This sentence is self-contradictory; if there are bug fixes, how can they not extensively retest (e.g., the other quote)? Or what's a bug, then?

hudson-ayers, 2021-10-04:
> This sentence is self-contradictory; if there are bug fixes, how can they not extensively retest (e.g., the other quote)? Or what's a bug, then?

There could be many bug fixes in the Tock kernel that do not affect the userspace API -- for the console example, if there is a bug where all UART hardware is always powered on rather than just those in use, or if there is a bug where console input buffers are accidentally being shared with another capsule. Bugs where the API behavior does not match the API documentation are just one class of bug, but they are the type of bug that requires a major version bump to be fixed.

All this being said, I think Brad is right that we don't really need to be having this conversation for this particular issue, because the console does in fact match the behavior prescribed in the API docs, since the code which looks like it could return an error is in fact unreachable. So for this particular issue we should just be removing the dead code, and updating the code comments to make it clear that returning an error is not an option in this upcall.