tock/tock #2909: Question on `tock-register-interface ReadWrite or WriteOnly`

nihalpasham, 2021-11-29:
```rust
 pub fn set(&self, value: T) { 
     unsafe { ::core::ptr::write_volatile(self.value.get(), value) } 
 } 
```
The `set method` for `ReadWrite or WriteOnly` takes an immutable reference to self and mutates the underlying value.  I'm guessing this is Ok as `Tock` is single threaded and the problem of simultaneous `aliasing + mutation` doesnt arise. 

My question is - in a multithreaded environment, where we hand out multiple immutable refs, could this potentially result in undefined behavior.  If yes, I presume an easy work-around would be to ensure `settable methods` use a exclusive ref i.e. a mutable borrow instead of an immutable borrow of self. 

```rust
 pub fn set(&mut self, value: T) { 
     unsafe { ::core::ptr::write_volatile(self.value.get(), value) } 
 } 
```
Given the above reasoning, would it make sense to change method signatures in the api to enforce this behavior?

lschuermann, 2021-11-29:
> The `set method` for `ReadWrite or WriteOnly` takes an immutable reference to self and mutates the underlying value. I'm guessing this is Ok as `Tock` is single threaded and the problem of simultaneous `aliasing + mutation` doesnt arise.

Indeed we use the concept of [interior mutability](https://doc.rust-lang.org/reference/interior-mutability.html) here. This is quite a common paradigm and well supported in the Rust core library, e.g. through types such as `Cell`. `Cell` uses `UnsafeCell` to accomplish this, which allows mutable aliasing of some memory. However, Rust still assumes to have full control over that memory and it must behave how Rust expects it to. This implies that the memory may not change if not written from the current thread, [explaining why `UnsafeCell` is `!Sync`](https://doc.rust-lang.org/nomicon/send-and-sync.html). Because `tock-registers` is intended to be used on memory-mapped registers, which can change their value at any time, all accesses are performed through volatile operations. This means that Rust can never assume that two accesses to the underlying memory will return the same value, and it will severely limit the amount of optimizations (e.g. reordering) which Rust is allowed to do on these operations.

> My question is - in a multithreaded environment, where we hand out multiple immutable refs, could this potentially result in undefined behavior.

I don't think so, for two reasons: first and foremost, the `tock-registers` types are `!Sync` because they wrap an `UnsafeCell`. This means that Rust will prevent references to these values from being shared between threads. Furthermore, even if we'd change the inner representation to something which would not be a wrapper around an `UnsafeCell` (but would still have to be the same size and memory layout as the represented `IntLike` type, to have the `tock-registers` type's `#[repr(transparent)]` cause it to have the same size and memory layout as the represented type), we still _only_ perform volatile operations on it. While performing multiple volatile reads / writes of the same memory from different threads will not necessarily have any _predictable_ behavior, it does not violate Rust's memory model.

> Given the above reasoning, would it make sense to change method signatures in the api to enforce this behavior?

Unfortunately, this is not trivial. Tock's asynchronous event-loop style kernel architecture causes a lot of circular references between different parts of the operating system. This cannot be represented through mutable references in Rust. As a consequence of that, if register accesses within a peripheral whose methods take immutable self-references would require a proper exclusive mutable reference, this would need to be obtained first before the registers can be accessed, either through a non thread-safe construct such as `RefCell` or a thread-safe construct such as a `Mutex`. This would likely have a significant performance impact. Luckily, as described above, we should be fine without it.

nihalpasham, 2021-11-30:
> the tock-registers types are !Sync because they wrap an UnsafeCell. This means that Rust will prevent references to these values from being shared between threads.

Ah, I didn't realize registers were wrapped in an `UnsafeCell`. That makes sense. 

PS: `Tock` sounds interesting.  Are there plans to add more video-material (i.e. talks) on `TockOS`?

hudson-ayers, 2021-11-30:
I know of a couple videos available today: https://www.youtube.com/watch?v=BvA5fICr5Gk is a recording from when @alevy presented the original Tock paper at SOSP in 2017. 

https://www.youtube.com/watch?v=3s23eOYUHAA is a similar talk on Tock and secure system design, from later that year.

lschuermann, 2021-12-13:
@nihalpasham Given this issue seems resolved, I'll close it. If you have any further questions feel free to reopen.