tock/tock #2910: SPI read length incorrect

jettr, 2021-11-29:
Accorrding to the comment in [spi_controller.rs][0], we should be able to perform a write operation without providing a read buffer. The way the code is written in [spi_controller][0] and [spi_peripheral][1] does not do this. It should instead be something like the following:

```rust
let wlen = app.app_write.enter(|w| w.len()).unwrap_or(0);
let rlen = app.app_read.enter(|r| r.len()).unwrap_or(0);
let len = if rlen == 0 { wlen } else { wlen.min(rlen) };
```

I want to fix this after #2906 lands as to not give myself more merge conflicts (that is also how I noticed the issue in the code).

[0]:https://github.com/tock/tock/blob/master/capsules/src/spi_controller.rs#L223
[1]:https://github.com/tock/tock/blob/master/capsules/src/spi_peripheral.rs#L213

phil-levis, 2022-01-14:
I think this is not quite correct: if I pass a read buffer of length 0 then there should be no read/write; this is distinct from not passing a read buffer, in which there can be a write-only operation.

Why doesn't the current code work?

phil-levis, 2022-01-14:
(Or has it been fixed, hence my confusion?)

jettr, 2022-01-14:
I couldn't even remember myself because on the surface is looks correct. The subtle issue comes from the fact that a non-shared buffer and 0-sized shared buffer both return 0 for the size and neither return an error. This subtle behavior existed before and after the change to move allow into common kernel code.

This is due to a process buffer of NULL is valid and returns as a 0-sized slice always.

So we really can't tell the different between not sharing a read buffer and sharing a 0-size buffer. I would suggested that we do not support sharing a 0-sized read buffer as a way to prevent the write operation. Thoughts?

phil-levis, 2022-01-14:
Ah! This is because of the transition to zero-length buffers denoting the absence of a buffer. That is, the buffer is no longer an Option. Makes sense.

PR #2945 (2022-01-17): spi: fix issue when read buffer not present
### Pull Request Overview

If a read buffer was not passed in, then the write operation would not occur since the read length returned would always be 0 (not none). Then the min would be 0, and the operation would fail since len was 0.

This fixes #2910

### Documentation Updated

- [x] None

### Formatting

- [x] Ran `make prepush`.
closed 2022-01-17 by bors[bot]