tock/tock #2940: Buffer mutability in hil::digest

phil-levis, 2022-01-14:
Currently, the `Digest` HIL takes only mutable buffers. This is a problem if you want to compute a digest over data in flash (e.g., verify the hash on a binary).

For example, `DigestData::add_data` https://github.com/tock/tock/blob/7094e5217e0399d29b3545f99491a4bec24a43bf/kernel/src/hil/digest.rs#L74 takes a `LeasableBuffer`, which wraps around a mutable buffer. This is evident from the return value of `add_data`.

We either need to take an approach similar to RSA (having separate traits for mutable and immutable buffers) in #2839 or come up with a general solution (which will eventually involve some unsafe code).

hudson-ayers, 2022-01-15:
My view of the solution space:

1. Use something like #2839, possibly combining the implementation of mut_imut_buffer with leasable_buffer into a single type that handles both things. This comes with the various overhead / downsides of mut_imut_buffer, but should work.
2. Decide that a given peripheral can support *either* RAM buffers of Flash buffers, but not both in the same binary. To achieve this, we can basically use the approach described in https://lab.whitequark.org/notes/2016-12-13/abstracting-over-mutability-in-rust/ for the HIL. This ultimately means that the underlying chip peripheral will also be generic over some type implementing `AsRef<[u8]>`. You only instantiate one chip peripheral, so at compile time you would have to set the type of that buffer to either be `&'static [u8]` or `&'static mut [u8]`. This would work for a situation where all digest data is either in flash or in RAM, minimizes code duplication, and does not require any unsafe. It does not address any situations where a capsule needs to calculate digests of some buffers in flash and some buffers in RAM (without copies out of flash at least).
3. Have two different traits, one for mutable buffers and one for immutable buffers. This works for any case, but will lead to a lot of duplicate code in chip peripheral implementations, requires peripherals hold space for two clients, space for two buffers, etc. The upside is that if a user of a chip knows they don't care about flash buffers they can just implement for RAM buffers with no overhead over what we do today.
4. A single trait, but with `add_ram_data()` / `add_flash_data()` and `RamDataClient` / `FlashDataClient. This is similar to above, with less duplicate code for supporting both RAM/Flash buffers but no opt-out from the overhead of storing two clients/two buffers. 
5. I wonder if it is possible for the HIL to use the approach in 2, if we are clever enough about the implementation of chip peripherals. Ideally we would have some compile-time detection of if different generic types are passed, and then decide how many buffer/client types we want to store based on this, or somehow isolate the generic types so that they are not on the implementation of the chip peripheral itself, but some other type that is hidden behind an `&dyn`. I couldn't think of any good solution here that did not involve procedural macros unfortunately.

bradjc, 2022-07-07:
Resolved in #3041