tock/tock #2941: register_structs! macro: reached recursion limit during macro expansion

imDema, 2022-01-14:
I'm using `svd2regs` to parse the SVD for the STM32MP157 (which can be obtained [here](https://github.com/stm32-rs/stm32-rs/blob/master/svd/vendor/STM32MP157x_v1r6.zip)) and get an generated source which won't compile due to recursion limits in macros.

## Replicating
+ Download and extract STM32MP157x svd
+ Run `svd2regs` for the RCC component ```python3 ./tools/svd2regs.py --svd stm32mp157.svd --save generated.rs RCC```
+ (Other small issue) Fix the outdated imports by changing `kernel::common` to `kernel::utilities`
+ Try compiling

Here's an archive with pre generated code, run `cargo check` to replicate [sv2regs_issue.tar.gz](https://github.com/tock/tock/files/7870019/sv2regs_issue.tar.gz) (The tock directory is missing for archive size reasons, clone or change the path in `Cargo.toml`)

I'd be happy to help if needed

alexandruradovici, 2022-01-14:
The `register_structs` macro is indeed recursive. To prevent infinite loops, Rust has a [limit](https://doc.rust-lang.org/reference/attributes/limits.html) for the number of recursions that it will execute.

You should be able to change it as describe [here](https://doc.rust-lang.org/reference/attributes/limits.html), but you might have to edit the [lib.rs](https://github.com/tock/tock/blob/master/libraries/tock-register-interface/src/lib.rs) file from the `tock-register-interface` crate the `register_structs` macro is defined.

This should be something similar to
```rust
// place here anything higher than 128
#![recursion_limit = "512"]
```

imDema, 2022-01-14:
I figured that was the culprit, however I was wondering if a one shot (nested) macro could be beneficial instead of just increasing the limit to an arbitrary amount.
From what I can tell, the recursion is only really needed for the padding fields, which need to know the offset of the next field in order to compute the difference, which otherwise would need to be captured twice.
However by using a nested implementation, it would be possible to first duplicate the offset and then recapture it in a nested macro to achieve the same result without recursion.

The idea is something like this:
```rust
// ...
(0x0D8 => rcc_ddritfcr: ReadWrite<u32, RCC_DDRITFCR::Register>),
(0x0DC => _reserved5),
(0x100 => rcc_mp_bootcr: ReadWrite<u32, RCC_MP_BOOTCR::Register>),
(0x108 => @END),

// Duplicate offset for each field

// ...
0xD8), (0x0D8 | rcc_ddritfcr: ReadWrite<u32, RCC_DDRITFCR::Register> |
0xDC), (0x0DC | _reserved5 |
0x100), (0x100 | rcc_mp_bootcr: ReadWrite<u32, RCC_MP_BOOTCR::Register> |
0x108),

// Reorder whitespace for readablilty

/* ... */ 0xD8),
(0x0D8 | rcc_ddritfcr: ReadWrite<u32, RCC_DDRITFCR::Register> | 0xDC),
(0x0DC | _reserved5 | 0x100),
(0x100 | rcc_mp_bootcr: ReadWrite<u32, RCC_MP_BOOTCR::Register> | 0x108),

// Now each field can be processed independently and we can compute the difference without recursion
```
This way the macro could work even with components with many registers

> Disclaimer: I'm not that experienced with macros, so there may be a better solution

alexandruradovici, 2022-01-14:
Even tough I'm not keen about the idea of duplicating offsets, not being able to write large register structures without modifying the upstream code is a problem. As I'm not very experienced in macros either, @lschuermann maybe you can take a look at this?

lschuermann, 2022-01-14:
It appears I have hit a major roadblock while trying to implement this. Conceptually, a solution to this issue should actually be rather simple. 

Fundamentally, the `register_structs` macro needs to be able to represent two basic kinds of fields: proper fields and padding fields. When we want to iterate over some tokens in a Rust macro non-recursively, we must be able to represent the iterables through the same match pattern and cannot perform any conditional expansion logic inside the `macro_rules` branch itself. To implement any conditional logic during macro expansion, the `macro_rules` match patterns must be used, by invoking a different or the same macro again recursively. This implies that, to iterate over the fields in the `register_structs` macro non-recursively, they have to be translated into a uniform representation.

Also, as per the  [macro expansion order hierarchy](https://rustc-dev-guide.rust-lang.org/macro-expansion.html#the-expansion-order-hierarchy), macros are expanded from outer to inner macro; that is the outer macro generates a token tree, when _then_ may invoke further macro expansions.

To convert the fields to a uniform specification, viewing the struct body as a list of field specifications (e.g. `(0x00 => name: Type),`), we let the macro traverse over the list, and at the same time traverse over the identical list with the head element removed (index shifted by one). This way we have a "lookahead" element from which we are able to determine the next offset, which can in turn be used to generate a proper amount of padding for fields. It is important that we append a dummy `@LOOKAHEADEND` field to the lookahead list, as otherwise the lists would be of unequal length, making it impossible to iterate over both simultaneously. This could look something like this:

```rust
macro_rules! register_fieldspecs {
    (@root
        $(#[$headattr:meta])* ($($headfield:tt)*),
        $($(#[$tailattr:meta])* ($($tailfields:tt)*)),* $(,)?
    ) => {
        register_fieldspecs!(@lookahead
            (($(#[$headattr:meta])* ($($headfield)*)),
             $(($(#[$tailattr:meta])* ($($tailfields)*))),*)
            ($(($($tailfields)*)),*,
             (@LOOKAHEADEND))
        )
    };

    (@lookahead
       ($(($($attrfield:tt)*)),*)
       ($(($($lookaheadattrfield:tt)*)),*)
    ) => {
        $(
            register_fieldspecs!(@field
                ($($attrfield)*) ($($lookaheadattrfield)*))
        ),*
    };

    (@field
        ($($attrfield:tt)*)
        ($($lookaheadattrfield:tt)*)
    ) => {
        // Generate the field specification consisting of a field name
        // and field type, something like
        abc: u32
    };
}
```

This will yield a list of field specifications in the form of `$($fname:ident : $ftype:ty),*`.

As a second stage, we "_simply_" need to turn this collection of field specifications into a struct. Easy enough, right? For us to be able to build a struct's body, we need to have one or more specifications of the form `$a:ident: $b:ty`, but the key issue is that macro expansion appears to be entirely impossible within the struct field's identifier specification:

```rust
macro_rules! type_macro { () => { u32 }; }
macro_rules! ident_macro { () => { abc }; }
macro_rules! spec_macro { () => { abc: u32 }; }

struct A {
    // works
    name: type_macro!(),
    // does not work, error: expected `:`, found `!`
    ident_macro!(): u16,
    // does not work, error: expected `:`, found `!`
    spec_macro!(),
}
```

It seems this fundamentally issue prevents us from using a macro invocation to generate the field specifications non-recursively, because to translate the original field specifications into the uniform representation, a macro invocation is required:

```rust
macro_rules! register_structs {
    (@root $(($($origfieldspec:tt)*),)*) => {
	struct A {
	    $(
                // Rust does not like this
		register_structs!(@field $($origfieldspec)*)
	    ),*
	}
    };

    (@field $($origfieldspec:tt)*) => {
	// Generate the field specification, e.g.
	abc: u32
    };
}
```

Furthermore, trying to generate the uniform representation first and then injecting it into the struct isn't an option either. The following snippet works

```rust
#[macro_export]
macro_rules! register_spec_to_struct {
    ($($fname:tt : $ftype:tt),*) => {
	struct A {
	    $($fname: $ftype),*
	}
    };
}

register_spec_to_struct!((abc, u32));
```

, however an invocation along the lines of

```rust
register_spec_to_struct!(
    register_fieldspecs!(
        (0x00 => a: u32),
        (0x04 => b: u16),
    )
);
```

does not, as `register_fieldspecs!` cannot be matched onto `$($fname:tt : $ftype:tt),*`. Also

```rust
#[macro_export]
macro_rules! register_spec_to_struct {
    ($($input:tt)*) => {
	struct A {
	    $($input)*
	}
    };
}

register_spec_to_struct!(
    register_fieldspecs!(
        (0x00 => a: u32),
        (0x04 => b: u16),
    )
);
```

does not work, even though the inner macro would generate an accepted token tree, as the `register_spec_to_struct!` macro is simply passed the token tree of the inner macro invocation. If we could change the macro expansion order hierarchy, an approach like this could work.

That's a bummer. So how does the `register_structs!` macro do it currently? By _iterating_ over the field specifications, it can directly match on the field specifications and extract the field identifier and type, which it can then use to recursively build the struct body. But given that a recursive macro is what's causing issues in the first place, that is not an option.

Maybe I have missed something, it's pretty hard to keep track of what gets expanded when. @imDema Do you perhaps have a PoC which can generate struct field specifications from a `macro_rules` macro non-recursively?

imDema, 2022-01-15:
Ok, so I managed to get a PoC of a macro which will compile and create a struct with fields generated by independent macros and having access to the offset of the field and of the following one.
This is the idea essentially: duplicate, discard the first offset, then by using a generic pattern with `$()?` that matches all fields write `$id: macro!(<generic pattern>),` and use the inner macro to pattern match and handle padding and fields accordingly.
```rs
#[macro_export]
macro_rules! register_fields_2 {
    // Macro entry point.
    (
        $(#[$attr_struct:meta])* $vis_struct:vis $name:ident $(<$life:lifetime>)? {
            $(
                $(#[$attr:meta])*
                ($offset_start:expr => $($desc:tt)*),
            )*
        }
    ) => {
        // Duplicate the offsets
        $crate::register_fields_2!(
            @prepare
            $(
                , $offset_start;
                $(#[$attr])*,
                $offset_start , ($($desc)*)
            )+
            ; / ($(#[$attr_struct])*
            $vis_struct $name $(<$life>)?)
        );
    };

    // Write the structs placing a type macro for each field
    // Optional tokens in field syntax to process them without distinction,
    // The per field macro will then make the distinction via pattern matching
    (
        @prepare, $_discard:expr;
        $(
            $(#[$attr:meta])*,
            $a:expr, ($(@)? $id:ident $(: $ty:ty)?)$(, $b:expr)?;
        )+
        / ($(#[$attr_struct:meta])*
        $vis_struct:vis $name:ident $(<$life:lifetime>)?)
    ) => {
        $(#[$attr_struct])*
        #[repr(C)]
        $vis_struct struct $name $(<$life>)?
        {$(
            $(#[$attr])*
            $id: $crate::register_fields_2!(@field ($a$(, $b)?) => ($id $(: $ty)?)),
        )+}
    };

    // Field macro for the @END token, this is a hack to keep the same API.
    // It will result in a warning for naming convention, there may be a way to avoid this.
    (
        @field ($a:expr) => ($id:ident $($ty:ty)?)
    ) => {
        [u8; 0]
    };

    // Field macro for padding spec (no type)
    (
        @field ($a:expr, $b:expr) => ($id:ident)
    ) => {
        [u8; $b - $a]
    };

    // Field macro for regular field
    (
        @field ($a:expr, $b:expr) => ($id:ident: $ty:ty)
    ) => {
        $ty
    };
}
```
[Small PoC](https://github.com/tock/tock/files/7875278/sv2regs_issue.tar.gz)

This does work, however as I said, I'm not really experienced with macros so take it as a dirty PoC which still needs work.

lschuermann, 2022-01-21:
@imDema Thanks for this small PoC! I knew I must've missed something, I didn't realize that we can use a common match pattern for the `@END` field, padding fields and proper fields, while still being able to extract a suitable attribute name from it which can be used to generate the struct field specifications. The `$(@)? $id:ident` match is really clever!

Actually, thinking about it now we can probably also just eliminate the last list iteration and thus completely remove the `@END` from the equation, also eliminating the warning it generates as an attribute identifier. I'll give it a shot this weekend.

bradjc, 2023-08-18:
Please comment if this is still something we should pursue.

lschuermann, 2023-08-18:
I'd say this depends on the general direction `tock-registers` is taking, at least in the short term. If we stick to our current types & macros for now, this seems like an improvement worth making (in fact, #3587 hit this again). @imDema's PoC seems workable, but we (I) would need to invest some time fully integrating it. If we're switching to an entirely different design anyways (like what @jrvanwhy proposed), that effort would not seem worth it.

baltazarortiz, 2024-06-18:
I was running into this today - for anyone else stumbling across this while trying to debug I actually only needed to add the recursion limit change to the local file using the macros (in fact, adding it to `lib.rs` didn't fix the issue), so while something that breaks a bit more nicely would be great it doesn't feel like as serious of an issue as something where the only workaround is modifying the upstream lib.

lschuermann, 2024-06-19:
For what its worth, we're pretty close to having a procedural-macro based rewrite of Tock registers (#4001), which should solve these issues. The plan is to do a stable major release of the current Tock registers before we switch to #4001.

If this is a significant problem for downstream users we may be able to alleviate this by integrating this PoC before performing this stable release, or as another minor release afterwards. However, increasing the recursion limit seems like a viable, no-effort alternative, esp. given it applies from the crate that actually contains the register definition. Given that fact, we should probably focus our efforts on #4001 instead.