tock/tock #2969: GNSS/serial device support

dcz-self, 2022-02-17:
Hi,

I'm playing with a GNSS device which is built into the board, communicating over UART. While working on it, I was wondering how to expose it to the user space. There are a couple considerations. First, arguments *for* a dedicated HIL:

1. The device will never need to switch UART parameters
2. Also, UART is not the only way to exchange messages
3. There are multiple protocols in common use (NMEA, UBX, etc)

so it makes sense to expose something more suited for GNSS devices. Or at least a buffer receiver that doesn't allow any reconfiguration, and doesn't expose the hardware interface.

Points against a GNSS HIL:
1. Parsing NMEA messages is not fun
2. The available protocols are commonly extended.

I think the best middle-ground would be to expose the GNSS device as a dumb, preconfigured pipe to the userspace. Does that make sense? Is that even in scope of the kernel to wrap such differences?

alistair23, 2022-02-17:
Interesting question!

First, I think the Tock kernel should not be parsing NMEA messages. The main reason for that is I'm not convinced that it's the job of a kernel to parse data like that. If we do it in the kernel we will need to write a parser capsule that provides the relevant data to userspace. Parses are also notoriously hard to get right and keeping it in the kernel if we don't have too seems like a bad idea.

If instead we parse them in userspace, we can re-use existing libraries (https://docs.rs/nmea-parser/0.9.0/nmea_parser/ for example) and then have the application do it. We could even use IPC to have one application parse the messages and send it to multiple others (like GPSD on Linux).

It still makes sense to have a GNSS capsule though. The capsule would just be a easy way to tell userspace that it's GNSS and expose the NMEA messages, which userspace then parses.

The only problem then is that generally you can send messages to the GNSS device to configure it. We don't really want multiple apps to be able to do that at the same time though. So a GNSS capsule might need to handle that, or we could just use syscall permissions to fix that.

alistair23, 2022-02-17:
> I think the best middle-ground would be to expose the GNSS device as a dumb, preconfigured pipe to the userspace

Should have read this part first :) I think that's the best bet as well

ppannuto, 2022-02-18:
FWIW, at least once one Tock platform supported GNSS and that was the approach it took: https://github.com/lab11/signpost-software/blob/f13b32e0661849024b4817ceb08f47fe980277ff/signpost/apps/libsignpost-tock/gps.c -- this was the motivation behind the UartAdvanced which listened for a logically cohesive string based on character timing; that way the kernel could pass up whole messages to the userspace driver.

Don't think this was necessarily the best solution, but it did work well at the time.

dcz-self, 2022-03-20:
I discovered that `hil::uart::UartData` exists, and it's a passable first approximation. That would only require settling on a device number for syscalls. I propose:

```
GNSS = 0x90004,
```

The way UartData works, there's a limitation in how it detects that a message is ready to submit. For accurate timing it would be a bad idea to have GNSS updates sit in the buffer for any approciable length of time, so timeouts (as in nrf52 uarte driver) are suboptimal. Therefore, the driver should ideally have some minimal knowledge about the protocols it handles. NMEA ends all messages with `\r\n` (that corresponds to the proposed `receive_until_terminator` thing), ubx has a length field in the middle (`receive_len_then_message`), etc. This would be necessary for a shim which makes sure that messages are retired in time, but it would still fit the API of UartData.

I think something on top of even the basic version is needed to identify the device to let users know what kind of data it produces/receives.

dcz-self, 2022-03-20:
I also had the little thought that if the driver behind the hil must know a bit about the protocol, then it might do things like parsing checksums and stuff, but that doesn't really help much if the user still has the task of actually parsing the payloads and doing something useful with them.