tock/tock #2981: Kernel ABI for 64-bit systems

jettr, 2022-02-28:
How do we want to handle the kernel/user ABI for 64-bit systems using Tock OS? This has come up while implementing a host emulated version of Tock OS and application on a 64-bit linux machine.

Most of the kernel/user ABI is defined using `usize` or `Register` nomenclature ([example][2]), but the newer [`SyscallReturnVariant`][0] enum specifically calls out `u32` variants instead of `usize` variants. Should, for example, `FailureU32` instead be `FailureUsize`?

Otherwise, when the kernel passes back architecture-sized data (e.g. pointer), the correct `SyscallReturnVariant` variant is difficult to chose in an platform independent manner.

Another option would be to add `usize` variants to `SyscallReturnVariant`, but then that gets confusing for 32-bit system to chose the correct return variant.

[TRS 104][1] specifically calls out that it only applies to "32-bit ARM Cortex-M and RISC-V RV32I", but the definitions in that TRD seem to have directly led to the `SyscallReturnVariant` enum, which exist in a platform independent location of the kernel.

Are there any other documents or discussions on this topic that I can gain some insight from to help forge a path forward?

[0]:https://github.com/tock/tock/blob/master/kernel/src/syscall.rs#L221
[1]:https://github.com/tock/tock/blob/master/doc/reference/trd104-syscalls.md
[2]:https://github.com/tock/tock/blob/master/kernel/src/syscall.rs#L159

lschuermann, 2022-02-28:
This has come up during the design of the Tock 2.0 system call interface.

> [`SyscallReturnVariant`](https://github.com/tock/tock/blob/master/kernel/src/syscall.rs#L221) enum specifically calls out `u32` variants instead of `usize` variants

This is deliberate choice we've made for the Tock 2.0 system call interface. While an `allow`'s pointer and length, as well as a `subscribe`'s callback pointer and user data (most likely a pointer) are clearly platform dependent and thus must be represented through the platform's native integer width, `command` has a different story. While capsules may in certain cases want to communicate types of the platform's native integer width, fundamentally the `command` interface is simply a mechanism to convey data from a userspace application to a capsule, and provide a return value as feedback of that operation. It is of major importance that the ABI provided by capsules does not implicitly change between platforms. While in practice transitioning from 32-bit to 64-bit is seldom going to cause issues, a capsule using native integer widths, written for a 64-bit system, can very likely not work on a 32-bit system.

This is reflected in the `SyscallReturn` enum already, by encoding `allow` and `subscribe` results in distinct variants (because while they _could_ be represented through regular Success and Failure variants are distinct in transporting native integer width data). `SyscallReturn` is then passed up to the `arch` crates which are ultimately responsible for implementing the register encoding (although they **may** call out fo `SyscallReturn::encode_syscall_return` as a helper method). This transition to fixed size integers as part of Tock 2.0 is not reflected everywhere yet, e.g. `Syscall::Command` should be changed to represent `driver_number`, `subdriver_number`, `arg0` and `arg1` as `u32`s.

This is not to say that we have decided to not support 64-bit systems. 64-bit systems would however need to have a different (extended) system call interface definition, which may or may not provide means of wider `command` arguments and return values. Very likely this should be represented by a different `Driver` trait, possibly with `Driver` as a supertrait.

> Should, for example, `FailureU32` instead be `FailureUsize`?

Following this reasoning, I would argue that `FailureUsize` is not required. A 64-bit system call interface can support all of the `SyscallReturn` variants we currently have, possibly a superset of those.

I hope this gives a bit of insight into why `syscall.rs` and the TRD104 is written in this way.

jettr, 2022-02-28:
Thank you for the quick response and explanation!

So how would a chip-independent capsule return a system-width value? I don't actually see a way to do this without using `cfg` to conditionally compile different branches in the capsule code based on the architecture width, e.g. `SuccessU32` for 32-bit and `SuccessU64` for 64-bit systems.

lschuermann, 2022-02-28:
> So how would a chip-independent capsule return a system-width value? I don't actually see a way to do this without using `cfg` to conditionally compile different branches in the capsule code based on the architecture width, e.g. `SuccessU32` for 32-bit and `SuccessU64` for 64-bit systems.

I think it's entirely reasonable to provide a method on `SyscallReturn` which takes `usize` and constructs the appropriate `SuccessU32` / `SuccessU64` and analog failure variants depending on the platform. This would make it an explicit mapping to different return variants based on the platform, which would still be a well-defined and portable ABI. The thing we want to avoid is implicitly adjusting the return value or parameter width depending on the target platform.

_However_, one could argue this violates our "one success and one failure return variant per command" policy. @phil-levis What do you think?

jettr, 2022-03-01:
> How about if core::mem::size_of::<usize>() == 8 { CommandReturn::success_u64(val as u64) } else { CommandReturn::success_u32(val as u32) } for the time being? I'm expecting this to be optimized away entirely.

We need this on both kernel and user side; this is one of the options we are entertaining for sure. I do hope that we can get more clarity on the long-term supported way to do this type of operation via this discussion/issue though.

hudson-ayers, 2022-03-01:
From https://github.com/tock/tock/commit/06ae51f238905472b82452c72e002e5dced00f8b: 
> Allowing arbitrary 32-bit variants to be added in the future maximizes the ABI's future flexibility. In particular, we *may* choose to extend the ABI to 64-bit architectures in a manner that adds future variants (e.g. "success with usize"). If we decided we do not want that flexibility in the future, we can further restrict the possible return variant discriminators without breaking compatibility with existing processes.

In general, https://github.com/tock/tock/pull/2402 is explicitly written to allow for adding something like `success_usize` to support 64 bit platforms. It seems like that might be the preferable route here.

phil-levis, 2022-03-04:
The idea behind the Tock approach is to model what other systems do and have distinct 32-bit and 64-bit ABIs. 

*I think it's entirely reasonable to provide a method on SyscallReturn which takes usize and constructs the appropriate SuccessU32 / SuccessU64 and analog failure variants depending on the platform.*

This could be OK, but my initial reaction is skeptical: I think it's going to end up with a lot of conditionals/matches with unclear error handling cases.

*The thing we want to avoid is implicitly adjusting the return value or parameter width depending on the target platform.*

Yes. Then you end up with checks on whether usize is a u32 or a u64.

*However, one could argue this violates our "one success and one failure return variant per command" policy.*

I guess my question is whether we ever expect a system to simultaneously support 32-bit and 64-bit ABIs. 

This all seems like a weird set of hoops to jump through in kernel architecture to support a host-emulation platform; can you just run in 32-bit mode?

lschuermann, 2022-03-05:
We did talk about this on the call yesterday, to provide a short (incomplete) summary:

- Tock 2.0's `allow` and `subscribe` have been designed to adjust to the target architecture's native integer width from the start. While they do currently map to a specific success and failure variant of the 32-bit ABI, a 64-bit ABI would just map them to an appropriate variant of it's return value layout definition.
- Tock 2.0's `command` specification actually defines that "[c]ommand calls should never pass pointers: those are passed with Allow calls, as they can adjust memory protection to allow the kernel to access them."
- Tock has been a 32-bit system from the start, although with viable RISC-V `rv64imac` platforms readily available it doesn't seem far fetched to potentially introduce a proper 64-bit ABI.
- Syscall return variants with a `usize` payload might make our current 32-bit ABI more portable, but less consistent. It seems cleaner to just provide a proper 64-bit ABI, if we have justification and eventually also hardware for that.

We decided to further discuss this with @jettr next week.

jettr, 2022-04-29:
After discussing more with Phil, we decided that maintaining the restriction that command calls do not deal in pointers is a good protection to keep in place. If we want to support specific pointer (i.e. memory related) usecases, then we can/should add new first-class syscall ID support for it. For example, the kernel owned buffer sharing, could/should introduce a `Request` syscall id that deals in pointers much like `Allow` currently does.

closed 2022-04-29 by jettr