tock/tock #2982: Cannot use this lib on the stable release channel

dawsonc623, 2022-03-01:
I am trying my hand at some embedded Rust development on a Raspberry Pi 4B, and one of my dependencies (`cortex-a`) depends on this library. My toolchain uses the stable release channel, and when trying to compile my project I get `error[E0554]: '#![feature]' may not be used on the stable release channel`. Is this by design, or is this library intended to work for devs using the stable release channel?

alexandruradovici, 2022-03-01:
Can you point us to the exact crate that you are using and the crate (library) that it depends on from this repository?

jrvanwhy, 2022-03-01:
> Can you point us to the exact crate that you are using and the crate (library) that it depends on from this repository?

I checked `cortex-a`'s `Cargo.toml`, and it depends on `tock-registers`. It may be easy to make `tock-registers` work on stable Rust (I might send a PR today).

jrvanwhy, 2022-03-01:
I edited the issue title to mention `tock-registers`, so that Tock developers understand at a glance which library this issue refers to.

The overarching issue for making Tock build on stable Rust is https://github.com/tock/tock/issues/1654.

dawsonc623, 2022-03-01:
Sorry, I should have dug a little deeper to give a better description (I did not realize this was a mono-repo and how big the scope of the project is). Thanks for making that clearer. I am not the most advanced Rust developer out there (especially in the embedded space), but if there is anything I can help with to get unblocked I am happy to try.

jrvanwhy, 2022-03-01:
PR opened: https://github.com/tock/tock/pull/2983

The specific unstable feature in use is likely to be stabilized soon, and the PR I opened has some drawbacks, so it's possible that we will not merge the PR.

jrvanwhy, 2022-03-01:
Making `tock-registers` work on stable Rust will not solve your issue on its own, because `cortex-a` depends on unstable Rust as well: https://github.com/rust-embedded/cortex-a/blob/fd6688419ac9884abd1d79a18b3ff64b4776e1f4/src/lib.rs#L82. It'll just move the compilation error from the `tock-registers` crate to the `cortex-a` crate.

dawsonc623, 2022-03-01:
Hmmm, okay. Well, thanks for your work on this all the same. I will open an issue with `cortex-a` next and see what they say.

brghena, 2022-03-01:
@dawsonc623 One thing to check, do you actually need to be on stable, or could you just move your toolchain to a nightly version of rust?

dawsonc623, 2022-03-02:
@brghena That is a good point. The stable channel is preferred since that is what the team has set up, but I do not think there is any policy holding us to stable right now. I will check into that; thanks for the suggestion!

jrvanwhy, 2022-03-02:
The consensus at #2983 was to wait for `feature(const_fn_trait_bound)` to be stabilized before moving `tock-registers` over to stable Rust. It sounds like moving to a nightly toolchain is a reasonable workaround in the meantime.

jrvanwhy, 2022-03-07:
The stabilization PR for `feature(const_fn_trait_bound)` was merged: https://github.com/rust-lang/rust/pull/93827#event-6197814397. Unless it is reverted, `feature(const_fn_trait_bound)` will be stable in 1.61.0.

qwandor, 2022-07-01:
Now that Rust 1.61 has gone to stable and #2988 is in, can we get a release of tock-registers that doesn't require nightly?

qwandor, 2022-08-19:
Hi, any update on this? Could a release be pushed that supports building on stable?

ppannuto, 2022-08-19:
Thanks for the nudge! Just opened #3151 — I'll push to crates.io once that's merged

closed 2022-08-24 by bors[bot]

ppannuto, 2022-08-24:
And we're live!

https://crates.io/crates/tock-registers

commit befe714a (2022-08-24): Merge #3151

3151: tock-registers: release v0.8 r=lschuermann a=ppannuto

### Pull Request Overview

Prepare a v0.8 release of tock-registers.

Fixes #2982.

---

`tock-registers` now supports stable Rust!

There is a small breaking change, documented below, required to support
Rust 2021 edition. Most of the remaining changes are improvements to the
internal self-testing infrastructure and documentation. There are also
some additions to `FieldValue` to improve ergonomics.

**Breaking Changes**

 - #2842: tock-registers: rename TryFromValue::try_from to try_from_value
 - #2838: Update to Rust 2 [... cut]
PR #3151 (2022-08-24): tock-registers: release v0.8
### Pull Request Overview

Prepare a v0.8 release of tock-registers.

Fixes #2982.

---

`tock-registers` now supports stable Rust!

There is a small breaking change, documented below, required to support
Rust 2021 edition. Most of the remaining changes are improvements to the
internal self-testing infrastructure and documentation. There are also
some additions to `FieldValue` to improve ergonomics.

**Breaking Changes**

 - #2842: tock-registers: rename TryFromValue::try_from to try_from_value
 - #2838: Update to Rust 2021 edition

Other Changes

 - #3126: [trivia [... cut]