tock/tock #2995: `static_init!` rejects generic params

dcz-self, 2022-03-15:
When trying to implement a capsule, I ran into this error here: https://github.com/dcz-self/tock/blob/056e4ee434783bca78973e825d3c39f2dd5c5e77/boards/components/src/block_storage.rs#L76 

```
   Compiling components v0.1.0 (/mnt/source/tock/boards/components)
error[E0401]: can't use generic parameters from outer function
  --> boards/components/src/block_storage.rs:76:57
   |
46 | impl<B, const W: u32, const E: u32> Component for BlockStorageComponent<B, W, E>
   |      - type parameter from outer function
...
76 |             block_storage_driver::BlockStorage<'static, B, W, E>,
   |                                                         ^ use of generic parameter from outer function
```

It seems that some scoping goes awry. The solution is to replace this:

```
        let syscall_driver = static_init!(
            block_storage_driver::BlockStorage<'static, B, W, E>,
            block_storage_driver::BlockStorage::new(
                self.device,
                self.board_kernel.create_grant(self.driver_num, &grant_cap),
                write_buffer,
            )
        );
```

with a similar call to `static_init_half!`:

```
        let u: MaybeUninit<block_storage_driver::BlockStorage<'static, B, W, E>> = MaybeUninit::uninit();
        let syscall_driver = static_init_half!(
            u,
            block_storage_driver::BlockStorage<'static, B, W, E>,
            block_storage_driver::BlockStorage::new(
                self.device,
                self.board_kernel.create_grant(self.driver_num, &grant_cap),
                write_buffer,
            )
        );
```

It's rather awkward, and confusing behaviour. I still don't know why the solution works, so maybe this should get documented or something.

hudson-ayers, 2022-03-26:
This is not really an issue with `static_init()`, but rather an issue with how Rust's monomorphization over generic types interacts with static global variables.

Here is a simple example from another component (that does not compile). Here I have replaced the call to `static_init!()` with what that macro would actually expand to.
```rust
impl<A: 'static + time::Alarm<'static>> Component for AlarmMuxComponent<A> {
    type StaticInput = &'static mut MaybeUninit<MuxAlarm<'static, A>>;
    type Output = &'static MuxAlarm<'static, A>;

    unsafe fn finalize(self, static_buffer: Self::StaticInput) -> Self::Output {
        //let mux_alarm = static_init!(MuxAlarm<'static, A>, MuxAlarm::new(self.alarm));
        static mut BUF: kernel::utilities::static_init::UninitializedBuffer<MuxAlarm<'static, A>> =
            kernel::utilities::static_init::UninitializedBuffer::new();
        let mut buf = kernel::utilities::static_init::StaticUninitializedBuffer::new(&mut BUF);
        let mux_alarm = buf.initialize(MuxAlarm::new(self.alarm));

        self.alarm.set_alarm_client(mux_alarm);
        mux_alarm
    }
}
```

The problem is that if you define BUF here based on the generic parameter, the expectation is that you would get a new instance of BUF for each different call to this function. In reality that is not the case, you would just get a new BUF for each different generic parameter, but if this function was called twice with the same generic parameter there would only be a single static variable. As such, Rust disallows this. This stack overflow answer provides a similar explanation: https://stackoverflow.com/a/52546611

hudson-ayers, 2022-03-26:
The `static_init_half!()` solution works because the expansion of that macro does not define a static variable

commit 84522e2a (2022-09-07): static_init: add note about generic parameters

Closes #2995.
closed 2022-09-07 by bors[bot]