tock/tock #2998: Process dump corrupted when using RTT

dcz-self, 2022-03-16:
I'm havig trouble seeing the process dump when connecting over the RTT console. It looks like multiple messages run into each other, leaving a coppupted amalgam in place. One version:

```
NRF52 HW INFO: Variant: AAC0, Part: N52840, Package: QI, Ram: K256, Flash: K1024
tock$ ───────────────────────── H

  R0 : 0xBAD00BAD    R6 : 0x00000000
  R1 : 0xBAD00BAD    R7 : 0x00000000
  R2 : 0xBAD00BAD    R8 : 0x00000000
  R3 : 0xBAD00BAD    R10: 0x00000000
  R4 : 0x0003006C    R11: 0x00000000
  R5 : 0x00000000    R12: 0xBAD00BAD
  R9�: ��──────
             │ Unused
  0x20006000 ┴───────────────────────────────────────────
             .....
  0x00032000 ┬─────────────────────────────────────────── F
             │ App Flash      8120                        L
  0x00030048 ┼─────────────────────────────────────────── A
             │ Protected        72                        S
  0x00030000 ┴─────────────────�bug, run `make lst` in the app's folder
and open the arch.0x30048.0x20006000.lst file.

```

What's notable here is that one of the messages I'm printing *just before* starting processes is missing too.

bradjc, 2022-03-16:
That code is fairly untested, and in general panic handling is difficult because it must happen synchronously when all other Tock code is written to be event-driven.

If I had to guess, the issue is here: https://github.com/tock/tock/blob/master/boards/nordic/nrf52840dk/src/io.rs#L83, if that delay isn't long enough in practice then it is not surprising it would overwrite data before it was completely sent.

dcz-self, 2022-03-24:
Indeed, the delay you pointed out helped alleviate the issue. I'm not closing because I don't know what the proper fix is.

hudson-ayers, 2022-03-26:
I think just increasing the length of that wait is an appropriate fix. I supposed ideally we would determine the minimum wait time long enough to print the longest possible panic message, but that feels like overkill. Even making the wait 10x longer will not make panic messages significantly slower in practice

ppannuto, 2022-08-06:
I think the right way to do this would be to wait for the attached JLink to drain the buffer, i.e. to replicate https://wiki.segger.com/RTT#SEGGER_RTT_GetAvailWriteSpace.28.29

One example of doing this:
https://github.com/bluekitchen/btstack/blob/87087689bb0c37ff4b9b3e3ba670b7019c306ebd/3rd-party/segger-rtt/SEGGER_RTT.c#L481
```c
static unsigned _GetAvailWriteSpace(SEGGER_RTT_BUFFER_UP* pRing) {
  unsigned RdOff;
  unsigned WrOff;
  unsigned r;
  //
  // Avoid warnings regarding volatile access order.  It's not a problem
  // in this case, but dampen compiler enthusiasm.
  //
  RdOff = pRing->RdOff;
  WrOff = pRing->WrOff;
  if (RdOff <= WrOff) {
    r = pRing->SizeOfBuffer - 1u - WrOff + RdOff;
  } else {
    r = RdOff - WrOff - 1u;
  }
  return r;
}
```

One risk/issue here is what happens if there's no debugger attached, i.e. I think this would hang waiting to print and not display the panic LED blink pattern, which would not be ideal.

commit aabd4679 (2023-08-18): boards: nrf52840dk: increase rtt wait

Increase the wait time to help panics.

Fixes: #2998
PR #3625 (2023-08-18): boards: nrf52840dk: increase RTT wait
### Pull Request Overview

Increase the wait time to help panics.

Fixes: #2998

### Testing Strategy

Didn't test, just figure we should do something with the issue.

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2023-08-18 by alevy