tock/tock #3005: Screen HIL brightness

dcz-self, 2022-03-31:
I am working on a screen driver, and I am somewhat confused about the brightness:

```
pub trait ScreenClient {
    /// Sets the display brightness and/or powers it off
    /// Screens must implement this function for at least two brightness values (in percent)
    ///     0 - power off,
    ///     otherwise - on, set brightness (if available)
    fn set_brightness(&self, brightness: usize) -> Result<(), ErrorCode>;

```

Does this turn on/off the entire display or just the backlight? I'm guessing it's just the backlight because there's no mention of initialization and the `screen_is_ready` callback, but I'd like to confirm.

Either way, flipping one means there's no knob to flip the other, which is somewhat suboptimal (even if I don't care yet).

alexandruradovici, 2022-04-04:
This was discussed a lot when designing the HIL. The general agreement was that setting the brightness to 0 would actually turn off the screen (https://github.com/tock/tock/pull/1837#discussion_r425440447). What that means is left up to the driver.

dcz-self, 2022-04-04:
Thanks for referencing that. That's a shortsighted solution though, because it doesn't work at all with transflective displays which, while they do have a backlight, can operate without it quite fine (and it's the main mode of operation for my case).

hudson-ayers, 2022-04-05:
IMO, that is a fine reason for us to accept a PR changing the HIL to add on()/off() functions. It is common for us to change new HILs as they are used by additional hardware with different requirements.

dcz-self, 2022-04-05:
I have proposed a HIL fixing that issue and a couple others: https://github.com/tock/tock/pull/3011 . It's a wallop though, so if it's preferred to do incremental HIL changes, please let me know and I'll split it up.

dcz-self, 2022-05-25:
This has been fixed in https://github.com/tock/tock/pull/3023

closed 2022-05-25 by dcz-self