tock/tock #3012: tock-registers: Allow setting registers by enum value

qwandor, 2022-04-05:
If I use the `register_bitfields` macro to declare a set of possible value for a register field, then I can use `Readable::read_as_enum` to get back a value of the appropriate `REG::FIELD::Value` enum type. However, unless I'm missing something, there doesn't seem to be any way to write such an enum value back into the field (i.e. to pass it to `write` or `modify`).

This could be solved by generating a `From` implementation for `FieldValue` alongside the generated enum type.

alistair23, 2022-04-06:
You can just use the enum directly, like this: https://github.com/tock/tock/blob/master/chips/earlgrey/src/aes.rs#L442

qwandor, 2022-04-06:
@alistair23 The code you linked doesn't seem to be using the enum type, only the constant `FieldValue` instances.

commit f4a1c734 (2022-04-15): Merge #3014

3014: tock-registers: Implement From field enum value type for FieldValue. r=brghena a=qwandor

### Pull Request Overview

This makes it easier to set a field based on the enum value.

Fixes #3012.

### Testing Strategy

This pull request was tested by `cargo test`.

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
commit 86a50dbb (2022-04-15): Merge #3014

3014: tock-registers: Implement From field enum value type for FieldValue. r=brghena a=qwandor

### Pull Request Overview

This makes it easier to set a field based on the enum value.

Fixes #3012.

### Testing Strategy

This pull request was tested by `cargo test`.

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
PR #3014 (2022-04-15): tock-registers: Implement From field enum value type for FieldValue.
### Pull Request Overview

This makes it easier to set a field based on the enum value.

Fixes #3012.

### Testing Strategy

This pull request was tested by `cargo test`.

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2022-04-15 by bors[bot]