tock/tock #3015: [question] Dose tock support posix Interface?

wjhtinger, 2022-04-07:
Many legacy software need posix, how to port them to tock?

alexandruradovici, 2022-04-07:
Tock does not intentionally support the POSIX standard as is based on a very different kernel / user space interface. On the other hand, the libtock-c implementation does provide `newlib` which in turn provides a subset of the libc API.

wjhtinger, 2022-04-12:
Thank your reply. 
Another quesiton，does Tock support theads in user space?  And, there is only one thead in kernel, right?

alexandruradovici, 2022-04-12:
No, it does not support threads. What you can do (if you need only one extra thread) is start two process and use IPC between them.

brghena, 2022-04-12:
And the kernel indeed only has one thread in Tock.

There are some basic thoughts on multi-core support, see #2873. Although no implementation yet.

lschuermann, 2022-08-05:
Closing this as the original question seems to have been resolved.

closed 2022-08-05 by lschuermann