tock/tock #3017: nrf52 Enhanced ShockBurst (ESB) support

Yohannfra, 2022-04-12:
Hello, is there any plan to support ESB for NRF52 chips ?

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_esb.html

Thanks you!

brghena, 2022-04-12:
There aren't any implementation plans for Enhanced ShockBurst right now. We are focused more on industry-standards like Bluetooth Low Energy and IEEE 802.15.4.

That being said, we do support many nRF52 boards and would be receptive to an ESB driver.

psyspy1, 2022-04-29:
@brghena Any updates on BLE?

brghena, 2022-04-29:
There's support for advertising (which we've had for quite a bit): https://github.com/tock/tock/blob/master/capsules/src/ble_advertising_driver.rs

I don't _think_ anything else is in progress right now. A few people have expressed interest, but I think they're busy with other tasks right now.

ppannuto, 2022-08-16:
We already have #981 to track BLE work. I don't think that ESB is going to be on the core team's roadmap any time soon, so I think it makes sense to close this issue. We are, of course, happy to take in PRs that add ESB support.

closed 2022-08-16 by ppannuto