tock/tock #3019: Tock-registers's test_fields macro fails in undocumented and non-obvious ways

jscatena88, 2022-04-14:
The test_fields macro seems to not be able to account for the size of a struct being influenced by the alignment constraints of a repr[C] struct; specifically the Rust docs say ["Finally, the size of the struct is the current offset rounded up to the nearest multiple of the struct's alignment."](https://doc.rust-lang.org/reference/type-layout.html). Take for example this code:
```
extern crate tock_registers;

use tock_registers::register_structs;
use tock_registers::registers::ReadWrite;

register_structs! {
    pub RegisterBlock {
        (0x00 => foo: ReadWrite<u64>),
        (0x08 => bar: ReadWrite<u32>),
        (0x0C => @END),
    }
}
fn main() {
    println!("{}", core::mem::size_of::<RegisterBlock>());
}
```

Running `cargo test` fails with the following error:
```
running 1 test
test test_register_structs::RegisterBlock::test_offsets ... FAILED

failures:

---- test_register_structs::RegisterBlock::test_offsets stdout ----
thread 'test_register_structs::RegisterBlock::test_offsets' panicked at 'Invalid size for struct RegisterBlock (expected 0xC but was 0x10)', src/main.rs:6:1
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

This is because the alignment of the `u64` element dictates that the size must be a multiple of 8. However even changing the `@END` marker to `0x10` as shown here:
```
extern crate tock_registers;

use tock_registers::register_structs;
use tock_registers::registers::ReadWrite;

register_structs! {
    pub RegisterBlock {
        (0x00 => foo: ReadWrite<u64>),
        (0x08 => bar: ReadWrite<u32>),
        (0x10 => @END),
    }
}
fn main() {
    println!("{}", core::mem::size_of::<RegisterBlock>());
}
```
Also generates an error, that is actually more confusing since it seems to indicate that the size is now smaller than before: 

```
running 1 test
test test_register_structs::RegisterBlock::test_offsets ... FAILED

failures:

---- test_register_structs::RegisterBlock::test_offsets stdout ----
thread 'test_register_structs::RegisterBlock::test_offsets' panicked at 'Invalid end offset for field bar (expected 0x10 but was 0xC)', src/main.rs:6:1
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

The only way to get the code to compile and pass test without error is add a "_reserved" memory location that fills out the rest of the alignment padding:
```
extern crate tock_registers;

use tock_registers::register_structs;
use tock_registers::registers::ReadWrite;

register_structs! {
    pub RegisterBlock {
        (0x00 => foo: ReadWrite<u64>),
        (0x08 => bar: ReadWrite<u32>),
        (0x0C => _reserved0),
        (0x10 => @END),
    }
}
fn main() {
    println!("{}", core::mem::size_of::<RegisterBlock>());
}
```

Ideally a developer shouldn't have to be aware of the alignment of every element of their struct, in order to find the max alignment and pad the struct manually with a reserved location in order to have the code pass builtin tests

brghena, 2022-04-15:
Thanks for description of what's going on!

How did you end up running into this? Is there some hardware out there that mixes register sizes?

jscatena88, 2022-04-15:
The register spec of our peripheral uses 32 bit registers, however in some cases in order to represent a 64 bit value/address the spec lists two adjacent registers as "lower" then "upper" of the same value. We were using a u64 at the offset of "lower" to capture both

brghena, 2022-04-15:
Yup, that's definitely a reasonable use case that we should be accommodating. And I agree that users shouldn't have to add their own `reserved` fields, although it's a reasonable workaround for now.

closed 2022-07-21 by bors[bot]