tock/tock #3021: TicKV: Add it to crates.io

riskable, 2022-04-15:
I'm about to try out TicKV in my project and noticed there's no crate for it at crates.io.  It looks like it's "finalized" so it would be nice if we could just put it in our `Cargo.toml` as a simple dependency instead of having to give it the git repo/path.

hudson-ayers, 2022-04-21:
@alistair23 thoughts on this?

riskable, 2022-04-22:
FYI: I ended up posting my TicKV project: https://github.com/riskable/rp2040-tickv

It uses the leftover space at the end of the RP2040's flash chip for TicKV storage (which is fantastic!).  Because TicKV isn't in crates.io I had to use this github repo directly in the `Cargo.toml`.  It would feel a lot better if I could just use `tickv = "0.1"` (well, whatever version is the latest) in the `[dependencies]` section though.

alistair23, 2022-04-27:
Cool! I'm glad to see TicKV being used in other projects.

The plan is to add TicKV to crates.io. I just wanted to make sure the library was reasonably stable before a `0.1` release. If you have been using it on hardware and can confirm it's stable that seems like a good time to release a `0.1`

riskable, 2022-05-02:
@alistair23 Yes I can confirm that it works on the RP2040.  I'm not using it in a production release yet but it has been working great so far in my testing 👍

alistair23, 2022-05-04:
Great! I have sent a PR to Tock to prepare for pushing to crates.io. It would be great if you can test it: https://github.com/tock/tock/pull/3034

Note that it won't work with any current data

alistair23, 2022-05-22:
Done! https://crates.io/crates/tickv

closed 2022-05-22 by alistair23