tock/tock #3022: TicKV: Typo? initalise() should be initialize() (or just init())

riskable, 2022-04-16:
There appears to be a typo in TicKV:

```rust
pub fn initalise(&self, hashed_main_key: u64) -> Result<SuccessCode, ErrorCode> {
...
```

Unless of course you meant it to be that way (keep everyone on their toes!) I recommend `init()` instead :smile:

lschuermann, 2022-04-16:
I wouldn't go as far as to say the British (and Austrailian, for that matter) spelling is a typo, but I agree it's good to be as consistent as possible within our own interfaces.

```
> git grep "initialize" | wc -l
568
> git grep "initialise" | wc -l
10
```

It seems that the American spelling is far more prevalent in our overall code base, which makes sense given Tock's origins. However, I believe TicKV's developer @alistair23 only vendored this library into the Tock repository because of policies regarding use of external crates. Thus, as far as I am concerned, they are at their liberty to keep the interfaces, and in particular spelling, how they wish.

For these other occurences, though, subjectively one can make the argument to rename them. I think this is a valuable discussion to have, as for APIs it seems unified spelling does matter.

riskable, 2022-04-16:
@lschuermann Actually, it has nothing to do with the S or the Z but the fact that it's missing an "i": **init**-*alise*

lschuermann, 2022-04-16:
> Actually, it has nothing to do with the S or the Z but the fact that it's missing an "i": **init**-_alise_

Yikes, sorry, I missed that! You proposing `initialize` and suggesting it might've been meant to be that way mislead me a little. :smile: Yeah, we should probably fix that. Nonetheless, I think my argument still holds in the discussion of whether we should prefer `initialize` (with a Z), or of course we can go the elegant route and use `init`.

riskable, 2022-04-17:
Yeah the Z or S is irrelevant (I type the Z automatically, hehe) but the missing "i" is seriously annoying...  I wrote a driver that lets you use TicKV with the RP2040 (https://github.com/riskable/rp2040-tickv) and it really threw me for a loop--trying to figure out why the compiler was complaining about my call to `initialise()` not being a function that exists :smile:

alistair23, 2022-04-19:
Just created a PR: https://github.com/tock/tock/pull/3026

closed 2022-04-27 by bors[bot]