tock/tock #3024: DynamicDeferredCall multiple register()

dcz-self, 2022-04-17:
Looking at DynamicDeferredCall, I am getting a little uncertain about getting Handles. The DynamicDeferredCallClient has `fn call(&self, handle: DeferredCallHandle);`, which suggests that it can dstinguish between multiple handles (cool), but then the handles come from `register(Client)`. I'm perhaps wrong about this, but `register(Client)` gives me vibes of idempotency: registering to a client multiple times wouldn't make a difference if it was SPI or something of the sort.

But in this case, registering multiple times actually makes sense: having two deferred call handles, each for a different purpose, is useful. A skim of the code doesn't show idempotency here either.

Perhaps this should be explicitly documented: "you may register many times, and you shall receive a new handle each time". Or, to minimize surprise, rename to `create_handle_for(client)`.

Or am I mistaken somewhere?

alexandruradovici, 2022-04-17:
If I'm not mistaking and looking at how the `register` function is implemented, I think you will get a different (incremented) handle each time you it. I think it would be a good idea to add this to the documentation.

However I think one of its authors might chip in to this.

lschuermann, 2022-05-07:
Original author here. 

> But in this case, registering multiple times actually makes sense: having two deferred call handles, each for a different purpose, is useful. A skim of the code doesn't show idempotency here either.

Indeed this is the case and its been deliberately built this way. It's pretty easy to conceive scenarios in which a component would like to schedule a deferred call in response to different events and handle them separately. There was a tradeoff to make between the following options:

1. Maintain the state about intentions for scheduling deferred calls in the client itself. This seems like a rather messy solution. Although the client may sometimes be able to infer the original scheduling intention of a deferred call purely based on its state, forcing clients to do this seemed like it has the potential to result in buggy, complex and duplicate code quickly.
2. Add additional context to scheduled deferred calls. This would incur a cost for each and every registered client, as we'd have to add state to the `DynamicDeferredCallClientState` structure.
3. Allow registering a client multiple times, with different handles.

In the end, option (3) seems to be the most reasonable tradeoff between resource requirements (consumes a `DynamicDeferredCallClientState` for each handle) and client complexity.

> Perhaps this should be explicitly documented: "you may register many times, and you shall receive a new handle each time". Or, to minimize surprise, rename to `create_handle_for(client)`.

I agree the documentation can be improved here. I will create a PR immediately.

lschuermann, 2022-05-07:
Also, I'll make the `DeferredCallHandle: Eq` so that clients can actually distinguish deferred calls.

commit c9177292 (2022-05-09): Merge #3038

3038: kernel/dynamic_deferred_call: document multi-handle client behavior r=lschuermann a=lschuermann

### Pull Request Overview

This documents that it is possible for a client reference to be registered with the dynamic deferred call mechanism multiple times, retrieving a different handle each time. It furthermore derives `Eq` and `PartialEq` for the `DeferredCallHandle`, for clients to distinguish scheduled deferred calls.

Fixes #3024.

Suggested-by: `@dcz` <gihu.dcz@porcupinefactory.org>

### Testing Strategy

This pull request was tested by CI.

### TODO or Help Wanted

N/A
 [... cut]
commit f4387b65 (2022-05-09): kernel/dynamic_deferred_call: document multi-handle client behavior

This documents that it is possible for a client reference to be
registered with the dynamic deferred call mechanism multiple times,
retrieving a different handle each time. It furthermore derives `Eq`
and `PartialEq` for the `DeferredCallHandle`, for clients to
distinguish scheduled deferred calls.

Fixes #3024.

Suggested-by: dcz <gihu.dcz@porcupinefactory.org>
PR #3038 (2022-05-09): kernel/dynamic_deferred_call: document multi-handle client behavior
### Pull Request Overview

This documents that it is possible for a client reference to be registered with the dynamic deferred call mechanism multiple times, retrieving a different handle each time. It furthermore derives `Eq` and `PartialEq` for the `DeferredCallHandle`, for clients to distinguish scheduled deferred calls.

Fixes #3024.

Suggested-by: @dcz <gihu.dcz@porcupinefactory.org>

### Testing Strategy

This pull request was tested by CI.

### TODO or Help Wanted

N/A

### Documentation Updated

- [x] ~Updated the rel [... cut]
closed 2022-05-09 by bors[bot]