tock/tock #3028: Slightly confusing description of rust in the readme

nico-abram, 2022-04-22:
The readme states:
> are written in Rust, a systems programming language that provides compile-time memory safety, type safety and strict aliasing

I find this use of "strict aliasing" confusing. Rust has "strict" aliasing rules w.r.t &/&mut but isn't this is easy to misunderstand since in gcc -fno-strict-aliasing disables Type-Based Alias Analysis, and "Strict Aliasing" is somewhat overloaded as a term. But rust has no type-based aliasing rules (A `*mut T` and `*mut U` can alias. The only restrictions come from reference types (Or Box, or the unstable Unique<T>), but as far as I know those are closer to using C's `restrict` than type-based aliasing), rust memory is "untyped"

lschuermann, 2022-04-22:
According to the [Rust Unsafe Code Guidelines Reference Glossary](https://rust-lang.github.io/unsafe-code-guidelines/glossary.html), _aliasing_ is defined as to occur "when one pointer or reference points to a "span" of memory that overlaps with the span of another pointer or reference." I think this is the definition we're implying in that statement.

> The only restrictions come from reference types

Indeed, the only case where aliasing as defined above is restricted in terms of soundness is with references. Perhaps we can rephrase to "[...] and strict reference aliasing rules."?

jrvanwhy, 2022-04-22:
"strict aliasing" is a well-known term in the C/C++ world, but is not present in the Rust language.

I think we can shorten the sentence to:

> First, the kernel and device drivers are written in Rust, a systems programming language that provides compile-time memory safety and type safety.

From this perspective, Rust's aliasing rules aren't a feature -- they are a tool Rust uses to achieve its "memory safety" feature.

commit 4a28b353 (2022-07-20): Merge #3074

3074: readme: update to clarify rust, formatting r=alevy a=bradjc

### Pull Request Overview

Fixes #3028 by being more clear about what Rust provides.

Updates formatting.

### Testing Strategy

n/a

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
PR #3074 (2022-07-20): readme: update to clarify rust, formatting
### Pull Request Overview

Fixes #3028 by being more clear about what Rust provides.

Updates formatting.

### Testing Strategy

n/a

### TODO or Help Wanted

n/a

### Documentation Updated

- [x] Updated the relevant files in `/docs`, or no updates are required.

### Formatting

- [x] Ran `make prepush`.
closed 2022-07-20 by bors[bot]