tock/tock #3030: Issue when building utilizing arch/cortex-m4/src/mpu.rs

KeepSummerSaf3, 2022-04-26:
The following error occurs when using make. Seems to be an issue using relative path here. Replacing/copying mpu.rs from cortex-m3 to cortex-m4, effectively eliminating the reference, is one method. Maybe this method is invalid for cmake?

```
error: expected item, found `..`
 --> arch/cortex-m4/src/mpu.rs:1:1
  |
1 | ../../cortex-m3/src/mpu.rs
  | ^^ expected item

error: aborting due to previous error

error: could not compile `cortexm4`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

KeepSummerSaf3, 2022-04-26:
yes indeed a fumble. On second look you are correct on this not being part of the repo; will close this issue out for now. Sorry for the mistake.

closed 2022-04-26 by KeepSummerSaf3

alexandruradovici, 2022-04-26:
I had this issue when trying to compile while being stored on a Windows drive (via Windows Subsystem for Linux). Symlinks seems to be an issue.