tock/tock #3035: Snync Screen syscalls

dcz-self, 2022-05-04:
I'm trying to implement a syscall driver for the Screen interface, and I'm seeing that all of the synchronous HIL calls have been turned async:

> ### Command number: `23` 
>
>    **Description**: Get the screen's resolution
>
>    **Argument 1**: unused
>
>  **Argument 2**: unused
>
>    **Returns**: Ok(()) followed by a callback with the resolution, BUSY if another command is in progress.

Is this actually desired? The HIL already says:

>     /// This function is synchronous as the driver should know this value without
>    /// requesting it from the screen.
>    fn get_resolution(&self) -> (usize, usize);

so it can be called synchronously even when some operation is in progress, and even if another client is currently performing an operation.

I'm just itching to make the syscalls synchronous. Should I have a go at it? Or is there a good reason for them to stay async?

hudson-ayers, 2022-05-05:
@alexandruradovici

alexandruradovici, 2022-05-05:
While the HIL's function is synchronous, the screen syscall driver services several applications at once. If one application has issued a command to the driver and the driver is busy processing it, another application will have to wait for the first application's request to finish. This is why the syscall command is asynchronous.

While you can make this particular syscall synchronous, image the following race:
- app1 requests a resolution change
- app2 asks for the current resolution and gets the "old" one
- app1 requests is done and now we have a new resolution (different from the one app2 received) 

Any suggestions are welcome though :smile:

dcz-self, 2022-05-05:
Thanks for answering. I see this as an example of a wider problem though:

1. Client A requests current resolution and gets Res1
2. Client B requests resolution change to Res2
3. Now Client A has an outdated resolution

The ordering doesn't matter, because changes need to be broadcast. And if we go in that direction, then suddenly we have to broadcast all state changes.

While this may be useful in some use cases, I think it's way overkill, and I'd rather make use of a simple driver that doesn't try to multiplex. Most people are going to use only a single screen client in the userspace anyway, including me.

So I don't really see much value in multiplexing, given its complexity. But I'll happily improve the single client situation. Should I go ahead?

alexandruradovici, 2022-05-05:
I see your point. While you still have to multiplex some requests (write, etc), I think the resolution read can be synchronous. I would say go ahead and modify it.

dcz-self, 2022-05-05:
Yes, I don't mean to throw away all multiplexing (although I consider locking the entire screen for one process a valid solution), just keep it on a best-effort basis.

dcz-self, 2022-07-09:
Should we close it now that #3047 is in?

alexandruradovici, 2022-07-10:
I think so.

closed 2022-07-11 by hudson-ayers