tock/tock #3040: What is the best way to hadnle a downstream version of Tock

alexandruradovici, 2022-05-10:
I am trying to build a downstream version of Tock and I am wondering what would be the best way to organize such a repository.

The main requirements that I am starting from are:
- Updates to upstream Tock should be easy to integrate (rebase might not be the best idea)
- There is a need to write additional crates (with chips and capsules)
- Boards need modifications to support additional peripherals

So far, the best idea that I came up with is the structure designed in the template https://github.com/WyliodrinEmbeddedIoT/tock-project. It includes Tock as a submodule, copies and modifies the board crates and adds a drivers crate (as there already is a crate named capsules). The issue that I am faceing is that Tock provides a workspace, so a new top level workspace seems not possible unless I delete the Crago.toml file from Tock.

As there are several authors here that manage downstream versions of Tock, any idea would be really appreciated. Based on these, I could write a tutorial explaining how to do this.

hudson-ayers, 2022-05-10:
What you are describing isn't really a "downstream version of Tock" (which I would consider as modifying the core kernel or core capsules) but rather an out-of-tree board/chip, and possibly some out of tree capsules.

A couple public examples of the same approach:
1. https://github.com/google/tock-on-titan , uses tock as a submodule while providing additional board / chip crates in the downstream repo
2. https://github.com/google/opensk uses tock as a submodule, defines additional boards but just uses the upstream chips. also applies a couple of git patches to the tock kernel / libtock-rs if I recall correctly

I know we had talked before about wanting it to be possible to not use Tock as a submodule and instead directly depend on the crates in the Tock repository from the Cargo.toml file (using git dependencies since not all of the crates are on crates.io), I don't see any reason that would not be possible for what you describe. Then you could have a top level workspace Cargo.toml in your repo and the workspace in the Tock repo should not affect you, as I understand it.

alexandruradovici, 2022-05-11:
I used the same approach as the two projects you describe. The issue that I faced was creating a top level workspace, as this means having nested workspaces (which cargo does not seem to like). For instance rust-analyzer fails if there is no top level workspace.

Thank you for the idea of having creates pulled directly out of git. I was under the impression that you can only pull a create out of git if it is in the root of the repository. I will try pulling them separately.

lschuermann, 2022-06-05:
> Thank you for the idea of having creates pulled directly out of git. I was under the impression that you can only pull a create out of git if it is in the root of the repository. I will try pulling them separately.

This is what I would've suggested as well. I've already experimented with that concept once and it seemed to work quite well at the time. Closing this given this seems like the canonical approach to maintain an out-of-tree board, if there's any issues feel free to reopen.

closed 2022-06-05 by lschuermann