tock/tock #3050: Have a way to *not* upgrade warnings to errors in CI

dcz-self, 2022-05-31:
I am currently maintaining a branch of Tock, which I intend to experiment on, and gradually upstream. With the amount of experimentation I did, I needed to take some technical debt, and my build has plenty of warnings.

The problem is with Tock upgrading warnings to failures behind my back, breaking the principle of least surprise.

When I run `make` on my laptop, it succeeds with warnings. When I run `make` in my local CI checker, it succeeds with warnings. When I run `make` in Gitlab CI, it suddenly changes its mind and fails: https://framagit.org/jazda/tock/-/jobs/1692651

I don't think this is good behaviour: the build system should not guess the intention of the person running it, because it will inevitable fail (as exemplified).

Moreover, there doesn't seem to be a way to override the mistaken detection. I tried to find when the detection happens, but my Makefile skills are not sufficient for that.

Could the automatic detection be replaced by an explicit setting of CI=true *outside* of the makefiles, so that it's controllable directly by the caller?

dcz-self, 2022-05-31:
Yes, but for me, the logic is wrong for me. I did remove that part in my downstream, but I don't like increasing the diff.

The flawed guessing for what is "CI=true" and what isn't is something I can't find.

dcz-self, 2022-05-31:
Oh, now I see, that's picked up from the runner environment. That makes things complicated, because the user no longer actually decides how things are being run, with runners typically being set up by someone else.

Working around users' bad infrastructure is not something projects should do, so I'll just suggest unhardcoding the assumption that CI means errors must be upgraded for everyone. That'll make downstream development easier.

dcz-self, 2022-05-31:
Yes, be sensitive to a custom environment variable: "DISABLE_UPGRADE_WARNINGS". The "disable" means you don't have to change your CI configuration, so it would stand out in a contribution. Meanwhile, anyone who wants to make use of such a flag can add the variable in their favorite way (gitlab allows setting variables out-of-band, meaning I would be able to do it without altering code).

That's a middle-of-the-way position, because running the same command will give different results depending on the environment that might not be obviously different. A more total solution would be to introduce an environment variable "UPGRADE_WARNINGS" instead. It could be explicitly present in CI pipelines (stopping a mistaken change needs a reviewer to catch it), or implicitly in CI runner config (only admin can switch it).

hudson-ayers, 2022-06-02:
Ok, I understand a little better what is happening now. Both github and gitlab automatically set an environment variable `CI=true` when running in the CI environment, so even though you are not running our predefined ci jobs you still are getting warnings treated as errors. This was also the case for Travis CI, our old CI platform.

I am not sure whether we are relying on this behavior for any other CI commands (i.e. are there any CI commands for which we do not currently manually pass `CI=true`, but for which warnings being treated as errors is important?). If not, the easiest fix here is probably just to rename our `CI` variable to something else (i.e. `TREAT_AS_CI` or similar).

commit 5990dff1 (2022-07-15): Merge #3075

3075: build: change `$CI` to `$NOWARNINGS` r=phil-levis a=bradjc

Tock uses different flags in CI so that warnings throw errors, while during normal development warnings are OK. We used the CI env variable to control this, however, this conflates Tock's policy on warnings in CI with running in _any_ CI. That is, a fork of Tock also ends up using Tock's CI policy on warnings, because CI tools tend to set the $CI variable as well.

This patch renames our variable to be more clear of what our use is. In theory we can dissallow warnings while not being in CI.

Fixes #3050.

### Testin [... cut]
PR #3075 (2022-07-15): build: change `$CI` to `$NOWARNINGS`
Tock uses different flags in CI so that warnings throw errors, while during normal development warnings are OK. We used the CI env variable to control this, however, this conflates Tock's policy on warnings in CI with running in _any_ CI. That is, a fork of Tock also ends up using Tock's CI policy on warnings, because CI tools tend to set the $CI variable as well.

This patch renames our variable to be more clear of what our use is. In theory we can dissallow warnings while not being in CI.

Fixes #3050.

### Testing Strategy

travis

### TODO or Help Wante [... cut]
closed 2022-07-15 by bors[bot]