tock/tock #3081: TicKV panics at certain value sizes

eflukx, 2022-07-12:
Hi I'm using (ok.. _planning_ to use) TickV in my application, looks like a great little tool! I use it apart from Tock, using nrf-hal (and RTIC). I wrote a simple implementation of `FlashController` for the nrf-hal `nvmc` peripheral and things mostly seem to work fine!

While testing the newly found capabilities of my app I ran into the following:

- Writes of values length 1-4067 bytes work fine
- Writes of values length >=4070 bytes result in an (probably expected) error code
- **Writes of values length 4068-4069 bytes result in a panic** 

_Note that I have been testing this with a single key, on a clean (empty, initialized) TicKV store._

The panic seems to have its [origins at this source line](https://github.com/tock/tock/blob/99cf1d32ce05dfd96a98e8f6989c162e753bba57/libraries/tickv/src/tickv.rs#L496) resulting in an out-of-bounds indexing operation.
Probably the simplest (and sane) solution is to tighten the `object_length` check in `append_key()` (?)

alistair23, 2022-07-12:
Hey! I'm glad to hear you find TicKV useful!

A panic is bad, so that needs to be fixed. It would be great if you could isolate a test case. If you could share a test case then we can fix the issue and ensure it doesn't occur again.

I'm guessing the value plus the header data is larger then the memory region, which is causing the panic. We should probably change the array access to use `get()`, that way we can remove a panic

bootstrap-prime, 2022-07-27:
Do you happen to have your TickV nrf-hal impl for FlashController publically available? I'm unable to implement it because the trait requires &self instead of allowing &mut self... are you using interior mutability?

eflukx, 2022-07-28:
> Do you happen to have your TickV nrf-hal impl for FlashController publically available? I'm unable to implement it because the trait requires &self instead of allowing &mut self... are you using interior mutability?

Yes, I do use interior mutability indeed.. :) I've posted [a gist with my current interface](https://gist.github.com/eflukx/8d85fad415b8fdc4d86dc4f382d8de7b). Probably useful for someone.

(I included the check-if-pager-empty before actually erasing the flash page, as was done in the nrf-hal nvmc example code, but I don't know if that's actually useful, as TickV probably very seldomly erases already-erased pages)

eflukx, 2022-07-28:
@bootstrap-prime if you could test if you get the same the panicing-behavior i got using my implementation that would be great.

Oh.. btw, you probably got this from the gist already, but this is how I initialize my TickV struct (also includes some other workaround (delay), without it my code trashes defmt-rtt output when called immediately after startup, did not investigate that further yet):

```rust
const NUM_PAGES: usize = 6; // Should be <= CONFIG_DATA_PAGES in memory.x

extern "C" {
    #[link_name = "_config"]
    static mut CONFIG: [u8; (NUM_PAGES * FLASH_PAGE_SIZE)];
}

// To run this example:
// cargo run --release --example nvmc_example
// or
// probe-run.exe --chip nRF52833_xxAA target/thumbv7em-none-eabihf/release/examples/_nvmc_example

#[cortex_m_rt::entry]
fn main() -> ! {
    let peripherals = bsp::hal::pac::Peripherals::take().unwrap();
    let nvmc = Nvmc::new(peripherals.NVMC, unsafe { &mut CONFIG });

    // Corrupted output when immediately erasing a page after start
    // Could this be a bug in the HAL or defmt or maybe just a HW quirk?
    cortex_m::asm::delay(450_000);

    let mut tickv_buffer = [0u8; FLASH_PAGE_SIZE];
    let tickv = NvmcFlash::tickv(nvmc, &mut tickv_buffer, NUM_PAGES);

// ... code cut here...
}
```

alistair23, 2022-08-03:
I just opened https://github.com/tock/tock/issues/3081 that should fix the failure for you

closed 2022-08-05 by bors[bot]