tock/tock #3101: Getting Verilator working on Mac

bradjc, 2022-07-22:
Goal: make verilator support cross platform and work on Mac.

Work in progress: https://github.com/tock/tock/pull/2975

lschuermann, 2022-07-23:
Would getting it to work in Docker (at least the fully CI workflow) qualify as "working on Mac"? I think that could relief some of the struggles people have with reconstructing CI results even on Linux, but more generally also on Windows and macOS.

ppannuto, 2022-07-24:
I think for the purposes of the "heavyweight" things, yes. This is more about ensuring a principle that any developer can replicate any CI failure locally; Docker is admittedly non-zero lift, but so is any realistic install of Verilator.