tock/tock #3103: Support for Wired Ethernet (IEEE 802.3)

ntninja, 2022-07-26:
From my studying of the source code so far there does not appear to be support for any wired ethernet adapters right now, as all networking examples and capsules presume the usage 6LoWPAN wireless networking instead. However, the existing network code has been carefully structured in a way that it does not actually require the presence of wireless components as such. (For instance, example [libtock-c/examples/tests/udp/udp_send/](https://github.com/tock/libtock-c/tree/master/examples/tests/udp/udp_send/) would happily send its “Hello World" packets to its hard-coded destination address, irrespective of the carrier of the source IPv6 address even though its README talks about IEEE 802.15.4 network anyways.)

So with that out of the way, how would one best implement support for a wired ethernet NIC like the [ENC28J60](https://ww1.microchip.com/downloads/en/DeviceDoc/39662a.pdf)? Based on reading the development section of the book I conclude that this would be done by implementing a capsule ([the book’s section talks about “sensors”](https://book.tockos.org/development/sensor.html), but its obviously the same) implementing some HIL. However the only HIL that comes close [is the “radio” one](https://github.com/tock/tock/blob/224bad4a6fb7cb757c06699b8e598bdfabf91dba/doc/reference/trd-radio.md). Supposedly, the trivial reason for that is that all networking devices so far have been IEEE 802.15.4 wireless devices, so nobody bothered to separate out the layers yet. (Correct?) Some digging through the source code also reveals that the `IP6SendStruct` in [tock/capsules/src/net/ipv6/ipv6_send.rs](https://github.com/tock/tock/blob/224bad4a6fb7cb757c06699b8e598bdfabf91dba/capsules/src/net/ipv6/ipv6_send.rs) is the first layer to hard-code the current *networking → 6LoWPAN* situation. And it appears as though, an alternate implementation passing the packets up to a certain maximum size (>=1280) as-is to a simplified MAC layer would already be pretty close to what is needed on the device-independent kernel code side.

hudson-ayers, 2022-07-26:
@lschuermann has an out-of-tree Ethernet stack that he has been planning on contributing for awhile, I believe

ntninja, 2022-07-26:
Thanks! I found this while digging through their stuff looking for what you mentioned: https://git.currently.online/leons/tock.git/log/?h=dev/enc28j60

Wouldn’t call that an “Ethernet stack”, but it is a driver/capsule for the chipset in question (of unknown quality of course, but at least it has all the basic steps spelled out already).

@lschuermann: Given that its from 2020 is it safe to assume that this abandoned until further notice? Or is there related updated code somewhere?

lschuermann, 2022-07-26:
Oh wow, what a coincidence this issue pops up just now, perfect timing! I have actually just PRed one implementation of an Ethernet driver (VirtIO `NetworkCard` device) here: #3110. Furthermore, there is an Ethernet driver for the LiteX boards [already in tree](https://github.com/tock/tock/blob/master/chips/litex/src/liteeth.rs).

Generally, in a recent meeting it seems that we have found Ethernet (and more generally IP) to be a viable communication stack for Tock, in parallel to extending our Bluetooth LE capabilities and introducing perhaps more stacks such as LoRa.

I don't have everything pushed to my private repos unfortunately, but yes, the ENC28J60 support also was a project I've been working on and I had IPv6 ICMP Echo & Neighbor Discovery working in kernel with that (although highly unreliable, perhaps because of running the SPI interface over long jumper cables). That could be picked up again. Furthermore, I have an i.MX RT1060 EVK lying around which I'd try to port next given that Tock already has basic support for this chip.

In terms of an actual network stack, our short term plan was to first of all get Ethernet working in the kernel and define a HIL once we have a few Ethernet chips supported (probably ~3), such that we can this crucial component right, which will also determine how buffers get passed around in the kernel. Then, we'd introduce a "TAP" driver for delivering raw Ethernet frames to userspace and using LwIP to establish basic IP capabilities. We'd like to build up this stack carefully and step by step, rather than trying to get it right in one shot. This may, for some time in the future, involve handling some parts of this stack in userspace, while others are in the kernel. I have a basic TAP driver and LwIP web server working since over a year now, actually, and will try to get them PRed shortly.

I'm really happy to see others being interested in this area as well and plan to spend a more significant portion of my time on this particular area in Tock. Once we get to the basic userspace support stage, things are going to get more interesting and I'll draft up a design document. @ntninja Do you have a specific chip you'd be interested to see supported? While I think we should wait a little more to work in the proper IP stack, getting more chips and Ethernet MACs supported for that initial HIL would be great.

ntninja, 2022-07-26:
> Oh wow, what a coincidence this issue pops up just now, perfect timing! I have actually just PRed one implementation of an Ethernet driver (VirtIO NetworkCard device) here: [...]

Cool. :slightly_smiling_face: 
I had actually only checked the capsule directory for Ethernet drivers, not chips. Hence, I didn’t notice the LiteEth one at all until you just pointed it out.

> Generally, in a recent meeting it seems that we have found Ethernet (and more generally IP) to be a viable communication stack for Tock, in parallel to extending our Bluetooth LE capabilities and introducing perhaps more stacks such as LoRa.

Obviously I agree – in particular also for IoT use-cases where devices don’t need to move post-installation going wired using Ethernet removes a lot of attack surface while still being able to use the spades of off-the-shelf equipment available that other wired options don’t tend to have.

> I don't have everything pushed to my private repos unfortunately, but yes, the ENC28J60 support also was a project I've been working on and I had IPv6 ICMP Echo & Neighbor Discovery working in [...]

Good to know its current state, I’ll keep that in mind if I manage to make it work on current TockOS.

> Do you have a specific chip you'd be interested to see supported? 

The good ol’ ENC28J60 chip exactly. Have several of these lying around here and they should do a decent job – just wish they also supported EEE; not aware of any “maker-friendly” (ie: pre-soldered to breakout board) chips being available that do, though.

lschuermann, 2022-07-28:
> The good ol’ ENC28J60 chip exactly. Have several of these lying around here and they should do a decent job

While I agree that the ENC28J60 is an interesting target because it is so low cost and you can attach it to virtually every board we already support, it also has some major issues, the most prevalent being:
- the serial interface it exposes requires some comparatively complex state machines on the host side (Tock), other chip-integrated MACs are much simpler to use,
- getting it to run at a reasonable speed requires SPI clocks upwards of 100kHz, which is where typical breadboard setups might already encounter signal integrity issues,
- it only supports 10Mbit/s Links, and it's astonishing how much network equipment today does not support that anymore (for good reason).

That being said, I have finally gotten around to solder my breakout board onto a "shield" for better signal integrity, borrowed a proper logic analyzer and will work on rebasing my changes and see how far I get. It'd be a shame to throw this way and I wouldn't want others to need to dig through that mess of code. :smile:

lschuermann, 2022-10-31:
Thanks again for opening this issue! I've created a tracking issue for Ethernet support in Tock: #3308.

closed 2022-10-31 by lschuermann