tock/tock #3116: Release 2.1

bradjc, 2022-08-02:
At the end of July 2022 we discussed doing a release before incorporating the AppID changes. This would 1) provide a tested commit before the AppID changes to the core kernel that developers could use as the infrastructure around credentials is built up, and 2) make it theoretically to do a release _after_ AppID as there would be fewer changes without release-level testing.

This thread initially serves as a place to discuss if this is what we want to do, and if so, what changes/PRs we want to get in for 2.1.

jrvanwhy, 2022-08-02:
One other reason we discussed for releasing 2.1 now is that `libtock-rs` is unsound on Tock 2.0, and releasing 2.1 would allow us to say "`libtock-rs` works on a stable release of Tock".

hudson-ayers, 2022-08-02:
I am in favor of a 2.1 release, and would advocate for not blocking on any outstanding changes other than to say that we will keep the merge window open until we start release testing. I would consider blocking on https://github.com/tock/tock/pull/3056 if it were making more progress, but it seems unlikely to be completed in the immediate near-term.

lschuermann, 2022-08-02:
I'd like to chime in in favor of doing a PR before AppID lands, to have a release we'd be confident would have a relatively stable kernel (just given that the AppID changeset is rather large and touches a significant amount of process code).

Personally, I'd like #2977 to make it in as well. There is known unsoundness in the kernel currently, but merging this changeset might also break some things? My schedule is super full currently in lieau of moving internationally. I can offer to try the following approach: invest a full day and try to finish it in the near future (given it's rather close), and afterwards look at the generated assembly. Then we could judge as to how confident we are in these changes. I don't think that the unsoundness affects current builds, but we can't be 100% sure and future compiler versions might change that.

Furthermore, there is the mutable DMA buffer issue, which is on my table as well. That might require a few more full days of work, though.

phil-levis, 2022-08-03:
I'm in favor of doing 2.1 before AppID lands if
  1. Start the release process soon,
  2. Halt major kernel changes for 2.1,
  3. Merge AppID right after the release. 
 
Maintaining AppID across large kernel changes has been quite difficult. There are still some open questions in it, which would be good to sort out once it is merged, particularly the exact format that different credentials take. I feel confident for the SHA credentials, but think the RSA ones need a careful design as well as an implementation (e.g., on a board with hardware RSA support).

hudson-ayers, 2022-08-04:
> Personally, I'd like https://github.com/tock/tock/pull/2977 to make it in as well. There is known unsoundness in the kernel currently, but merging this changeset might also break some things? My schedule is super full currently given I am moving internationally. I can offer to try the following approach: invest a full day and try to finish it in the near future (given it's rather close), and afterwards look at the generated assembly. Then we could judge as to how confident we are in these changes. I don't think that the unsoundness affects current builds, but we can't be 100% sure and future compiler versions might change that.

I personally would advocate for starting release testing tomorrow, rather than blocking on any PRs. While there is known unsoundness, the fix is not ready, and I don't think this is the only instance of known unsoundness in the kernel (e.g. the mutable buffer DMA issue, and the fact that `static_init!()` can introduce unsoundness without use of unsafe https://github.com/tock/tock/issues/1662 ). I think on tomorrow's core call we can discuss whether tomorrow is a reasonable timeline to start release testing, or if people want to vote for a later hard cutoff for merging PRs (say, 1-2 weeks from now).

bradjc, 2022-08-04:
I agree that we should do the release now. We're probably overdue for a release anyway.

It would be good to get testing done for at least one board before starting the official process to get most issues sorted before the main testing starts. So perhaps we close for the release in a week.

hudson-ayers, 2022-08-05:
I can volunteer to try the Imix tests tomorrow, and see if anything major shows up

hudson-ayers, 2022-08-05:
# Imix

- Tests run on most Boards
  - [x] `examples/sensors`
    - humidity still reads as 0 (same as 2.0)
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - The board I brought to the east coast with me still has R6 attached, so no BLE tests.
  - [ ] sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - Fails when run simultaneously (expected: console not virtualized)
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - **FAIL** `examples/tests/console_timeout` --> reentered grant region
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
  - [x] `examples/tests/adc_continuous`
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`

 
- Imix-specific
  - [ ] `examples/tests/imix` (requires bluetooth)
  - [ ] `examples/ble-uart` (requires bluetooth)
  - [x] `examples/ip_sense`: Application that samples a sensor and puts it in a UDP broadcast segment
  - **FAIL** `examples/tests/udp/udp_send`: sends a UDP datagram to a destination port:
    - `Error Sending Packet: -1`
  - [ ] `examples/tests/udp_receive`: receives datagrams on a port an blinks when received: pair with udp_send
  -  [ ] **In-Kernel Tests** (waiting on these)
      - [ ] `boards/imix/src/test/aes_ccm_test.rs`
      - [ ] `boards/imix/src/test/aes_test.rs`
      - [ ] `boards/imix/src/test/icmp_lowpan_test.rs`
      - [ ] `boards/imix/src/test/linear_log_test.rs`
      - [ ] `boards/imix/src/test/log_test.rs`
      - [ ] `boards/imix/src/test/rng_test.rs`
      - [ ] `boards/imix/src/test/udp_lowpan_test.rs`
          - [ ] `start()`
          - [ ] `start_rx()`
          - [ ] `start_with_app()`
          - [ ] `start_dual_rx()`
      - [ ] `boards/imix/src/test/virtual_uart_rx_test.rs`
      - [ ] `boards/imix/src/test/multi_alarm_test.rs`

- Miscellaneous
  - [x] **Scheduler Tests**
    - [x] `whileone` + `blink`
    - [x] restart `whileone` using process console
    - [x] stop and start `whileone`
    - [x] `whileone` + `blink` + `hello_loop` using MLFQ scheduler
      - Note: See timeslice expirations for blink + hello_loop, which I did not expect

hudson-ayers, 2022-08-05:
So far, just two significant failures. `getnstr_abort()` seems to be causing a re-entered grant region panic in `console_timeout`, and then the UDP tests are failing. I need to dig some more into the UDP tests, but those are unlikely to affect other boards anyway (with the exception of the nrf52dk).

hudson-ayers, 2022-08-09:
Turns out the UDP failure was just that I was missing a jumper, and I found a fix for the other failure. So no remaining problems that I have found which should affect other boards, and I think we are ok to start general release testing.

hudson-ayers, 2022-08-09:
Per the discussion on the last core call, I wanted to organize release testing. Now that I have all of the tests I can run on my Imix passing, we are ready to open up to full release testing. I have tagged a testing release: https://github.com/tock/tock/releases/tag/release-2.1-rc1 .

This testing list is just borrowed from the 2.0 release testing, if you cannot test the board I have assigned to you please make a comment indicating you need someone else to test it. If you can, go ahead and check the box next to your name.

@tock/core-wg @tock/opentitan-wg 

General release testing board signup list:

- [ ] acd52832 
- [ ] arty_e21 (@alexandruradovici)
- [ ] clue_nrf52840 (@alexandruradovici)
- [ ] earlgrey-nexysvideo (@alistair23 )
- [ ] hail (@ppannuto)
- [x] imix (@hudson-ayers )
- [ ] imxrt1050-evkb (@valexandru )
- [ ] litex/sim (@lschuermann )
- [ ] litex/fpga (@lschuermann )
- [ ] microbit_v2 (@brghena)
- [ ] msp_exp432p401r (@lebakassemmerl )
- [ ] nano33ble (@alexandruradovici)
- [x] nrf52dk (@hudson-ayers )
- [ ] nrf52840dk (@lschuermann)
- [ ] nrf52840_dongle (@ppannuto )
- [ ] nucleo_f429zi (@alexandruradovici )
- [ ] nucleo_f446re (@alexandruradovici )
- [ ] raspberry pi pico (@alexandruradovici)
- [ ] redboard artemis nano (@alistair23 )
- [ ] stm32f3discovery (@alexandruradovici)
- [ ] stm32f412gdiscovery (@alexandruradovici)
- [ ] swervolf (@alistair23 )
- [ ] teensy40 (@mciantyre )
- [ ] weact_f401ccu6 (@yusefkarim )

Here is the testing template. Please paste this to a new comment and mark off tests as they are confirmed to work. Also feel free to add more tests, this is just a pretty good starting point. And note, not all tests are expected to work on all boards. A good starting point for any board is to look at the tests performed at the last release: https://github.com/tock/tock/issues/2429 .

```
# <Board Name>

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`
```

If you have new non-board specific tasks to add to this list, please edit this comment or reply to this comment to mention them, and I will add them!

mciantyre, 2022-08-11:
# Teensy 4.0

✅ Passed the same tests from [the 2.0 release test](https://github.com/tock/tock/issues/2429#issuecomment-899931340). Tested with Tock `release-2.1-rc1` and libtock-c commit `1518ff49a`.

- [x] `examples/c_hello` and `examples/tests/printf_long`
  - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
- [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
- [x] `examples/blink`
- [x] `examples/rot13_client` and `examples/rot13_service`
- [x] `examples/blink` and `examples/c_hello` ~~and `examples/buttons`~~
  - *No buttons available.*
- [x] `examples/lua-hello`
- [x] `examples/tests/console_timeout`
- [x] `examples/tests/malloc_test01`
- [x] `examples/tests/malloc_test02`
- [x] `examples/tests/stack_size_test01`
- [x] `examples/tests/stack_size_test02`
- [x] `examples/tests/mpu_stack_growth`
- [x] `examples/tests/mpu_walk_region`
  - *No buttons available.*
- [x] `examples/tests/multi_alarm_test`
  - *Only one configured LED on board; combined with `whileone` to make it interesting.*
- [x] `examples/tests/whileone`

alistair23, 2022-08-12:
- [ ] OpenTitan (earlgrey-cw310)
  - [X] Unit tests
  - [ ] Board/app tests
- [ ] redboard artemis
  - [ ] Unit tests
  - [ ] Board/app tests
- [ ] swervolf
  - [ ] Manual tests
- [ ] ESP32-C3
  - [ ] Unit tests
  - [ ] Board/app tests

# OpenTitan (earlgrey-cw310)

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [X] ~~`examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`~~
    - ~~sense_to_ble: Application that samples a sensor and advertises value with BLE~~
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [X] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

#  redboard artemis nano

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [X] ~~`examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`~~
    - ~~sense_to_ble: Application that samples a sensor and advertises value with BLE~~
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
[... 36 lines cut ...]
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [X] ~~`examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`~~
    - ~~sense_to_ble: Application that samples a sensor and advertises value with BLE~~
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`

jettr, 2022-08-12:
I would like to suggest adding a line to the release note/breaking changes:

 * `VirtualMuxAlarm`s now requires the `setup()` function to be called in board set up code after they are created (e.g. [link](https://github.com/tock/tock/blob/master/boards/litex/arty/src/main.rs#L314)) after this [change](https://github.com/tock/tock/commit/3e6adf5f4f79589371c7dda8a9d342ab055d758d)

brghena, 2022-08-15:
# Microbit v2

Actually tested on a Microbit v2.2.

- General Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - **Still fails for 2.0** -2 EBUSY
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
    - Not sure if this is working or failing? See https://github.com/tock/libtock-c/issues/290
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
    - ADC on the nRF52 does not support retrieving a buffer of samples
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [x] `examples/tests/gpio` with mode set to 0
    - Requires modifying the app to use pin 8 for interrupts
  - [x] `examples/tests/whileone`
  

 - Microbit-specific Tests
   - [x] `examples/accel-leds`
   - [x] `examples/music`
   - [x] `examples/tests/ble/ble_advertise`

 - Removed Tests
   - ~[ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`~
     - ~sense_to_ble: Application that samples a sensor and advertises value with BLE~
     - Only works with BLE serialization from an nRF51822. Not supported by the Microbit v2
   - ~[ ] `examples/tests/adc_continuous`~
     - ADC on the nRF52 does not support continuous sampling

hudson-ayers, 2022-08-19:
With https://github.com/tock/tock/pull/3152, all Imix tests now pass.

hudson-ayers, 2022-08-19:
# nrf52dk

* All Boards
  * [x]  `examples/sensors`
  * [x]  `examples/c_hello` and `examples/tests/printf_long`
  * [x]  `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    Only individually - as expected
  * [x]  `examples/blink`
  * [x]  `examples/rot_client` and `examples/rot_service`
  * [x]  `examples/blink` and `examples/c_hello` and `examples/buttons`
  * [x]  `examples/lua-hello`
  * [x]  `examples/tests/console_timeout`
  * [x]  `examples/tests/malloc_test01`
  * [x]  `examples/tests/malloc_test02`
  * [x]  `examples/tests/stack_size_test01`
  * [x]  `examples/tests/stack_size_test02`
  * [x]  `examples/tests/mpu_stack_growth`
  * [x]  `examples/tests/mpu_walk_region`
     regular loop does not crash, button presses crash both reads as expected
  * [x]  `examples/tests/multi_alarm_test`
     slow, peaceful blinking of each LED in order
  * [x]  `examples/tests/adc`
     `No ADC driver!`
  * [x]  `examples/tests/adc_continuous`
     `No ADC driver!`
  * [x]  `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
     `Number of LEDs: 4` .. then each LED randomly turns on and off
  * [x]  `examples/tests/gpio` with mode set to 0
    Touch `P0.03` to Vdd (randomly tested with a few other pins to _not_ fire an interrupt):
 
* nRF specific
  * [x]  `examples/ble_advertising`
  * [x]  `examples/ble_passive_scanning`

hudson-ayers, 2022-08-19:
# Nano 33 BLE

- Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [x] `examples/ble_advertising`
     - Should a device named "TockOS" with a "Health Thermometer" (1809) Service and refuses connections.
  - [x] examples/ble_passive_scanning
     - Should find some nearby devices (large printout featuing hex-encoded MAC addresses and data)
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - Fails with "[SHORT] Error during UART receive: -2", expected given console is not virtualized.
    - apps work individually
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/malloc_test02`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
    - works, no user button to test overrun though
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
    - works, but only for single sample mode
  - [x] `examples/tests/adc_continuous`
    - Fails as continuous sampling is not supported
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [x] `examples/tests/gpio` with mode set to 0
    - works, but you have to modify the test to pass pin 2 instead of pin 0, since pin 0 is now unassigned
  - [x] `examples/tests/whileone`

lschuermann, 2022-08-23:
# LiteX Arty-A7

(tests have been conducted with the patches from #3149 and on the gateware of tock-litex 2022081701)

- Tests run on most Boards
  - [x] `examples/c_hello` and `examples/tests/printf_long`
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - Fails when run simultaneously (expected: console not virtualized)
  - [x] `examples/blink`
  - [x] `examples/rot_client` and `examples/rot_service`
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
    - no button support as of now, but blink and c_hello work happily together
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
    - no button support, overrun of both flash and RAM triggered manually
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
    - reports "Number of LEDs: 4", nothing happens as there is no RNG present

- Not applicable:
  - `examples/lua-hello`: only available on ARM
  - `sense_to_ble`: Application that samples a sensor and advertises value with BL
  - `examples/sensors`: No sensors present
  - `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - fails (as expected) given lack of BLE hardware
  - `examples/tests/adc` (`No ADC driver!`, which is correct)
  - `examples/tests/adc_continuous` (`No ADC driver!`, which is correct)

bradjc, 2022-08-23:
# Hail

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [ ] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
    - sense_to_ble: Application that samples a sensor and advertises value with BLE
  - [ ] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [ ] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.5 (console capsule not fully virtualized), and might work for 1.6.
  - [ ] `examples/blink`
    - blink: blinks LEDs
  - [ ] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
  - [ ] `examples/tests/adc_continuous`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`
 

- Hail
  - [ ] `examples/tests/hail`
  - [ ] `examples/witenergy`
  - [ ] `examples/accel_leds`
  - [ ] `examples/ble-uart`
  - [ ] `examples/tests/hail` and `examples/sensors`

alexandruradovici, 2022-08-24:
# STM32F412g Discovery

- Tests
  - [ ] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - fails with [SHORT] Error doing UART receive: -2
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [ ] `examples/screen`
    - fails, working to fix it
  - [ ] `examples/lvgl`
  - [x] `examples/touch`
  - [ ] `examples/tests/console_timeout`
    - faults with `Attempted to re-enter a grant region.`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
    - works, but displays some weird values when connected to 3.3 V
  - [ ] `examples/tests/adc_continuous`
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [x] `examples/tests/gpio` with mode set to 0
  - [x] `examples/tests/whileone`

alexandruradovici, 2022-08-26:
# Nucleo429zi

- Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
    - depends on #3159 
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - fails with [SHORT] Error doing UART receive: -2
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
    - depends on #3160 
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
    - test might be wrong, it computes the millivolts using 12 bits (should be u16 as per trait)
    - depends on #3165
  - [x] `examples/tests/adc_continuous`
    - fails as continuous sampling is not supported
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
    - depends on #3164
  - [x] `examples/tests/gpio` with mode set to 0
  - [x] `examples/tests/whileone`

bradjc, 2022-08-26:
I tried all of the Tock 2.0 tabs (https://github.com/tock/tock-www/tree/master/assets/tabs) on hail and they all still work.

brghena, 2022-08-26:
This message is to ping @lebakassemmerl (msp_exp432p401r) and @yusefkarim (weact_f401ccu6) and see if you're able to do board testing in the near future (within this week?).

hudson-ayers, 2022-08-26:
# nRF52840dk

- Tests
  - [x] `examples/sensors`
         26 deg C
  - [x] `examples/services/ble-env-sense` and `examples/services/ble-env-sense/test-with-sensors`
        Fails with `BLE ERROR Code = 16385`, expected because no nRF51 serialization driver is present on the board.
  - [x] `examples/ble_advertising`
         Advertises a device named "TockOS" with a "Health Thermometer" (1809) Service and refuses connections.
  - [x] `examples/ble_passive_scanning`
         Seems to find some nearby devices (large printout featuing hex-encoded MAC addresses and data)
  - [x] `examples/c_hello` and `examples/tests/printf_long`
         Outputs "Hello World!\nHi welcome to Tock. This test makes sure that a greater than 64 byte message can be printed.\nAnd a short message."
         Virtualization works.                                           
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
         Fails with "[SHORT] Error during UART receive: -2", expected given console is not virtualized.
  - [x] `examples/blink`
  - [x] `examples/rot_client` and `examples/rot_service`
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
        ~`examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
         After a short duration, returns characters sent to the app.
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/malloc_test02`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
         Quickly faults with the stack memory exceeded.
  - [x] `examples/tests/mpu_walk_region`
         Works without button 0 pushed, faults when overrunning both flash and RAM. 
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc` and `examples/tests/adc_continuous`
        Error "No ADC driver!", which is correct.
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
        Mesmerizing random LED patterns.
  - [x] `examples/tests/gpio` with mode set to 0
         Tricky, given that on this board with it's additional pins GPIO 0 is on `P1.01` (Arduino D0) instead of `P0.00` :)
  - [x] `examples/tests/whileone`
        Process console confirms timeslice expirations

yusefkarim, 2022-08-27:
# weact_f401ccu6

NOTE 1: I didn't test parallel apps, I don't have easy way to upload two apps for now. If anyone can recommend a fast way for me to achieve parallel apps, I can complete parallel tests.

NOTE 2: The [WeAct has discontinued their STM32F401CCU6 board](https://github.com/WeActStudio/WeActStudio.MiniSTM32F4x1#the-parameters-of-the-board-chip-we-produced-are-compared), they now only sell a STM32F401CEU6 and STM32F411CEU6 variant. If you want to remove support for the WeAct board, you are welcome. If not, I can still test for you all as releases come through.

- Tests
  - [x] `examples/sensors`
    - Don't have sensors, but I observed app running and continually print the newline each interrupt
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - "Hello World!"
    - "Hi welcome to Tock. This test makes sure that a greater than 64 byte message can be printed.
                                                                                            And a short message."
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
      - Didn't test in parallel
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - Didn't test in parallel
    - console_recv_short: sfdsfsdfsdf
    - console_recv_long: aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa
  - [x] `examples/blink`
  - [ ] `examples/rot_client` and `examples/rot_service`
    - Both run individually, didn't test in parallel
  - [ ] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
    - **Panicked after timeout**
    - `panicked at 'Attempted to re-enter a grant region.', kernel/src/grant.rs:1372:17
	Kernel version release-2.1-rc1`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
    - Panicked (expected)
  - [x] `examples/tests/mpu_walk_region`
    - Faulted when KEY button pressed (expected), works otherwise
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
    - Single Samples - Channel 0, 1, 2, 3, 4, and 5  -> works
    - Buffered Samples - Channel 0, 1, 2, 3, 4, and 5 -> `Error sampling ADC: -6`
  - [x] `examples/tests/adc_continuous`
    - `ADC driver exists with 6 channels\nset buffer error: -6`
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
    - Skipped, no parallel apps
  - [x] `examples/tests/gpio` with mode set to 0
    - **Does not work**.
    - This board doesn't have pin 0, so I used pine 11 (PA1). The interrupt did not get triggered. I had this problem for release 2.0, but I believe it was fixed (see #2782),
    - I tested with mode = 2 to double check my pin logic was actually changing (it was), so something with interrupt again
  - [x] `examples/tests/whileone`

alexandruradovici, 2022-08-27:
# Nucleo446re

depends on #3166 and #3174

- Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
    - depends on #3159 
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - fails with [SHORT] Error doing UART receive: -2
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
    - depends on #3160 
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
    - as it works on other stm boards with a similar mcu, it probably works here too, the board has only one LED so only one alarm is actually set
  - [x] `examples/tests/adc`
    - test might be wrong, it computes the millivolts using 12 bits (should be u16 as per trait)
    - depends on https://github.com/tock/libtock-c/pull/291
    - depends on #3165
  - [x] `examples/tests/adc_continuous`
    - fails as continuous sampling is not supported
  - [ ] ❌ `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
    - depends on #3164
    - kernel hardfault #3173 
  - [ ] ❌ `examples/tests/gpio` with mode set to 0
   - does not seem to trigger interrupts
  - [x] `examples/tests/whileone`

alexandruradovici, 2022-08-27:
# STMe32F303 Discovery

depends on #3166 

❌ The board reboots upon faulting while displaying the fault message

- Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - fails with [SHORT] Error doing UART receive: -2
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
    - it timeouts only when a character is read, working to fix it
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
    - test might be wrong, it computes the millivolts using 12 bits (should be u16 as per trait)
    - depends on https://github.com/tock/libtock-c/pull/291
    - works, but the process seems te enter in infinite loop after a few buffered sampling calls, event though the board returns an error
  - [ ] ❌ `examples/tests/adc_continuous`
    - kernel hardfault
  - [ ] ❌  `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
    - kernel hardfault
    - this should fail as there is no rng implemented
  - [ ] ❌ `examples/tests/gpio` with mode set to 0
    - does not seem to trigger interrupts
  - [x] `examples/tests/whileone`

alexandruradovici, 2022-08-28:
# CLUE nRF52840

- Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - fails with [SHORT] Error doing UART receive: -2
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/lua-hello`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tests/multi_alarm_test`
  - [x] `examples/tests/adc`
    - test might be wrong, it computes the millivolts using 12 bits (should be u16 as per trait)
    - depends on https://github.com/tock/libtock-c/pull/291
  - [x] `examples/tests/adc_continuous`
    - fails as continuous sampling is not supported
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [x] `examples/tests/gpio` with mode set to 0
    - used gpio 6
    - stack overflows when many interrupts are triggered, possible from printf (which yields)?l
  - [x] `examples/tests/whileone`

-CLUE-specific Tests
  - [x] `examples/screen`
    - depends on https://github.com/tock/libtock-c/pull/292
  - [x] `examples/lvgl`
  - [x] examples/music
  - [x] examples/tests/ble/ble_advertise
  - [x] examples/ble/passive_scanning

valexandru, 2022-08-29:
# i.MX RT 1052 EVKB

- Tests
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [x] `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [x] `examples/tests/console_timeout`
  - [x] `examples/tests/malloc_test01`
  - [x] `examples/tests/stack_size_test01`
  - [x] `examples/tests/stack_size_test02`
  - [x] `examples/tests/mpu_stack_growth`
  - [x] `examples/tests/mpu_walk_region`
  - [x] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [x] `examples/tests/whileone`

bradjc, 2022-08-29:
@alexandruradovici It seems like we need to figure out these kernel hardfaults. Do you have the panic handler output?

> # STM32F303 Discovery
>
> * ❌ `examples/tests/adc_continuous`  
>   * kernel hardfault
> * ❌ `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
>   * kernel hardfault

bradjc, 2022-08-29:
# Release Sign Off

Please check next to your name when you approve releasing Tock 2.1.

- [ ] Hudson Ayers, [hudson-ayers](https://github.com/hudson-ayers), Stanford
- [ ] Brad Campbell, [bradjc](https://github.com/bradjc), UVA
- [ ] Branden Ghena, [brghena](https://github.com/brghena), Northwestern
- [ ] Philip Levis, [phil-levis](https://github.com/phil-levis), Stanford
- [ ] Amit Levy (chair), [alevy](https://github.com/alevy), Princeton University
- [ ] Pat Pannuto, [ppannuto](https://github.com/ppannuto), UCSD
- [ ] Leon Schuermann, [lschuermann](https://github.com/lschuermann), University of Stuttgart
- [ ] Johnathan Van Why, [jrvanwhy](https://github.com/jrvanwhy), Google

alexandruradovici, 2022-08-29:
> @alexandruradovici It seems like we need to figure out these kernel hardfaults. Do you have the panic handler output?

No, it starts printing out the panic and reboots immediately.

hudson-ayers, 2022-08-29:
Does the tutorial_05 failure at least go away after the latest change to remove the RNG from that board?

bradjc, 2022-08-29:
> > > # STM32F303 Discovery
> > > 
> > > * ❌ `examples/tests/adc_continuous`
> > >   
> > >   * kernel hardfault

> 
> No, it starts printing out the panic and reboots immediately.
> 
> EDIT: if it helps we can try a live debugging meeting

The stm32f303 adc doesn't support continuous sampling:

https://github.com/tock/tock/blob/6e41531c3e75bf79607b4dd8020bf0db7dbee13c/chips/stm32f303xc/src/adc.rs#L696-L703

I modified the sam4l adc to return the same error:

```diff
diff --git a/chips/sam4l/src/adc.rs b/chips/sam4l/src/adc.rs
index 69306130c..8f6874d6a 100644
--- a/chips/sam4l/src/adc.rs
+++ b/chips/sam4l/src/adc.rs
@@ -659,6 +659,8 @@ impl hil::adc::Adc for Adc {
     /// - `channel`: the ADC channel to sample
     /// - `frequency`: the number of samples per second to collect
     fn sample_continuous(&self, channel: &Self::Channel, frequency: u32) -> Result<(), ErrorCode> {
+        return Err(ErrorCode::NOSUPPORT);
+
         let res = self.config_and_enable(frequency);

         if res != Ok(()) {
```

and I get expected output:

```
[INFO   ]  ----- Waiting for serial port to reconnect...
Initialization complete. Entering main loop.
[Tock] ADC Continuous Test
ADC driver exists with 6 channels
Beginning continuous sampling on channel 0 at 10 Hz
continuous sample error: -10
tock$
```

Does the `adc_continuous` app print anything before the hardfault?

lschuermann, 2022-08-29:
We should validate that #3175 either doesn't change the efficieny of system call return value packing & unpacking or is a strict improvement on both ARM and RISC-V platforms before release signoff. I expect to have the LiteX simulation & QEMU rv32 virt tested today or tomorrow and will verify this as part of these tests.

alexandruradovici, 2022-08-29:
# Raspberry Pi Pico / Pico Explorer Board

WIP

- Tests
  - [x] `examples/sensors`
    - sense_to_console: Application that samples a sensor and outputs values to console
  - [x] `examples/c_hello` and `examples/tests/printf_long`
    - uart_tx_small and uart_tx_large: applications that write to console with small and large buffers; run both in parallel to properly test virtualization
  - [x] `examples/tests/console_recv_short` and `examples/tests/console_recv_long`
    - We expect this fail for 1.6 (console capsule not fully virtualized), and might work for 2.0.
    - fails with [SHORT] Error doing UART receive: -2
  - [x] `examples/blink`
    - blink: blinks LEDs
  - [ ] ❌ `examples/rot_client` and `examples/rot_service`
    - rot_ipc: tests IPC with a simple service
    - the service seems to panics at a `command` system call `strb	r0, [r4, #0]`
    - it does not panic if the contents of `rot13_callback` is commented
  - [x] `examples/blink` and `examples/c_hello` and `examples/buttons`
  - [ ] `examples/lua-hello`
  - [ ] `examples/tests/console_timeout`
  - [ ] `examples/tests/malloc_test01`
  - [ ] `examples/tests/stack_size_test01`
  - [ ] `examples/tests/stack_size_test02`
  - [ ] `examples/tests/mpu_stack_growth`
  - [ ] `examples/tests/mpu_walk_region`
  - [ ] `examples/tests/multi_alarm_test`
  - [ ] `examples/tests/adc`
    - test might be wrong, it computes the millivolts using 12 bits (should be u16 as per trait)
    - depends on https://github.com/tock/libtock-c/pull/291
  - [ ] `examples/tests/adc_continuous`
    - fails as continuous sampling is not supported
  - [ ] `examples/tutorials/05_ipc/led` and `examples/tutorials/05_ipc/rng` and `examples/tutorials/05_ipc/logic`
  - [ ] `examples/tests/gpio` with mode set to 0
  - [ ] `examples/tests/whileone`
- Pico Explorer specific tests
  - [ ] `examples/screen`
    - depends on https://github.com/tock/libtock-c/pull/292
  - [ ] `examples/lvgl`

hudson-ayers, 2022-08-29:
@alexandruradovici and I did some pair-debugging on the stm32f3 issues -- we found one bug (incorrect memory layout) which was probably responsible for the adc_continuous failure. With the correct memory layout, adc_continuous does not fit in the available SRAM, so we can't actually confirm the fix.

However, there is still a bug in the normal `adc` app -- it runs correctly for a while, then enters an infinite loop (probably the exit loop in libtock-c). Removing a panic in the kernel makes this bug go away, and the app runs correctly, even though that panic is never hit in either case. Definitely sounds suspiciously like https://github.com/tock/tock/issues/3109

hudson-ayers, 2022-08-29:
@alexandruradovici do you have a panic print for the rpi pico IPC panic?

bradjc, 2022-08-30:
We have to decide what to do about https://github.com/tock/tock/pull/3175. Even if it is an efficiency improvement (or neutral), if some capsule out there is using Tock 2.0 and used this return type, and then updated to 2.1, their code would break in a way we stated would be stable.

If the decision was entirely up to me, I would revert #3175, update the TRD to match the implementation, and use this as further evidence we should follow Tock's policy of not implementing features we don't use. In general, I feel strongly that we should stick to our stability promises.

However, I'm ok with calling this a hopefully inconsequential bug and basically arguing that we promised to stabilize the TRD, not our bad implementation of it. But, we need to decide what we want to do.

lschuermann, 2022-08-30:
> We have to decide what to do about #3175.

Agreed. I think that unrelated to the downstream impact of this (which might be high) the actual impact on _our_ release is low and a revert is going to be quick and unproblematic. Should we talk about this on Friday then?

bradjc, 2022-08-30:
> the actual impact on _our_ release is low and a revert is going to be quick and unproblematic. Should we talk about this on Friday then?

Well, since we are aware of this problem, it seems like whatever we include in this release we are committing to. Since the entire purpose of this release is to allow libtock-rs development to proceed against a tagged release, I really can't see changing our mind _after_ 2.1.

lschuermann, 2022-08-30:
> I really can't see changing our mind _after_ 2.1.

Sorry, I was trying to say that we can make up our minds synchronously just before the release, assuming we don't tag it until Friday on the call.

bradjc, 2022-09-01:
> If the decision was entirely up to me, I would revert #3175, update the TRD to match the implementation, and use this as further evidence we should follow Tock's policy of not implementing features [...]

Merging #3175 actually _fixed_ libtock-rs, as libtock-rs used the TRD for its implementation, and hasn't been tested against the kernel (since nothing in-tree uses this command return type). As it seems quite unlikely #3175 actually breaks anything for anyone in practice, it seems like we should fix our bug (i.e. leave #3175 in) and document the breaking implementation change.

hudson-ayers, 2022-09-01:
I agree, given that libtock-c used one and libtock-rs used the other we inevitably break one or the other, so we should default to the one in the TRD

lschuermann, 2022-09-04:
# QEMU RISC-V 32-bit "virt" Platform

It seems that this board is currently broken with upstream QEMU, in the sense that it cannot run any apps with Physical Memory Protection Enabled (PMP), at least given Tock's particular memory model. I think that I have identified the appropriate upstream QEMU issue and a potential fix for it: https://lists.nongnu.org/archive/html/qemu-riscv/2022-09/msg00007.html I am not entirely sure why I have not run into this issue before, especially given I used this "board" with apps before. The most likely explanation might be that I used a QEMU version in which this issue (introduced mid-2020) was not included.

Nonetheless, the situation is that we have a board which runs only in QEMU, and for which no release supporting this board exists. I will happily carry out release tests with the patched QEMU version, but we can also exclude it from the release.

bradjc, 2022-09-05:
I pushed releases of both [elf2tab](https://crates.io/crates/elf2tab/0.10.0) and [tockloader](https://pypi.org/project/tockloader/1.9.0/).

bradjc, 2022-09-06:
@phil-levis @jrvanwhy Any further comments/thoughts? https://github.com/tock/tock/issues/3116#issuecomment-1230330230

jrvanwhy, 2022-09-06:
None from me, thanks for the reminder to check the box.

hudson-ayers, 2022-09-06:
@alexandruradovici I am adding a "known issues" section to the release notes -- can you confirm that the stm32f303discovery board still enters a reboot loop during panics, even after https://github.com/tock/tock/pull/3187 and https://github.com/tock/tock/pull/3179 ?

alexandruradovici, 2022-09-06:
> @alexandruradovici I am adding a "known issues" section to the release notes -- can you confirm that the stm32f303discovery board still enters a reboot loop during panics, even after #3187 and #3179 ?

Yes it does still enter the reboot loop.

bradjc, 2022-09-07:
https://github.com/tock/tock/releases/tag/release-2.1

closed 2022-09-07 by bradjc