tock/tock #3141: Release 2.1 Breaking Changes and Release Notes

lschuermann, 2022-08-12:
This issue-thread shall serve as a collection of breaking changes and otherwise noteworthy information to be part of the Tock 2.1 release notes. This is in addition to a final pass over the merged commits before tagging the final release, so that we don't miss anything important. If you feel like something shall be included, add a comment below.

As discussed on the call today, compiling this list of breaking changes was motivated by an inquery from @kupiakos. However, the release issue (#3116) tends to get cluttered with testing notes & feedback. Hopefully it'll make discussions around these breaking changes easier to read when they are kept in a seperate issue.

lschuermann, 2022-08-12:
I would like to suggest adding a line to the release note/breaking changes:

 * `VirtualMuxAlarm`s now requires the `setup()` function to be called in board set up code after they are created (e.g. [link](https://github.com/tock/tock/blob/master/boards/litex/arty/src/main.rs#L314)) after this [change](https://github.com/tock/tock/commit/3e6adf5f4f79589371c7dda8a9d342ab055d758d)

_Originally posted by @jettr in https://github.com/tock/tock/issues/3116#issuecomment-1213306202_

lschuermann, 2022-08-29:
Assuming #3175 stays in, we're breaking the implemented ABI contract for downstream users.

bradjc, 2022-09-01:
I have absolutely no idea how I got the list of PRs for 2.0. I can't seem to figure out whatever tool I used. Best I've got is using the github command line tool: `gh pr list --state merged --limit 5000` which returned this list since the date of the 2.0 release:

```
3184	rp2040: fix adc number of bits	WyliodrinEmbeddedIoT:ro2040_adc_fix	MERGED	2022-08-30 17:21:27 +0000 UTC
3180	stm: fix usart baud	WyliodrinEmbeddedIoT:stm32f303_uart_baud	MERGED	2022-08-29 19:28:44 +0000 UTC
3179	stm: fix memory length for stm32f303	WyliodrinEmbeddedIoT:stm32f3303_memory_fix	MERGED	2022-08-29 19:18:32 +0000 UTC
3175	syscall: Fix SuccessU32U64 format	dcz-self:abi	MERGED	2022-08-29 12:02:07 +0000 UTC
3174	stm: add adc, gpio pins and temperature	WyliodrinEmbeddedIoT:stm32f446re_adc_rng_fixes	MERGED	2022-08-27 14:32:04 +0000 UTC
3169	Update stored kernel version to 2.1	update-kernel-version-2.1	MERGED	2022-08-26 15:57:06 +0000 UTC
3165	stm: use the adc1 channels only (A0-A2, A6, A7)	WyliodrinEmbeddedIoT:stm_adc_num_channels_fix	MERGED	2022-08-26 12:33:50 +0000 UTC
3164	stm: use rng peripherals for all stm32f4xx	WyliodrinEmbeddedIoT:stm_rng_fix	MERGED	2022-08-26 12:25:24 +0000 UTC
3160	stm32f4: use deferred tasks to return buffers	WyliodrinEmbeddedIoT:stm_usart_deferred_tasks	MERGED	2022-08-24 23:15:48 +0000 UTC
3159	Fix temperature_stm driver to use 16 bits	WyliodrinEmbeddedIoT:temperature_stm_driver_fix	MERGED	2022-08-24 22:00:46 +0000 UTC
3156	kernel: wrap config.rs comments	kernel-config-wordwrap	MERGED	2022-08-23 19:37:05 +0000 UTC
3155	pconsole: capitalize BSS	process-console-bss	MERGED	2022-08-23 19:36:21 +0000 UTC
3154	NRF serialization: Handle restarted app in 2.0 world	nrf-serialization-restarts-2.0	MERGED	2022-08-23 18:48:03 +0000 UTC
3153	capsules: fxos8700cq: fix i2c deactivate buf len	fxos8700cq-deactivate-len	MERGED	2022-08-22 18:39:42 +0000 UTC
3152	restore UDP kernel_buffer in event of incorrect length error	hudson-ayers:udp-fix-1	MERGED	2022-08-19 21:09:38 +0000 UTC
3151	tock-registers: release v0.8	tock-registers-v0.8	MERGED	2022-08-19 16:50:08 +0000 UTC
3149	boards/litex: update pinned tock-litex release to 2022081701	lschuermann:dev/tock-litex-update-2022081701	MERGED	2022-08-18 10:10:16 +0000 UTC
3148	boards/esp32-c3-devkitM-1: Prepare for the 2.1 release	alistair23:alistair/esp32-c3	MERGED	2022-08-17 04:17:53 +0000 UTC
3147	arch/cortex-m: Correct end stack printing	alistair23:alistair/arm-stack	MERGED	2022-08-16 23:27:54 +0000 UTC
3145	Notes from Core Working Group call, August 12	core-notes-2022-08-12	MERGED	2022-08-15 02:48:00 +0000 UTC
3144	readmes: update links	doc-links-2.1	MERGED	2022-08-12 18:22:55 +0000 UTC
3143	doc: boards: add CW310 readme	ot-cw310-readme-2.1	MERGED	2022-08-12 18:00:46 +0000 UTC
3142	Update capsule documentation for 2.1 release	capsule-doc-2.1	MERGED	2022-08-12 17:55:07 +0000 UTC
3140	boards/redboard_artemis_nano: Fixup app loading	alistair23:alistair/artemis-app-load	MERGED	2022-08-12 07:56:41 +0000 UTC
3139	boards/redboard_artemis: Workaround Hard Fault Exception	alistair23:alistair/artemis-fixup	MERGED	2022-08-12 07:50:35 +0000 UTC
3136	Fix secondary alarm from firing immediately during alarm callback	jettr:alarm-fix	MERGED	2022-08-10 17:28:18 +0000 UTC
3134	use deferred calls to report aborted reception in sam4l uart	hudson-ayers:sam4l-uart-fix	MERGED	2022-08-09 14:49:02 +0000 UTC
3133	Add core notes for 2022-07-29	jrvanwhy:core-notes-2022-07-29	MERGED	2022-08-09 12:11:41 +0000 UTC
3132	[up-keep] arch: fixup typos	twilfredo:twilfred_fixup_arch_typos	MERGED	2022-08-06 22:46:09 +0000 UTC
3131	Tools: Catch error for size change estimates on new boards	no-new-board-errrorr	MERGED	2022-08-05 19:38:28 +0000 UTC
3130	doc: add core notes aug 5 2022	notes-aug-05	MERGED	2022-08-05 17:45:47 +0000 UTC
3128	boards: housekeeping: fixup typos	twilfredo:twilfred_fixup_boards_typos	MERGED	2022-08-05 00:01:37 +0000 UTC
3126	[trivial] tock-registers: mark two methods as `const`	brandenburg:master	MERGED	2022-08-04 10:21:41 +0000 UTC
3121	capsules: fixup up typos	twilfredo:wilf_fixup_capsule_typos	MERGED	2022-08-03 09:44:48 +0000 UTC
3120	libraries: tickv: Convert panics to errors	alistair23:alistair/tickv-panics	MERGED	2022-08-03 01:41:42 +0000 UTC
3119	tbf: make length errors all NotEnoughFlash	tbf-flash-len-errors	MERGED	2022-08-02 21:52:32 +0000 UTC
3118	Capsules: Add an AirQuality capsule and a CCS811 Capsule	alistair23:alistair/ccs811	MERGED	2022-08-02 03:44:14 +0000 UTC
3117	chips/apollo3: IOM: A collection of fixes for I2C	alistair23:alistair/apollo3-iom	MERGED	2022-08-02 03:15:03 +0000 UTC
3115	capsules: sha256: reset total_length on init	capsules-sha256-fix	MERGED	2022-08-01 21:14:43 +0000 UTC
3113	Correct kernel data/code disambiguation in print_tock_memory_usage	ptmu_symbol_counting	MERGED	2022-07-31 20:28:11 +0000 UTC
3111	boards/litex: README upkeep (fix RISC-V extensions, mention & explain tockloader flash-file support)	lschuermann:dev/readme-update-litex	MERGED	2022-07-27 20:00:44 +0000 UTC
3108	kernel: hil: sensors: AirQualityDriver improvements	alistair23:alistair/air-quality	MERGED	2022-07-26 12:22:02 +0000 UTC
3107	SparkFun RedBoard Artemis Nano Improvements and fixes	alistair23:alistair/redboard	MERGED	2022-07-26 12:19:38 +0000 UTC
3106	Seven Segment Display Capsule	WyliodrinEmbeddedIoT:7_segment_display	MERGED	2022-07-26 07:44:50 +0000 UTC
3105	doc: tbf: Fix Persistent ACL doc	doc-tbf-persistent-acl	MERGED	2022-07-26 04:17:01 +0000 UTC
[... cut ...]
2806	rv32i/pmp: add clarifying comment to enumeration	jettr:kernel-region	MERGED	2021-09-02 21:09:40 +0000 UTC
2804	OpenTitan: Bump the SHA and add CW310 support	alistair23:alistair/opentitan-bump	MERGED	2021-09-01 01:54:03 +0000 UTC
2803	kernel: Add new userspace readable allow syscall	alistair23:alistair/shared-allow	MERGED	2021-08-31 00:45:58 +0000 UTC
2802	add core notes 2021-08-27	gemarcano:master	MERGED	2021-08-27 20:23:40 +0000 UTC
2801	Add LiteX simulator based CI	lschuermann:dev/litex-sim-ci	MERGED	2021-08-27 12:55:13 +0000 UTC
```

ppannuto, 2022-09-01:
[This tool](https://github.com/suchipi/prs-merged-since) gave a slightly prettier output? Still kind of a giant list though.. :/

`$ prs-merged-since --repo tock/tock --tag release-2.0 --format markdown | tee /tmp/pr_log.md`

- stm: add adc, gpio pins and temperature ([#3174](https://github.com/tock/tock/pull/3174) by [@alexandruradovici](https://github.com/alexandruradovici))
- rp2040: fix adc number of bits ([#3184](https://github.com/tock/tock/pull/3184) by [@alexandruradovici](https://github.com/alexandruradovici))
- Update stored kernel version to 2.1 ([#3169](https://github.com/tock/tock/pull/3169) by [@bradjc](https://github.com/bradjc))
- stm: fix usart baud ([#3180](https://github.com/tock/tock/pull/3180) by [@alexandruradovici](https://github.com/alexandruradovici))
- stm: fix memory length for stm32f303 ([#3179](https://github.com/tock/tock/pull/3179) by [@alexandruradovici](https://github.com/alexandruradovici))
- syscall: Fix SuccessU32U64 format ([#3175](https://github.com/tock/tock/pull/3175) by [@dcz-self](https://github.com/dcz-self))
- boards/redboard_artemis: Workaround Hard Fault Exception ([#3139](https://github.com/tock/tock/pull/3139) by [@alistair23](https://github.com/alistair23))
- stm: use rng peripherals for all stm32f4xx ([#3164](https://github.com/tock/tock/pull/3164) by [@alexandruradovici](https://github.com/alexandruradovici))
- stm: use the adc1 channels only (A0-A2, A6, A7) ([#3165](https://github.com/tock/tock/pull/3165) by [@alexandruradovici](https://github.com/alexandruradovici))
- stm32f4: use deferred tasks to return buffers ([#3160](https://github.com/tock/tock/pull/3160) by [@alexandruradovici](https://github.com/alexandruradovici))
- capsules: fxos8700cq: fix i2c deactivate buf len ([#3153](https://github.com/tock/tock/pull/3153) by [@bradjc](https://github.com/bradjc))
- boards/litex: update pinned tock-litex release to 2022081701 ([#3149](https://github.com/tock/tock/pull/3149) by [@lschuermann](https://github.com/lschuermann))
- Fix temperature_stm driver to use 16 bits ([#3159](https://github.com/tock/tock/pull/3159) by [@alexandruradovici](https://github.com/alexandruradovici))
- NRF serialization: Handle restarted app in 2.0 world ([#3154](https://github.com/tock/tock/pull/3154) by [@bradjc](https://github.com/bradjc))
- tock-registers: release v0.8 ([#3151](https://github.com/tock/tock/pull/3151) by [@ppannuto](https://github.com/ppannuto))
- kernel: wrap config.rs comments ([#3156](https://github.com/tock/tock/pull/3156) by [@bradjc](https://github.com/bradjc))
- pconsole: capitalize BSS ([#3155](https://github.com/tock/tock/pull/3155) by [@bradjc](https://github.com/bradjc))
- boards/esp32-c3-devkitM-1: Prepare for the 2.1 release ([#3148](https://github.com/tock/tock/pull/3148) by [@alistair23](https://github.com/alistair23))
- restore UDP kernel_buffer in event of incorrect length error ([#3152](https://github.com/tock/tock/pull/3152) by [@hudson-ayers](https://github.com/hudson-ayers))
- arch/cortex-m: Correct end stack printing ([#3147](https://github.com/tock/tock/pull/3147) by [@alistair23](https://github.com/alistair23))
- boards/redboard_artemis_nano: Fixup app loading ([#3140](https://github.com/tock/tock/pull/3140) by [@alistair23](https://github.com/alistair23))
- Fix secondary alarm from firing immediately during alarm callback ([#3136](https://github.com/tock/tock/pull/3136) by [@jettr](https://github.com/jettr))
- doc: boards: add CW310 readme ([#3143](https://github.com/tock/tock/pull/3143) by [@bradjc](https://github.com/bradjc))
- Notes from Core Working Group call, August 12 ([#3145](https://github.com/tock/tock/pull/3145) by [@brghena](https://github.com/brghena))
- doc: add core notes aug 5 2022 ([#3130](https://github.com/tock/tock/pull/3130) by [@ppannuto](https://github.com/ppannuto))
- Add core notes for 2022-07-29 ([#3133](https://github.com/tock/tock/pull/3133) by [@jrvanwhy](https://github.com/jrvanwhy))
- readmes: update links ([#3144](https://github.com/tock/tock/pull/3144) by [@bradjc](https://github.com/bradjc))
- Update capsule documentation for 2.1 release ([#3142](https://github.com/tock/tock/pull/3142) by [@bradjc](https://github.com/bradjc))
- use deferred calls to report aborted reception in sam4l uart ([#3134](https://github.com/tock/tock/pull/3134) by [@hudson-ayers](https://github.com/hudson-ayers))
- Tools: Catch error for size change estimates on new boards ([#3131](https://github.com/tock/tock/pull/3131) by [@ppannuto](https://github.com/ppannuto))
- [up-keep] arch: fixup typos ([#3132](https://github.com/tock/tock/pull/3132) by [@twilfredo](https://github.com/twilfredo))
- Capsules: Add an AirQuality capsule and a CCS811 Capsule ([#3118](https://github.com/tock/tock/pull/3118) by [@alistair23](https://github.com/alistair23))
- libraries: tickv: Convert panics to errors ([#3120](https://github.com/tock/tock/pull/3120) by [@alistair23](https://github.com/alistair23))
- chips/apollo3: IOM: A collection of fixes for I2C ([#3117](https://github.com/tock/tock/pull/3117) by [@alistair23](https://github.com/alistair23))
- Add QEMU riscv32 "virt" board ([#2516](https://github.com/tock/tock/pull/2516) by [@lschuermann](https://github.com/lschuermann))
- tbf: make length errors all NotEnoughFlash ([#3119](https://github.com/tock/tock/pull/3119) by [@bradjc](https://github.com/bradjc))
[... cut ...]
- I2C for RPi Pico ([#2690](https://github.com/tock/tock/pull/2690) by [@Gerharddc](https://github.com/Gerharddc))
- debug: rename begin_debug_fmt and add non-newline version ([#2724](https://github.com/tock/tock/pull/2724) by [@jettr](https://github.com/jettr))
- kernel: add public process_map_or_external fn ([#2726](https://github.com/tock/tock/pull/2726) by [@jettr](https://github.com/jettr))
- doc: misc updates ([#2799](https://github.com/tock/tock/pull/2799) by [@bradjc](https://github.com/bradjc))

bradjc, 2022-09-06:
This is fantastic.

jrvanwhy, 2022-09-06:
Thanks for the great summary, Leon!

> Other changes include the introduction of a _read-only allow_ system call

Read-only allow was in Tock 2.0, perhaps you mean userspace-readable allow?

hudson-ayers, 2022-09-06:
I updated Leon's comment to fix a couple nits and to add a "Known Issues" section

bradjc, 2022-09-07:
https://github.com/tock/tock/releases/tag/release-2.1

closed 2022-09-07 by bradjc