tock/tock #3173: Kernel hardfault for IPC application

alexandruradovici, 2022-08-27:
When running the ipc (led, rng and logic) applications on a Nucleo446re board, I get a kernel hardfault:

```
␀Initialization complete. Entering main loop
Number of LEDs: 1

panicked at 'Kernel HardFault.
        Kernel version tock-registers-v0.8-33-ge24d4c9df
        r0  0x20002df0
        r1  0x50060800
        r2  0x800e505
        r3  0x1
        r12 0x0
        lr  0x800953f
        pc  0x800e506
        prs 0x1000000 [ N 0 Z 0 C 0 V 0 Q 0 GE 0000 ; ICI.IT 0 T true ; Exc 0-Thread Mode ]
        sp  0x20001900
        top of stack     0x20002000
        bottom of stack  0x20000000
        SHCSR 0x0
        CFSR  0x8200
        HSFR  0x40000000
        Instruction Access Violation:       false
        Data Access Violation:              false
        Memory Management Unstacking Fault: false
        Memory Management Stacking Fault:   false
        Memory Management Lazy FP Fault:    false
        Instruction Bus Error:              false
        Precise Data Bus Error:             true
        Imprecise Data Bus Error:           false
        Bus Unstacking Fault:               false
        Bus Stacking Fault:                 false
        Bus Lazy FP Fault:                  false
        Undefined Instruction Usage Fault:  false
        Invalid State Usage Fault:          false
        Invalid PC Load Usage Fault:        false
        No Coprocessor Usage Fault:         false
        Unaligned Access Usage Fault:       false
        Divide By Zero:                     false
        Bus Fault on Vector Table Read:     false
        Forced Hard Fault:                  true
        Faulting Memory Address: (valid: false) 0x50060800
        Bus Fault Address:       (valid: true) 0x50060800
', arch/cortex-m/src/lib.rs:496:5
        Kernel version tock-registers-v0.8-33-ge24d4c9df

---| No debug queue found. You can set it with the DebugQueue component.

---| Cortex-M Fault Status |---
No Cortex-M faults detected.

---| App Status |---
𝐀𝐩𝐩: org.tockos.tutorials.ipc.led   -   [Yielded]
 Events Queued: 0   Syscall Count: 11   Dropped Upcall Count: 0
 Restart Count: 0
 Last Syscall: Yield { which: 1, address: 0x0 }
 Completion Code: None

 ╔═══════════╤══════════════════════════════════════════╗
 ║  Address  │ Region Name    Used | Allocated (bytes)  ║
 ╚0x20006000═╪══════════════════════════════════════════╝
             │ Grant Ptrs       56
             │ Upcalls         360
             │ Process         880
  0x20005AF0 ┼───────────────────────────────────────────
             │ ▼ Grant          76
  0x20005AA4 ┼───────────────────────────────────────────
             │ Unused
  0x200049FC ┼───────────────────────────────────────────
             │ ▲ Heap            0 |   4264               S
[... 553 lines cut ...]
    Sub-region 3: [0x20008C00:0x20009000], Enabled
    Sub-region 4: [0x20009000:0x20009400], Enabled
    Sub-region 5: [0x20009400:0x20009800], Disabled
    Sub-region 6: [0x20009800:0x20009C00], Disabled
    Sub-region 7: [0x20009C00:0x2000A000], Disabled
  Region 1: [0x08041800:0x08043800], length: 8192 bytes; UnprivilegedReadOnly (0x2)
    Sub-region 0: [0x08041800:0x08041C00], Disabled
    Sub-region 1: [0x08041C00:0x08042000], Disabled
    Sub-region 2: [0x08042000:0x08042400], Disabled
    Sub-region 3: [0x08042400:0x08042800], Enabled
    Sub-region 4: [0x08042800:0x08042C00], Enabled
    Sub-region 5: [0x08042C00:0x08043000], Enabled
    Sub-region 6: [0x08043000:0x08043400], Enabled
    Sub-region 7: [0x08043400:0x08043800], Disabled
  Region 2: Unused
  Region 3: Unused
  Region 4: Unused
  Region 5: Unused
  Region 6: Unused
  Region 7: Unused

To debug, run `make debug RAM_START=0x20008000 FLASH_INIT=0x8041875`
in the app's folder and open the .lst file.

```

bradjc, 2022-08-29:
Does the Nucleo446re board have a RNG on it?

```
Bus Fault Address:       (valid: true) 0x50060800
```

is the most operative line in that panic handler.

In the stm32f4 rng:

https://github.com/tock/tock/blob/e24d4c9df68a1505ad50fad5ab819c63f1dc2c2e/chips/stm32f4xx/src/trng.rs#L13-L14

Maybe something is wrong with that peripheral or its address?

alexandruradovici, 2022-08-29:
It should, let me check. If it does not, the implementation of RNG is in the wrong place.

alexandruradovici, 2022-08-29:
Sorry, Nucleo446re should have the same category of MCU as Nucleo429zi, which works.

alexandruradovici, 2022-08-29:
It seems that stm32f412g and stm32f429zi do have a RNG, while stm32f446re does not have one. I have no other idea on how to fix this other than adding the RNG in each pf the two crates that support it.

In the future, I suggest we add a set of peripherals crates that can be imported by chips.

closed 2022-08-29 by alexandruradovici

bradjc, 2022-08-29:
We do this for the sifive boards. The peripheral code is shared, and each
chip just instantiates the peripherals at the correct address.