tock/tock #3197: Release 2.2

bradjc, 2022-09-07:
With [Tock 2.1](https://github.com/tock/tock/releases/tag/release-2.1) fresh off the presses (in Sep 2022), it is time to start planning Tock 2.2!

This thread serves as a discussion for our goals for the 2.2 release.

## Planned Features for 2.2
- AppID (#2809)
- Revamp of Cortex-M low level syscall handling (fix #3109)

bradjc, 2022-09-07:
Also, I think that now is the time to have the discussion about deprecating boards. That way we have time to 1) make sure they should actually be deprecated, and 2) move them into tock-archive properly.

lschuermann, 2022-09-07:
I would like to see #2977 and the proposed DMABuffer design -- which would also incorporate the mutable/immutable buffer handling -- to be on the roadmap for the next release. I realize those two are on my table, I'll work on getting them ready, just to have that be on the radar.

hudson-ayers, 2022-09-07:
> Also, I think that now is the time to have the discussion about deprecating boards. That way we have time to 1) make sure they should actually be deprecated, and 2) move them into tock-archive properly.

The following boards did not receive release testing:
- imxrt1050-evkb 
- msp_exp432p401r
- arty_e21
- acd52832
- clue_nrf52840

The one other board we might want to consider deprecating is the weact board, since the version we support has been discontinued, though we do have someone available to perform testing for it if we decide to keep it.

alexandruradovici, 2022-09-07:
I tested the clue_nrf52840 and it passed all tests. It is one of the boards that is fully supported, it works perfectly with tockloader and tock bootloader.

I wanted to suggest placing it in Tier 1.

imxrt1050-evkb was tested by @valexandru.

hudson-ayers, 2022-09-07:
Ah sorry I must have missed the testing list for those two boards

brghena, 2024-01-19:
https://github.com/tock/tock/pull/3798 feels important to me. And it'll be merged fairly soon

hudson-ayers, 2024-06-14:
After the call today we had some brief discussion about where this release should land relative to the updated Yield WaitFor PR.

I would like to propose that we reduce the required amount of release testing for minor releases. We should only block on running the full suite of release tests for Imix + nrf52840dk, plus any automated tests that we can run through Leon's hardware CI system, and then set a one week timer for additional testing. Perhaps this would require some updates to our tiered board system. 

At a high level, I think our current approach discourages us too much from making new releases, which has various downsides for the project. 

We would still require a full suite of release testing for major version increases. Happy to hear what others think about this approach!

bradjc, 2024-07-23:
When we do a release, why do we not update the version numbers in the crates? Is there a strategic reason to not do that, or is it more because we have so many crates and there isn't a particularly good reason to go through the effort?

hudson-ayers, 2024-07-24:
I thought it was because updating the version numbers of the crates above 1.0 could imply certain semver guarantees for those crates that we are not ready to make. For example, we aren't ready to commit to no breaking changes of any of the public APIs in the arch crate

bradjc, 2024-07-24:
> version numbers of the crates above 1.0 could imply certain semver guarantees

Oh right, that makes sense.

I want to suggest we consider including the version number with a prepended 0 (i.e., this release would be `0.2.2` in the crates). That's a little odd, but so is always using the version `0.1.0`.

lschuermann, 2024-12-18:
Superseded by #4272.

closed 2024-12-18 by lschuermann