tock/tock #3203: TBF process binary alignment

jrvanwhy, 2022-09-08:
Rust binaries can contain data with arbitrary alignment requirements. For example:

```rust
// Ignore -- this is all to generate a minimal ELF binary.
#![no_main]
#![no_std]
#[panic_handler]
fn panic(_: &core::panic::PanicInfo) -> ! { loop {} }

// Instantiate a type with a large alignment as a static
#[repr(align(32))]
struct Foo { data: [u8; 4] }
static FOO: Foo = Foo { data: [1, 2, 3, 4] };

// Use FOO so it cannot be optimized away
#[no_mangle]
extern fn _start() {
    unsafe {
        core::arch::asm!("mv {x}, {x}", x = in(reg) &FOO);
    }
}
```

generates a `.rodata` section with a required alignment of 32 (`2**5`):

```
jrvanwhy@penguin:~/playground$ riscv64-unknown-elf-objdump -h target/riscv32imc-unknown-none-elf/release/playground

target/riscv32imc-unknown-none-elf/release/playground:     file format elf32-littleriscv

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .rodata       00000020  000100c0  000100c0  000000c0  2**5
                  CONTENTS, ALLOC, LOAD, READONLY, DATA
  1 .text         0000000c  000110e0  000110e0  000000e0  2**1
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  2 .riscv.attributes 00000026  00000000  00000000  000000ec  2**0
                  CONTENTS, READONLY
  3 .comment      00000013  00000000  00000000  00000112  2**0
                  CONTENTS, READONLY
```

As far as I can tell, this alignment requirement is lost in the translation from ELF to TBF, so process binaries can potentially be loaded and executed at addresses that don't meet their alignment requirements.

bradjc, 2022-09-08:
This is almost certainly an elf2tab bug. It looks like we check for 4 byte alignment (https://github.com/tock/elf2tab/blob/master/src/convert.rs#L472) but not a per-section alignment.

jrvanwhy, 2022-09-08:
It seems reasonable for `elf2tab` to check alignment this in non-relocatable binaries, but what about PIC binaries? It seems to me the loader would need to know the alignement requirement.

bradjc, 2022-09-08:
> It seems reasonable for `elf2tab` to check alignment in non-relocatable binaries, but what about PIC binaries? It seems to me the loader would need to know the alignment requirement.

This is a good point, and one that we haven't stumbled upon because the cortex-m MPU causes us to align apps at the largest possible alignment (ie the size of the entire app).

There has always been a tension between ELF and TAB: how much of the information in the elf needs to be shared with a tool like tockloader or some sort of loader service on the board or in the kernel? It seems like we (at least) need to add alignment information to TABs.

bradjc, 2024-07-03:
What's the largest reasonable alignment? We can build something arbitrarily complex (probably a TBF header which specifies the desired alignment) which could help risc-v apps if we ever got good PIC support.

But, it probably just makes more sense to define a minimum alignment all TBFs can expect and guarantee that a TBF will always be loaded at at least that alignment.

IPC does complicate this on cortex-m as we do need to align larger buffers for the MPU. But, that should continue to work because the entire TBF is aligned to a large alignment.

ppannuto, 2024-07-05:
> There has always been a tension between ELF and TAB: how much of the information in the elf needs to be shared with a tool like tockloader or some sort of loader service on the board or in the kernel?

I'm not sure how feasible this is in practice, as IIRC the implementation reads 'what it needs' from the ELF and ignores the rest, but it might be nice to have a debug mode of elf2tab or something where it lets the user know what ELF information has been discarded, as a way of helping making things discoverable.

lschuermann, 2024-07-05:
I don't think there's a reasonable minimum alignment; for any alignment requirement that an ELF allows you to specify, we may always be faced with a program that's relying on that alignment in the future. The [Rust Reference on alignment modifiers](https://doc.rust-lang.org/reference/type-layout.html#the-alignment-modifiers) specifies, for instance:

> The alignment is specified as an integer parameter in the form of #[repr(align(x))] or #[repr(packed(x))]. The alignment value must be a power of two from 1 up to 229. For packed, if no value is given, as in #[repr(packed)], then the value is 1.

I want to avoid imposing some minimum alignment, and then later adding a header anyways. And I do believe it's essential for us to reject applications whose correct flash placement we cannot ensure, whether that's the `elf2tab`, Tockloader, or app loading step.