tock/tock #3204: CI emails broken

dcz-self, 2022-09-09:
I'm not sure if this is misconfigured CI, or github's fault, but since July CI failure emails come without the date header, breaking sorting in my email client.

Last correct email:

```
Date: Tue, 28 Jun 2022 10:59:34 -0700
From: Dorota <notifications@github.com>
Reply-To: tock/tock <noreply@github.com>
To: tock/tock <tock@noreply.github.com>
Cc: Ci activity <ci_activity@noreply.github.com>
Message-ID: <tock/tock/check-suites/CS_kwDOAivZPs8AAAABqQsYeg/1656439163@github.com>
```

An undated bad one:

```
From: Dorota <notifications@github.com>
To: 
Subject: [tock/tock] PR run failed: tock-ci - boards: Add sma_q3 (caa7851)
Cc: Ci activity <ci_activity@noreply.github.com>
List-Id: tock/tock <tock.tock.github.com>
Precedence: list
X-GitHub-Reason: ci_activity
List-Post: noreply@github.com
Message-ID: <tock/tock/check-suites/CS_kwDOAivZPs8AAAAB3-JeXw/1661869606@github.com>
```

This is kind of annoying :) It applies to both tock and libtock-rs runs.

lschuermann, 2022-09-09:
Noticed this as well. Almost certainly GitHub's fault.

closed 2022-09-09 by lschuermann

lschuermann, 2022-11-01:
I have received feedback from GitHub (a while ago) that a fix for this issue has been deployed and can confirm that emails look sensible again.