tock/tock #3217: SiFive chips: Overwriting reserved bits?

mateibarbu19, 2022-09-13:
## Writing to reserved bits: a problem or not?

Hello,

I saw in more than one place that configuring registers for peripheral registers is done using the [write](https://github.com/tock/tock/blob/100b9a47fe37f663d8f94c027989b938275c5186/libraries/tock-register-interface/src/interfaces.rs#L265) call. For example, this small [commit](https://github.com/tock/tock/commit/9f06a7fbfa2ab89e44dc0773771690d5c794dd76) enables UART transmission setting `txctrl::txen`.

As I am a beginner, I was wondering why wasn't the configuration done using [modify](https://github.com/tock/tock/blob/100b9a47fe37f663d8f94c027989b938275c5186/libraries/tock-register-interface/src/interfaces.rs#L306)? The way I see it, the `txctrl` register contains two **reserved** sections and writing to  them would lead to undefined behavior.

@bradjc could you please help me?

---

### Notes

I looked into how SiFive defines it's registers, and saw in their [sifive-blocks repo](https://github.com/sifive/sifive-blocks/blob/545a396f3486132b01ceef3cbce2085608984478/src/main/scala/devices/uart/UART.scala#L148) that they use a `RegFieldGroup`, which I found [here](https://github.com/chipsalliance/rocket-chip/blob/279a87faa47d419c107c5877748dfec643da87b8/src/main/scala/regmapper/RegFieldDesc.scala#L99). But due to my lack of experience I wasn't able to deduce any further conclusions...

bradjc, 2022-09-13:
Hi @mateibarbu19, there are a couple ways to think about what is happening here.

First, a reserved field in a register doesn't mean that writing or reading those bits leads to any sort of undefined behavior, just that software should ignore those bits on a read, and set them to 0 on a write. This is the general expectation of any microcontroller. However, it is _possible_ that the hardware could do or expect something different, but it would have to be very clearly specified in the datasheet.

Second, the smallest unit of writing memory in any modern microcontroller is a single byte. So in this case of the SiFive UART, even if we tried to write only the `txen` and `nstop` bits in the register, we would be forced to write the other six LSBs as well.

Third, the Tock registers library `modify()` function essentially is a three step operation: first it reads the entire register, then it changes only bits in the fields specified by the arguments to `modify()`, then it writes the entire 32 bit value back to the register. So as far as the UART peripheral is concerned, the operation looks effectively the same as the current use of `set()`. However, since we are writing all valid portions of the register, there is no reason to do the first read, as none of the current contents of the register would be preserved anyway.

mateibarbu19, 2022-09-13:
Thank you for your detailed answer. So I should use `write()` when changing an entire register and `modify()` for a small bit field?

May I ask why didn't you disable the transmit interrupt in the synchronous send?

bradjc, 2022-09-13:
> Thank you for your detailed answer. So I should use `write()` when changing an entire register and `modify()` for a small bit field?

Yes.

> May I ask why didn't you disable the transmit interrupt in the synchronous send?

Tock doesn't normally use any synchronous I/O operations. That `transmit_sync()` call is only used to implement the `panic!()` handler. At the point of a panic, the system has effectively crashed, and we weren't too worried about the status of individual interrupts.

closed 2022-09-16 by mateibarbu19