tock/tock #3228: tock_register_interface: 0.8 doesn't allow to use `register_structs!` multiple times in same file anymore

andre-richter, 2022-09-19:
Hi,

it seems that something like the following is not possible anymore when switching from 0.7 to 0.8: https://github.com/rust-embedded/rust-raspberrypi-OS-tutorials/blob/8ab5417c776ddf8bbab3f6d01f4b75f1864e2ed2/19_kernel_heap/kernel/src/bsp/device_driver/bcm/bcm2xxx_interrupt_controller/peripheral_ic.rs#L26-L44

The error this produces is:
```
error[E0428]: the name `static_validate_register_structs` is defined multiple times
  --> kernel/src/bsp/device_driver/bcm/bcm2xxx_interrupt_controller/peripheral_ic.rs:26:1
   |
26 | / register_structs! {
27 | |     #[allow(non_snake_case)]
28 | |     WORegisterBlock {
29 | |         (0x00 => _reserved1),
...  |
33 | |     }
34 | | }
   | |_^ `static_validate_register_structs` redefined here
35 |
36 | / register_structs! {
37 | |     #[allow(non_snake_case)]
38 | |     RORegisterBlock {
39 | |         (0x00 => _reserved1),
...  |
43 | |     }
44 | | }
   | |_- previous definition of the module `static_validate_register_structs` here
   |
   = note: `static_validate_register_structs` must be defined only once in the type namespace of this module
   = note: this error originates in the macro `register_structs` (in Nightly builds, run with -Z macro-backtrace for more info)
```

CC @lschuermann as author of https://github.com/tock/tock/commit/b2f2ae7cba6e8702e22b67a53322e0c1b82d8bce

Might be the removal of the `#[cfg(test)]` that is unmasking this now for me? But having a macro that can only be used once per file would seem a bit odd?

lschuermann, 2022-09-20:
This is very unfortunate. I looked into revision `b2f2ae7~1` and it seems that indeed the removal of `#[cfg(test)]` is unmasking this behavior. I agree that you should be able to have multiple instantiations of this macro in a single module.

These changes coincidentally also moved all of what is generated as part of the `test_fields!` macro into a single `const _: () = { ... };` block. So it should in theory be possible for us to just remove the `mod static_validate_register_structs` module surrounding these checks. I'll look into the expanded macros for this though, to ensure we don't spill any `const`s into the user's module through this macro.

lschuermann, 2022-09-20:
Okay, so this is what the expanded macro looks like without modules surrounding the const-assertions:

https://gist.github.com/lschuermann/854c71bf724e23461981e31473900337#file-register_structs_expanded-rs

I'll create a PR to remove these modules, as it's perfectly fine to have multiple `const _: () = { ... };` spilled out by the macro.

lschuermann, 2022-09-20:
@andre-richter By the way: something which you can do in the meantime is just use one `register_structs!` invocation to define multiple register structs (aka. `structs` being plural is no accident):

```rust
register_structs! {
    #[allow(non_snake_case)]
    WORegisterBlock {
        (0x00 => _reserved1),
        (0x10 => ENABLE_1: WriteOnly<u32>),
        (0x14 => ENABLE_2: WriteOnly<u32>),
        (0x18 => @END),
    },

    #[allow(non_snake_case)]
    RORegisterBlock {
        (0x00 => _reserved1),
        (0x04 => PENDING_1: ReadOnly<u32>),
        (0x08 => PENDING_2: ReadOnly<u32>),
        (0x0c => @END),
    }
}
```

v0.8.0 will also error on this definition given that `@END` for `WORegisterBlock` should be at address `0x18` instead of `0x24`.

andre-richter, 2022-09-20:
Hah, I tried putting them under the same invocation yesterday night but got an error and was wondering why it doesn’t work!

I guess I wasn’t reading the error message carefully in this case 😅